mr-manuel/venus-os_dbus-serialbattery #20: Stuck in absorption, although BMS OVP hit

wasnertobias, 2024-02-28:
### Describe the problem

As the reported SOC of my JKBMS was way lower than expected (85 % instead of 100 %) I tried the setting SOC_RESET_VOLTAGE. However, although (multiple) cells hit the OVP of the BMS (3.65 V) it does not seem to go back into float mode.

I also enabled SOC_CALCULATION, as this is expected to yield much better results.

My biggest problem is: **It does not let me discharge the battery**, the Grid-Setpoint is not being followed any longer. I assume this happens as it is still stuck with "Bulk".

(I tried to disable the feature by leaving "SOC_RESET_AFTER_DAYS" blank again.)

[image]

[image]
[image]
[image]
[image]
[image]
[image]

### Driver version

v1.2.20240227beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.22

### BMS type

JKBMS / Heltec BMS

### Cell count

8

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

BMS_TYPE = Jkbms
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:EE:2B:05

VOLTAGE_DROP = 0.76

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.600
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.550
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.633
SOC_RESET_AFTER_DAYS = 14

; --------- SOC calculation ---------
SOC_CALCULATION   = True
SOC_RESET_CURRENT = 7
SOC_RESET_TIME    = 60

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING      = 3.633, 3.55, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION    =     0, 0.01, 0.05,  0.5,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.60, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1
```

### Relevant log output

```shell
2024-02-28 10:08:09.583138500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-28 10:08:09.584070500 INFO:SerialBattery:Venus OS v3.22
2024-02-28 10:08:09.584350500 INFO:SerialBattery:dbus-serialbattery v1.2.20240227beta
2024-02-28 10:08:25.869683500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EE:2B:05
2024-02-28 10:08:25.869947500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EE:2B:05
2024-02-28 10:08:30.382036500 INFO:SerialBattery:BAT: JKBMS 11.XW 8 cells (20231014)
2024-02-28 10:08:30.382897500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-02-28 10:08:30.383520500 INFO:SerialBattery:SOC initialized from BMS and set to 95%
2024-02-28 10:08:30.384009500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c8478cee2b05
2024-02-28 10:08:30.384265500 INFO:SerialBattery:========== Settings ==========
2024-02-28 10:08:30.384623500 INFO:SerialBattery:> Connection voltage: 29.17V | Current: 3.7A | SoC: 95.0%
2024-02-28 10:08:30.384933500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2024-02-28 10:08:30.385246500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-02-28 10:08:30.385584500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.6V | MAX CELL VOLTAGE: 3.55V
2024-02-28 10:08:30.385956500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A
2024-02-28 10:08:30.386264500 INFO:SerialBattery:> CVCM:     True
2024-02-28 10:08:30.386562500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-02-28 10:08:30.386846500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-02-28 10:08:30.387131500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-02-28 10:08:30.387409500 INFO:SerialBattery:Serial Number/Unique Identifier: 3050444076
2024-02-28 10:08:30.539308500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-02-28 10:08:30.686160500 INFO:SerialBattery:DeviceInstance = 1
2024-02-28 10:08:30.687100500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-02-28 10:08:30.687390500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-02-28 10:08:30.687679500 INFO:SerialBattery:com.victronenergy.battery.ble_c8478cee2b05
2024-02-28 10:08:30.688085500 INFO:SerialBattery:BAT: JKBMS 11.XW 8 cells (20231014)
2024-02-28 10:08:30.703026500 INFO:SerialBattery:publish config values = True
2024-02-28 10:09:32.130179500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:32.673536500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:33.236202500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:33.761060500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:34.323834500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:34.869437500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:35.393140500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:35.955407500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:36.480310500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:37.042731500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:37.586305500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:38.130020500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:38.674305500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:39.217834500 INFO:SerialBattery:SOC set to 100%
2024-02-28 10:09:44.111621500 INFO:SerialBattery:SOC set to 100%
```

mr-manuel, 2024-02-28:
> As the reported SOC of my JKBMS was way lower than expected (85 % instead of 100 %) I tried the setting SOC_RESET_VOLTAGE. However, although (multiple) cells hit the OVP of the BMS (3.65 V) it does not seem to go back into float mode.

This are two different things.
1. JKBMS resets the SOC to 100%, if one cell reaches OVP.
2. The driver switches to float, if `SOC_RESET_VOLTAGE` * cell count is reached. This means that if you set `SOC_RESET_VOLTAGE` to `3.65` the driver will not switch to float, because this situation is very unlikely. What was the battery voltage when the first cell hit OVP? Get this voltage divide it by your cell count and use it as `SOC_RESET_VOLTAGE`. This is only an initial value and you have to play with it. If OVP is never reached, increase it a bit. If `SOC_RESET_VOLTAGE` * cell count is never reached, decrese it a bit.

Great that it lets you discharge again. Else I would had proposed a reboot of the Venus OS.

ak68-hub, 2024-05-01:
CAVE: Dieser Fehler trat bei mir nach einem Multiplus-Firmware-Upgrade auf :(
Alle Spannungsschwellen müssen per VE.BUS System Connect neu eingestellt werden !