mr-manuel/venus-os_dbus-serialbattery #21: Eelectrical power is displayed incorrectly

Kapitaen31, 2024-03-08:
### Describe the problem

For me, the electrical power and current are displayed too small by a factor of 2. This is when loading and unloading.
I don't know where I can change that.
Can you help me?

### Driver version

1.1.20240121

### Venus OS device type

Raspberry Pi

### Venus OS version

3.12

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 70.0
FLOAT_CELL_VOLTAGE = 3.4
BMS_TYPE = LltJbd 
SOC_LOW_WARNING = 0
SOC_LOW_ALARM = 0
CUSTOM_BATTERY_NAMES = /dev/ttyACM0:KapitaensHall1
```

### Relevant log output

```shell
@4000000065dcee122152b554 INFO: Start service gps-dbus.ttyACM0 once
@4000000065dcee1b32d5735c INFO: Start service vedirect-interface.ttyACM0 once
@4000000065dcee1f037af3fc INFO: Start service dbus-serialbattery.ttyACM0 once
@4000000065dcef5f09580954 INFO: Start service mk2-dbus.ttyUSB1
@4000000065d6ef1c38b5dc44 *** CCGX booted (30253) ***
@4000000065d6ef1d2390261c *** starting serial-starter ***
@4000000065d6ef1d35dd8b0c serstart starting
@4000000065d6ef1d3631d52c INFO: loading config file /etc/venus/serial-starter.conf
@4000000065d6ef1e2eb5d8fc INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000065d6ef1f338c4bf4 ERROR: no service template for dbus-serialbattery
@4000000065e2b3e61d3c757c INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@4000000065e2b3e81edd7f34 INFO: Create daemontools service mk2-dbus.ttyUSB1
@4000000065e2b3ec2adca224 INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065e2b3ec2fdb4dfc INFO: Create daemontools service gps-dbus.ttyACM0
@4000000065e2b3ee33643a24 INFO: Start service mk2-dbus.ttyUSB1
@4000000065e2b3f239bdbb0c INFO: Start service gps-dbus.ttyACM0 once
@4000000065e2b3fc1cde7ae4 INFO: Create daemontools service vedirect-interface.ttyACM0
@4000000065e2b4022012fdac INFO: Start service vedirect-interface.ttyACM0 once
@4000000065e2b4032ec61d0c INFO: Create daemontools service dbus-serialbattery.ttyACM0
@4000000065e2b40931a77aac INFO: Start service dbus-serialbattery.ttyACM0 once
@4000000065e342e139ac7cfc *** CCGX booted (0) ***
@4000000065e342e218b18a74 *** starting serial-starter ***
@4000000065e342e22ab3d0ec serstart starting
@4000000065e342e22b9bb434 INFO: loading config file /etc/venus/serial-starter.conf
@4000000065e342e31ef1322c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000065e342e426b2da4c INFO: Create daemontools service dbus-serialbattery.ttyACM0
@4000000065e342f528fd00fc INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@4000000065e342f70fca8c94 INFO: Create daemontools service mk2-dbus.ttyUSB1
@4000000065e342f915ecb19c INFO: Start service dbus-serialbattery.ttyACM0 once
@4000000065e342fb3a709c04 INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065e342fd2397e294 INFO: Start service mk2-dbus.ttyUSB1
@4000000065e342e1397603ac *** CCGX booted (30253) ***
@4000000065e342e21a7137ec *** starting serial-starter ***
@4000000065e342e21a71572c serstart starting
@4000000065e342e21a7166cc INFO: loading config file /etc/venus/serial-starter.conf
@4000000065e342e31d4dde84 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000065e342e418b5d034 INFO: Create daemontools service dbus-serialbattery.ttyACM0
@4000000065e3f8323623469c INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@4000000065e3f83427c2b024 INFO: Create daemontools service mk2-dbus.ttyUSB1
@4000000065e3f83539efb404 INFO: Start service dbus-serialbattery.ttyACM0 once
@4000000065e3f8390b371224 INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065e3f83b017471b4 INFO: Start service mk2-dbus.ttyUSB1
@4000000065e56da425321264 INFO: Start service mk2-dbus.ttyUSB1
@4000000065e839e4377526b4 *** CCGX booted (0) ***
@4000000065e839e51e79a89c *** starting serial-starter ***
@4000000065e839e53a91c834 serstart starting
@4000000065e839e53b1e29b4 INFO: loading config file /etc/venus/serial-starter.conf
@4000000065e839e62a309b14 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000065e839e7316adb24 ERROR: no service template for dbus-serialbattery
@4000000065e839f922bc9dc4 INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@4000000065e839fb23d0d89c INFO: Create daemontools service mk2-dbus.ttyUSB1
@4000000065e839ff31ff27fc INFO: Create daemontools service gps-dbus.ttyACM0
@4000000065e839ff392c507c INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065e83a0130adefdc INFO: Start service mk2-dbus.ttyUSB1
@4000000065e83a0606da0d1c INFO: Start service gps-dbus.ttyACM0 once
@4000000065e83a0f2bb191dc INFO: Create daemontools service vedirect-interface.ttyACM0
@4000000065e83a1533b4138c INFO: Start service vedirect-interface.ttyACM0 once
@4000000065e83a170ee6c02c INFO: Create daemontools service dbus-serialbattery.ttyACM0
@4000000065e83a1d12302a84 INFO: Start service dbus-serialbattery.ttyACM0 once
@4000000065e844b137c1375c *** CCGX booted (0) ***
@4000000065e844b21d1c5ea4 *** starting serial-starter ***
@4000000065e844b304c50414 serstart starting
@4000000065e844b307b31f6c INFO: loading config file /etc/venus/serial-starter.conf
@4000000065e844b33ab2aa2c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000065e844b5025a592c INFO: Create daemontools service dbus-serialbattery.ttyACM0
@4000000065e844c61a3290dc INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@4000000065e844c808709754 INFO: Create daemontools service mk2-dbus.ttyUSB1
@4000000065e844ca052ca36c INFO: Start service dbus-serialbattery.ttyACM0 once
@4000000065e844cc228516ec INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065e844ce1f50a7ac INFO: Start service mk2-dbus.ttyUSB1

@4000000065e844df06d411dc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A (read from BMS)
@4000000065e844df06d438ec INFO:SerialBattery:> CVCM:     True
@4000000065e844df06d440bc INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@4000000065e844df06d45444 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000065e844df06d45ffc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000065e844df06d46f9c INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@4000000065e844df06d634bc INFO:SerialBattery:Serial Number/Unique Identifier: 56_105.0Ah
@4000000065e844df2cc12d1c INFO:SerialBattery:Found existing battery with DeviceInstance = 1
@4000000065e844e000bfdb3c INFO:SerialBattery:DeviceInstance = 1
@4000000065e844e000cbcdd4 INFO:SerialBattery:Used device instances: ['1']
@4000000065e844e000d6bab4 INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@4000000065e844e02313455c INFO:SerialBattery:publish config values = True
@4000000065e8d8ef153b2c4c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065e8e8682cdc469c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065e9b1842593b67c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065ea3cf92873ef24 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065eab8d4290b5cc4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065eac84c37a016e4 ERROR:SerialBattery:>>> ERROR: No reply - returning
```

mr-manuel, 2024-03-08:
Never heard of that. In the BMS the current is displayed correctly?

Kapitaen31, 2024-03-08:
i had check it now. but is the same..

mr-manuel, 2024-03-08:
Is the same as the driver shows or it‘s the same as it should be? Since the driver shows the same as the BMS os reporting you have to fix that in the BMS, since this will also reflect in wrong SOC and other values. If you don‘t know how, then contact your BMS seller.

Kapitaen31, 2024-03-08:
the same as the driver.
OK, I will then contact the seller.
Thank you.

closed 2024-03-08 by mr-manuel