mr-manuel/venus-os_dbus-serialbattery #26: Temperature sensors don’t show up correctly in Bluetooth driver

Pim57, 2024-04-04:
### Describe the bug

When using the Bluetooth interface with my JKbms BD6A20S15P the names of the temperature sensors don’t show correctly in Victron remote control.

I have one sensor named ‘Temp Fuse’ at the fuse holder and the second temperature sensor is named ‘Temp Batt’ placed on the battery cells.

[image]
[image]

### How to reproduce

To reproduce:
1. Have a JKbms BD6A20S15P
2. Have different names for the two temperature sensors in the config.ini
3. Make a serial connection to the Victron Cerbo GX 
4. Go to the remote control 
5. Go to the serial JKbms -> details page and look at the names of the temperature sensors
6. Repeat step 1-5 with Bluetooth interface.

### Expected behavior

I would expect the names for the temperature sensors in both the serial or Bluetooth interface to be similar.

### Driver version

V1.2.20240227beta

### Venus OS device type

Cerbo GX

### Venus OS version

V3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
Battery temperature
; Specify how the battery temperature is assembled
; 0 Get mean of temperature sensor 1 to sensor 4
; 1 Get only temperature from temperature sensor 1
; 2 Get only temperature from temperature sensor 2
; 3 Get only temperature from temperature sensor 3
; 4 Get only temperature from temperature sensor 4
TEMP_BATTERY = 1

; Temperature sensor 1 name
TEMP_1_NAME = Temp Batt

; Temperature sensor 2 name
TEMP_2_NAME = Temp Fuse
```

### Relevant log output

```shell
No relevant log output.

The system works fine, it’s just the naming of the temperature sensors in the remote control interface of the Cerbo GX.
```

mr-manuel, 2024-04-05:
Can you post a screenshot from `dbus-spy` where the temperature values are visible?

Pim57, 2024-04-05:
Here are screenshots of the temperature values 
[image]

[image]

Pim57, 2024-04-05:
I accidentally closed the issue. Now reopening

mr-manuel, 2024-04-05:
Did it work in a previous version of dbus-serialbattery?

mr-manuel, 2024-04-05:
Can you enable `DEBUG` and send the logs of the bluetooth connected BMS?

Pim57, 2024-04-05:
Here is the line from the log that shows temperature 1 and 2>>>>

2024-04-05 14:07:01.301724500 DEBUG:SerialBattery:{'model_nbr': 'BK-BLE-1.0', 'device_info': {'hw_rev': '11.XA', 'sw_rev': '11.41', 'uptime': 4918800, 'vendor_id': 'JK_BD6A20S15P', 'manufacturing_date': '240123', 'serial_number': '311153B317', 'production': 'Input Us'}, 'last_update': 1712326020.703545, 'settings': {'cell_uvp': 2.5500000000000003, 'cell_uvpr': 2.65, 'cell_ovp': 3.65, 'cell_ovpr': 3.5500000000000003, 'balance_trigger_voltage': 0.005, 'power_off_voltage': 2.5, 'max_charge_current': 65.0, 'max_discharge_current': 120.0, 'max_balance_current': 65.0, 'cell_count': 16, 'charging_switch': True, 'discharging_switch': True, 'balancing_switch': True}, 'cell_info': {'voltages': [3.341, 3.341, 3.341, 3.341, 3.341, 3.341, 3.341, 3.341, 3.341, 3.341, 3.34, 3.341, 3.341, 3.34, 3.341, 3.341], 'average_cell_voltage': 3.341, 'delta_cell_voltage': 0.001, 'max_voltage_cell': 0, 'min_voltage_cell': 9, 'resistances': [0.147, 0.147, 0.146, 0.147, 0.15, 0.152, 0.15, 0.152, 0.151, 0.149, 0.151, 0.151, 0.149, 0.146, 0.14400000000000002, 0.145, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'total_voltage': 53.461, 'current': 0.0, 'temperature_sensor_1': 26.900000000000002, 'temperature_sensor_2': 26.400000000000002, 'temperature_mos': 23.400000000000002, 'balancing_current': 0.0, 'balancing_action': 0.0, 'battery_soc': 73, 'capacity_remain': 219.673, 'capacity_nominal': 300.0, 'cycle_count': 15, 'cycle_capacity': 4505.1990000000005, 'charging_switch_enabled': True, 'discharging_switch_enabled': True, 'balancing_active': False, 'error_bitmask_16': '0x0', 'error_bitmask_2': '0000000000000000', 'power': 0.0}, 'warnings': {'resistance_too_high': False, 'cell_count_wrong': False, 'charge_overtemp': False, 'charge_undertemp': False, 'discharge_overtemp': False, 'cell_overvoltage': False, 'cell_undervoltage': False, 'charge_overcurrent': False, 'discharge_overcurrent': False}}

file is:
[ble_battery0_current.txt](https://github.com/mr-manuel/venus-os_dbus-serialbattery/files/14886092/ble_battery0_current.txt)

Pim57, 2024-04-05:
And this is all from the dbus-ble-sensors log >>>

2024-04-05 14:04:43.782631500 *** CCGX booted (0) ***
2024-04-05 14:05:05.200572500 *** starting dbus-ble-sensors ***
2024-04-05 14:05:06.401129500 dbus-ble-sensors 0.16
2024-04-05 14:05:06.758639500 opening hci1
2024-04-05 14:05:06.768011500 opening hci0

Pim57, 2024-04-05:
> Did it work in a previous version of dbus-serialbattery?

This is my first and only version I am using.

mr-manuel, 2024-04-28:
I found no hint why that happens. The strangest is, that it is seen from the Bluetooth part. Can you write me on Discord?

Pim57, 2024-04-28:
Yes, I’m on Discord username Pim57

mr-manuel, 2024-04-28:
Fixed problem with second temperature sensor, which was introduced with `v1.1.20240128dev` and commit https://github.com/mr-manuel/venus-os_dbus-serialbattery/commit/2da6bae211a171a1b76344b3ed3f4347469f2457.

commit 647f0558 (2024-04-28): Fixes #26 - JKBMS BLE 2nd temp sensor
https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/26
commit eceec5f4 (2024-04-28): Fixes #26 - JKBMS BLE 2nd temp sensor
https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/26
closed 2024-04-28 by mr-manuel