mr-manuel/venus-os_dbus-serialbattery #27: Zellspannungen bei zwei Akkupacks loggen

RRonGit, 2024-04-06:
### Describe the problem

Hallo Zusammen,

Ich habe folgende Umgebung:

Raspi4 mit VenusOS Verion v3.30
2x Seplos Batteriepacks BMS Seplos BMS V16 (10E) Software Version: 16.6
venus-os_dbus-serialbattery von mr-manuel installiert Version: 1.1.20240121
BatteryAggregator installiert via Setup Helper Version: v3.0.20

Die Seplos BMS sind beide separat per USB to RS485 an den Rapsi angeschlossen.

Das Ganze funktioniert soweit.

Gerne möchte ich aber die einzelnen Zellspannungen loggen.

Mit einem BMS geht das via Node-RED problemlos.
Bei zwei BMS werden die Werte immer abwechselnd von den beiden BMSen angezeigt.

[image]

[image]

Gibt es dazu ein Lösung?

### Driver version

1.1.20240121

### Venus OS device type

Raspberry Pi

### Venus OS version

3.30

### BMS type

Seplos

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; --------- Battery Current limits ---------

MAX_BATTERY_CHARGE_CURRENT = 140.0
MAX_BATTERY_DISCHARGE_CURRENT = 195.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
[... 162 lines cut ...]
2024-04-06 13:41:25.048175500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-04-06 13:41:25.052404500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-04-06 13:41:25.333721500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-04-06 13:41:25.335307500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2024-04-06 13:41:25.609043500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-04-06 13:41:25.610479500 INFO:SerialBattery:Testing Seplos
2024-04-06 13:41:26.729694500 INFO:SerialBattery:Connection established to Seplos
2024-04-06 13:41:26.730012500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB0
2024-04-06 13:41:26.730291500 INFO:SerialBattery:========== Settings ==========
2024-04-06 13:41:26.730615500 INFO:SerialBattery:> Connection voltage: 55.17V | Current: 0.0A | SoC: None%
2024-04-06 13:41:26.730929500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-04-06 13:41:26.731345500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-04-06 13:41:26.731701500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0A | MAX BATTERY DISCHARGE CURRENT: 195.0A
2024-04-06 13:41:26.732005500 INFO:SerialBattery:> CVCM:     True
2024-04-06 13:41:26.732320500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.062V | MAX CELL VOLTAGE: 3.531V
2024-04-06 13:41:26.732618500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-04-06 13:41:26.732924500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2024-04-06 13:41:26.733204500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-04-06 13:41:26.733557500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-04-06 13:41:26.932515500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-04-06 13:41:27.043422500 INFO:SerialBattery:DeviceInstance = 1
2024-04-06 13:41:27.043732500 INFO:SerialBattery:Used device instances: ['1']
2024-04-06 13:41:27.044031500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-04-06 13:41:27.060029500 INFO:SerialBattery:publish config values = True
```

mr-manuel, 2024-04-06:
Please install the latest dev and the problem should be shown. Probably your batteries have the same VRM ID.

RRonGit, 2024-04-07:
@mr-manuel, thank you very much, it worked. I have now installed version 1.2.20240404dev. Then I had to change the capacity of a battery from 280Ah to 281Ah so that the batteries got different IDs. Now I can log the two batteries separately.

closed 2024-04-07 by mr-manuel