mr-manuel/venus-os_dbus-serialbattery #29: SOC reset, cells voltage, balanced cells

Honusnap, 2024-04-22:
### Describe the bug

Hello All,

This is not per say a bug, maybe i'm doing something wrong.

My cells are well balanced and i do not charge them all the way up, i keep at 54.65V Bulk and 54.4 Float.
Problem is that the condition where SOC_RESET_VOLTAGE has to be higher then MAX_CELL_VOLTAGE do not work that well in those situation where there is near no deviation between cells... they all at the same voltage withtin 0.008V (at 54.6V total).
Which mean .. i'm kind locked between having a "High voltage" alarm .. ot not meeting the conditions on the SOC_RESET.

Why would there be a condition on the  SOC_RESET_VOLTAGE ? I want to be able to use a value lower then MAX_CELL_VOLTAGE.

Alarm high voltage when > MAX_CELL_VOLTAGEx16
=> No room here when there is not deviation.
3.416 : SOC_RESET_VOLTAGE 
3.415 ; MAX_CELL_VOLTAGE

Charge is not let to the BMS in my system, charge controller is doing the job, Chargers are set to 54.65V 
CVCM_ENABLE = False, is disabled : 

### How to reproduce

Charge pack up to 54.65V => SOC reset do not happend.

### Expected behavior

Reset SOC at the desired voltage.

### Driver version

v1.0.202404

### Venus OS device type

Cerbo GX

### Venus OS version

3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 120.0
[... 402 lines cut ...]
2024-04-22 08:25:44.152612500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20230331)
2024-04-22 08:25:44.154196500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-04-22 08:25:47.230669500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-04-22 08:25:47.753293500 INFO:SerialBattery:DeviceInstance = 1
2024-04-22 08:25:47.755405500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-04-22 08:25:47.756384500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-04-22 08:25:47.757569500 INFO:SerialBattery:com.victronenergy.battery.ble_c8478ce90ce6
2024-04-22 08:25:47.758934500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20230331)
2024-04-22 08:25:47.840511500 INFO:SerialBattery:publish config values = True
2024-04-22 08:25:47.881884500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c8478ce90ce6
2024-04-22 08:25:47.882567500 INFO:SerialBattery:========== Settings ==========
2024-04-22 08:25:47.883492500 INFO:SerialBattery:> Connection voltage: 54.62V | Current: 3.6A | SoC: 90.48%
2024-04-22 08:25:47.884304500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
2024-04-22 08:25:47.885108500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-04-22 08:25:47.885938500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.415V
2024-04-22 08:25:47.886778500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2024-04-22 08:25:47.887694500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 140.0A (read from BMS)
2024-04-22 08:25:47.888909500 INFO:SerialBattery:> CVCM:     True
2024-04-22 08:25:47.890060500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-04-22 08:25:47.891165500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-04-22 08:25:47.898761500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-04-22 08:25:47.901395500 INFO:SerialBattery:Serial Number/Unique Identifier: 2092416547
2024-04-22 08:25:48.037109500 INFO:SerialBattery:SOC initialized from dbus and set to 90.48%
2024-04-22 08:45:26.595390500 INFO:SerialBattery:JK BMS SOC reset finished.
```

mr-manuel, 2024-04-24:
See https://github.com/Louisvdw/dbus-serialbattery/issues/1018

Honusnap, 2024-04-24:
> See [Louisvdw#1018](https://github.com/Louisvdw/dbus-serialbattery/issues/1018)

Hello,
Thanks for the link but .. i can't see any solution here ... i'm using Calculated SOC, meaning it's not the BMS SOC that is now displayed but the calculated one from serialbattery. Why should we have  SOC_RESET_VOLTAGE has to be higher then MAX_CELL_VOLTAGE ? It generate high voltage error before reaching SOC_RESET_VOLTAGE... !
This is not a normal way, just let it be a arbitrary value ... i want to reset SOC at the first cell reaching 3.62V .. even if my MAX_CELL_VOLTAGE = 3.64V. It prevent the high voltage error... and still is vebry accurate since there is near no Wh between those 2 voltages.

mr-manuel, 2024-04-24:
I missunderstood something. Indeed, if you are using the calculated SOC you do not need the `SOC_RESET_VOLTAGE`.

I optimized the procedure in the latest nightly version, please install it and test it. Set the `MAX_CELL_VOLTAGE` to your desired value.

Honusnap, 2024-04-25:
> I missunderstood something. Indeed, if you are using the calculated SOC you do not need the `SOC_RESET_VOLTAGE`.
> 
> I optimized the procedure in the latest nightly version, please install it and test it. Set the `MAX_CELL_VOLTAGE` to your desired value.

Or maybe there is something i do not get ... 
**" if you are using the calculated SOC you do not need the SOC_RESET_VOLTAGE."**
I use calculated SOC yes, but SOC_RESET_VOLTAGE is the value that reset SOC to 100%, i still need it right ? I do not want to rely on Jkbms SOC since it reset only at OVP and i do not want to reach this value cause it shutdown charge "violently". Yes this "violence" can be reduced by selecting the right current value at high SOC cia config.ini. I think it's better to let things simple. Voltage reach 3.41 on one cell... SOC=100%, my cells are balanced within 0.008V at this voltage .. so .. all good .. :D

Thanks Manuel, will try your nightly.

mr-manuel, 2024-04-25:
`SOC_RESET_VOLTAGE` is only needed, if the BMS needs a specific cell voltage to reset the SOC back to 100%. For older JKBMS this is for example the OVP value.

If you are using the SOC calculation of the driver, then the SOC is reset when one cell reaches `MAX_CELL_VOLTAGE` and the current is below `SOC_RESET_CURRENT` for `SOC_RESET_TIME` seconds.

Honusnap, 2024-04-25:
> If you are using the SOC calculation of the driver, then the SOC is reset when one cell reaches `MAX_CELL_VOLTAGE` and the current is below `SOC_RESET_CURRENT` for `SOC_RESET_TIME` seconds.

OOoooo i did not know that .. ! I suppose i missed something in the doc/config.ini.

With the changes that you did in nightly the SOC_RESET_VOLTAGE is still not used in Driver SOC calculation mode .. right ? Or is it ? why not use this same SOC_RESET_VOLTAGE instead of MAX_CELL_VOLTAGE even in SOC calculation mode ?

In my case ..... since my cells are balanced... MAX_CELL_VOLTAGE on one cell is reached at the same time (or near) as High voltage alarm (not all the time but often) which make things not really confortable.
What do you think ?

Honusnap, 2024-04-26:
> > With the changes that you did in nightly the SOC_RESET_VOLTAGE is still not used in Driver SOC calculation mode .. right ?
> 
> Correct.
> 
> Other settings do not make sense. Incease OVP voltage or lower `MAX_CELL_VOLTAGE`.

OVP protection as nothing to do here... i never reach OVP.
**"then the SOC is reset when one cell reaches MAX_CELL_VOLTAGE and the current is below SOC_RESET_CURRENT for SOC_RESET_TIME seconds."**
Then i will have a "high voltage" alarm each time i want a SOC RESET.... cause my pack reach MAX_CELL_VOLTAGE**x16** at the same time one my cell reach MAX_CELL_VOLTAGE.

mr-manuel, 2024-04-26:
You can set `MAX_CELL_VOLTAGE` to whatever you want. If you have OVP set to 3.60 on the BMS then you can set `MAX_CELL_VOLTAGE` to 3.55 in the driver.

If it's still not clear please post some screenshots of the high voltage warning/error and all the settings of your BMS.

Honusnap, 2024-04-26:
> You can set `MAX_CELL_VOLTAGE` to whatever you want. If you have OVP set to 3.60 on the BMS then you can set `MAX_CELL_VOLTAGE` to 3.55 in the driver.
> 
> If it's still not clear please post some screenshots of the high voltage warning/error and all the settings of your BMS. If you are not reaching OVP then probably the alarm is not comming from the BMS.

..... i do not get it ..? You read what i've written ?
Changing MAX_CELL_VOLTAGE will have no impact on this problem.

mr-manuel, 2024-04-27:
Without screenshots I'm not able to help you.

1. Screenshot of the alarm in the Remote Console/GUI
2. Screenshots of the settings in the JKBMS app

Honusnap, 2024-04-27:
> Without screenshots I'm not able to help you. If the alarm is raised by the BMS it is triggered directly by the BMS, there are no settings for that in the driver.
> 
> 1. Screenshot of the alarm in the Remote Console/GUI
> 2. Screenshots of the settings in the JKBMS app

Alarm is not trigered by the BMS.
Can you explain what is :   * Sum of cell voltages >= self.max_battery_voltage - VOLTAGE_DROP_
IS self.max_battery_voltage = MAX_CELL_VOLTAGE * Cells Count (16 in most cases) ?

mr-manuel, 2024-04-27:
You don't have to always quote the text I wrote, if you directly reply to it.

I don't know where you got this code from. Since you are not answering to my questions and posting the requested screenshots, I think we are not getting to a point. I will reply again, if you posted the requested screenshots.

Honusnap, 2024-04-27:
This code ? It coems freom the config.ini ?
"Sum of cell voltages >= self.max_battery_voltage - VOLTAGE_DROP" <= Config.ini file

mr-manuel, 2024-04-27:
Check the latest update to the `config.default.ini`:

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/2624bd117902b21c12333a950263f413eb6fe1c4/etc/dbus-serialbattery/config.default.ini#L182-L197

Still waiting for the requested screenshots.

mr-manuel, 2024-04-27:
> Can you explain what is : * Sum of cell voltages >= self.max_battery_voltage - VOLTAGE_DROP_
> IS self.max_battery_voltage = MAX_CELL_VOLTAGE * Cells Count (16 in most cases) ?

Yes. See https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/2624bd117902b21c12333a950263f413eb6fe1c4/etc/dbus-serialbattery/battery.py#L414-L423

Honusnap, 2024-04-29:
Hello,
Allright, you are right, i was making things far more complex then they are .. ! I was stupid.
Problem is solved, no more alarm, reset happend perfectly where it should now, i've used the detailed log and from there i changed the MAX_CELL_VOLTAGE (decreased just by .. 0.05V) and reset now triger after the 60s under 3A as i choosed too. It's now nearly aligned with my juntek shunt. 
Perfect, thanks and sorry for makling things overly complex, i really though that this SOC_RESET_VOLTAGE  was used in this calculated SOC mode too.

Honusnap, 2024-04-29:
If you are in "SOC calculation" mode, the SOC_RESET_VOLTAGE is not used. Just set the MAX_CELL_VOLTAGE to the correct value, go a little ovber it with your Charge controller and set those settings to your needs : 

SOC_CALCULATION   = True
SOC_RESET_CURRENT = 3
SOC_RESET_TIME    = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS = -200, -100, -50, -25, -10, -5.9, -5, -3.4, -3.0, -2.4, -2.2, -1.4, -0.5, 0, 0.5, 1, 1.8, 2.2, 2.6, 2.8, 3, 3.35, 3.8, 4, 4.5, 5, 5
SOC_CALC_CURRENT_MEASURED_BY_USER = -200, -100, -50, -25, -10, -6.2, -5, -3.2, -2.70, -2.2, -2.1, -1, 0, 0, 0, 0.5, 1, 1.5, 2, 2.3, 2.55, 3, 3.55, 3.8, 4.35, 4.9, 5

Thanks again

closed 2024-04-29 by Honusnap