mr-manuel/venus-os_dbus-serialbattery #31: newest snapshot works only with LOGGING = DEBUG

stixif, 2024-05-30:
### Describe the bug

does not connect to venusOS

### How to reproduce

LOGGING = INFO

### Expected behavior

LOGGING = DEBUG

### Driver version

1.3.20240527dev

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.32

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

18

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
LOGGING = DEBUG
```

### Relevant log output

```shell
no...
```

mr-manuel, 2024-05-30:
I cannot reproduce that. How did you install/update?

mr-manuel, 2024-05-30:
Fixed with `v1.3.20240530dev`. Thank you for reporting!

closed 2024-05-30 by mr-manuel