mr-manuel/venus-os_dbus-serialbattery #35: Voltage High Alarm

dominikfe, 2024-06-19:
### Describe the problem

Venus OS shows a Voltage High alarm everyday, I guess caused by the SOC reset feature.
Is this normal behavior or do I have wrong settings applied?

### Driver version

1.2.20240401

### Venus OS device type

Raspberry Pi

### Venus OS version

3.20~50

### BMS type

JKBMS / Heltec BMS

### Cell count

18

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; MAX_CELL_VOLTAGE   = 3.55
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
; Description:
;     May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;     Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;       - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;     As you have to adopt this value to your system, I reccomend to start with
;     OVP voltage - 0.030 (see Example).
;       - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;         the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
[... 401 lines cut ...]
@40000000667030f72dd423bc INFO:SerialBattery:BAT: JKBMS 10.XW 18 cells (20230221)
@40000000667030f72de0d5bc INFO:SerialBattery:Connection established to Jkbms_Ble
@40000000667030f800225ed4 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
@40000000667030f807117c84 INFO:SerialBattery:DeviceInstance = 1
@40000000667030f807200efc INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
@40000000667030f80724645c INFO:SerialBattery:Used DeviceInstances = ['1']
@40000000667030f8072b2abc INFO:SerialBattery:com.victronenergy.battery.ble_c8478ce542f9
@40000000667030f807323f3c INFO:SerialBattery:BAT: JKBMS 10.XW 18 cells (20230221)
@40000000667030f8083d757c INFO:SerialBattery:publish config values = True
@40000000667030f8090dfc4c INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c8478ce542f9
@40000000667030f8091a6fcc INFO:SerialBattery:========== Settings ==========
@40000000667030f8091f2ea4 INFO:SerialBattery:> Connection voltage: 63.92V | Current: 0.4A | SoC: None%
@40000000667030f809232644 INFO:SerialBattery:> Cell count: 18 | Cells populated: 36
@40000000667030f8092721cc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000667030f8092b772c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@40000000667030f8092fb134 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A
@40000000667030f80933ef24 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0A | MAX BATTERY DISCHARGE CURRENT: 90.0A (read from BMS)
@40000000667030f80937db0c INFO:SerialBattery:> CVCM:     True
@40000000667030f8093bf9bc INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000667030f80940109c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000667030f8094413f4 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000667030f80947ec54 INFO:SerialBattery:Serial Number/Unique Identifier: 2052303064
@40000000667143ef189ae594 INFO:SerialBattery:JK BMS SOC reset finished.
@400000006672b38614458d64 INFO:SerialBattery:JK BMS SOC reset finished.
```

mr-manuel, 2024-06-19:
Please provide the logs in the correct format and also add a screenshot of the alarm in the Remote Console/GUI.

dominikfe, 2024-06-19:
[image]

mr-manuel, 2024-06-19:
Can you also post a screenshot of your BMS settings in the JK BMS?

dominikfe, 2024-06-19:
[image]
[image]
[image]

dominikfe, 2024-06-19:
Here you go 👍🏽

mr-manuel, 2024-06-19:
Hmm, maybe you have imbalanced cells? You can check in the VRM portal the min and max cell voltages. The settings seem fine. Else you have to check the voltages when the alarm gets triggered.

dominikfe, 2024-06-19:
Can I disable the SoC reset feature?

dominikfe, 2024-06-21:
Thank you! Just for clarification:
My understanding is, that the SoC reset with JK BMS is only possible, if one of the cells runs into Cell OVP that was set on the BMS directly (in my case 3.580V). To achieve that dbus-serialbattery has to set the charging voltage high enough to let this happen. This will lead to a cell voltage higher than set in config.ini (in my case MAX_CELL_VOLTAGE   = 3.450). 
The cell voltage that SoC reset feature aims to reach in my case is SOC_RESET_VOLTAGE = 3.45.
So the triggered high voltage alarm comes from dbus-serialbattery as the cell voltage exceeded my MAX_CELL_VOLTAGE   = 3.450, right?

Graph of cell voltages:
[image]

mr-manuel, 2024-06-21:
Not with JKBMS connected over BLE or Daly. There the SOC is reset when switching to float.

dominikfe, 2024-06-21:
Ah, this is interesting! Thank you, I will review it from that view angle again.
Does this mean that with serial connected JK BMS you change the Cell OVP on the BMS to achieve the 100% SoC?

mr-manuel, 2024-06-21:
Yes, that is correct.

Check the feature page and the comparison table: https://louisvdw.github.io/dbus-serialbattery/general/features#soc-reset-voltage

closed 2024-06-21 by mr-manuel

dominikfe, 2024-06-21:
Thank you, great support!