mr-manuel/venus-os_dbus-serialbattery #36: Charge Current is limited on the wrong temperature sensor

Pim57, 2024-06-22:
### Describe the bug

On my JK bms there are 2 temperature sensors. One sits on the battery cells and the other sits on de battery fuse.
In issue #26 the problem is fixed that the wrong sensor showed up in the VRM dashboard. (Temp Fuse instead of Temp Batt)
However the calculation for the CCL is still on the wrong sensor. (Temp Fuse instead of Temp Batt)

### How to reproduce

1. Have a JKbms (ie. BD6A20S15P)
2. Have different names for the two temperature sensors in the config.ini
3. Make a blutooth connection to the Victron Cerbo GX
4. Set steps to reduce battery current for temperature as in config.default.ini
5. Charge the battery and watch Temp Fuse rise to go over 35 degrees while Temp Batt is below 35 (22 in my case)
6. Check in VRM Remote Control that CCL goes from max charge current to a Temperature limited current

### Expected behavior

The CCL should be calculated with the value from the temperature sensor as specified in config.ini

TEMP_BATTERY = 2

; Temperature sensor 1 name
TEMP_1_NAME = Temp Fuse

; Temperature sensor 2 name
TEMP_2_NAME = Temp Batt

### Driver version

V1.2.20240227beta

### Venus OS device type

Cerbo GX

### Venus OS version

V3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 65.0
MAX_BATTERY_DISCHARGE_CURRENT = 120.0
[... 262 lines cut ...]
; Temperature sensor 2 name
TEMP_2_NAME = Temp Batt

; --------- Voltage drop ---------
; If you have a voltage drop between the BMS and the charger because of wire size or length
; then you can specify the voltage drop here. The driver will then add the voltage drop
; to the calculated CVL to compensate.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 V and the battery is now charged until the charger reaches 55.20 V.
;     The BMS now measures 55.05 V since there is a voltage drop of 0.15 V on the cable.
;     Since the dbus-serialbattery reads the voltage of 55.05 V from the BMS the max voltage
;     of 55.20 V is never reached and max voltage is kept forever.
;     By setting the VOLTAGE_DROP to 0.15 V the voltage on the charger is increased and the
;     target voltage on the BMS is reached.
VOLTAGE_DROP = 0.10
```

### Relevant log output

```shell
none
```

mr-manuel, 2024-06-22:
This is no bug, it's a feature :-) This beahaviour is wanted, since you want also to reduce the CCL and DCL if the fuse gets too hot. Normally the battery temperature is used when it's going to be cold and the fuse temperature when it's going to be hot. Under normal conditions the battery will always be cooler than the fuse on big currents.

Here an example:

[image]

Pim57, 2024-06-22:
Okay, then better update the documentation and default.config.ini to say that CCL/DCL will be calculated on the hottest temperature sensor.Thanks and regards, Pim Snoeks

grafik.png (view on web)

mr-manuel, 2024-06-22:
That statement would be wrong, but I got what you like to say. I will update it.

closed 2024-06-22 by mr-manuel