mr-manuel/venus-os_dbus-serialbattery #37: #8 internal calculation error

ckvsoft, 2024-06-27:
### Describe the bug

#8 internal calculation error

go back to 1.2.20240401. This works without this error

### How to reproduce

Let it run for some time and then the error will appear on its own

### Expected behavior

??

### Driver version

v1.3.20240624

### Venus OS device type

Raspberry Pi 4

### Venus OS version

3.33

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
default
```

### Relevant log output

```shell
@40000000667aa18723684714 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000667aa187237779b4 WARNING:SerialBattery:Error while executing calcMaxChargeCurrentReferringToCellVoltage(). Using default value instead.
@40000000667aa187237ca9d4 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_CHARGING: [3.55, 3.5, 3.45, 3.3] • MAX_CHARGE_CURRENT_CV: [0.0, 2.5, 25.0, 50.0]
@40000000667aa18723817464 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1100
@40000000667aa187238a230c WARNING:SerialBattery:Error while executing calcMaxDischargeCurrentReferringToCellVoltage(). Using default value instead.
@40000000667aa187238ee1e4 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_DISCHARGING: [2.6, 2.8, 2.9, 3.1] • MAX_DISCHARGE_CURRENT_CV: [0.0, 8.0, 40.0, 80.0]
@40000000667aa18723936624 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1140
@40000000667aa187255b1664 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1061
@40000000667aa1882371055c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000667aa188237f41e4 WARNING:SerialBattery:Error while executing calcMaxChargeCurrentReferringToCellVoltage(). Using default value instead.
@40000000667aa18823845e7c WARNING:SerialBattery:CELL_VOLTAGES_WHILE_CHARGING: [3.55, 3.5, 3.45, 3.3] • MAX_CHARGE_CURRENT_CV: [0.0, 2.5, 25.0, 50.0]
@40000000667aa1882389196c ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1100
@40000000667aa188239179f4 WARNING:SerialBattery:Error while executing calcMaxDischargeCurrentReferringToCellVoltage(). Using default value instead.
@40000000667aa18823964484 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_DISCHARGING: [2.6, 2.8, 2.9, 3.1] • MAX_DISCHARGE_CURRENT_CV: [0.0, 8.0, 40.0, 80.0]
@40000000667aa188239ac8c4 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1140
@40000000667aa18823be98bc ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1061
```

mr-manuel, 2024-06-27:
Please provide the logs in the correct format like everywhere described.

ckvsoft, 2024-06-27:
hash 8 is not supposed to be a reference, the error message I get is
# 8 internal calculation error

ckvsoft, 2024-06-27:
> Please provide the logs in the correct format like everywhere described.

?? Where ist the description?

ckvsoft, 2024-06-27:
It seems as if the error also occurs in version 1.2. I skipped 1.2 during the update.
I've now gone back to 1.1.

mr-manuel, 2024-06-27:
When you open an issue:

[image]

If you click on the here link:

[image]

[image]

From where did you copy the command to get the logs?

ckvsoft, 2024-06-27:
I didn't copy any commands from anywhere.
It says you should post the relevant parts if you want to report a new bug.

They are simply the relevant parts from the logfile /data/log/dbus-serialbattery.ttyUSB0/current.

Hope this version is now as it should be

2024-06-25 11:26:50.048507500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A
2024-06-25 11:26:50.048918500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 55.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A (read from BMS)
2024-06-25 11:26:50.049283500 INFO:SerialBattery:> CVCM:     True
2024-06-25 11:26:50.049669500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.6V | MAX CELL VOLTAGE: 3.45V
2024-06-25 11:26:50.050110500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-06-25 11:26:50.050481500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-06-25 11:26:50.050851500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-06-25 11:26:50.051207500 INFO:SerialBattery:Serial Number/Unique Identifier: JKBMS1
2024-06-25 11:26:50.195410500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-06-25 11:26:50.309649500 INFO:SerialBattery:DeviceInstance = 1
2024-06-25 11:26:50.309954500 INFO:SerialBattery:Used device instances: ['1']
2024-06-25 11:26:50.310251500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-06-25 11:26:50.327686500 INFO:SerialBattery:publish config values = True
2024-06-25 12:52:11.340132500 INFO:SerialBattery:
2024-06-25 12:52:11.340429500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-25 12:52:11.341546500 INFO:SerialBattery:Venus OS v3.33
2024-06-25 12:52:11.341864500 INFO:SerialBattery:dbus-serialbattery v1.3.20240624
2024-06-25 12:52:27.350638500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-25 12:52:27.350941500 INFO:SerialBattery:Testing Jkbms
2024-06-25 12:52:43.310608500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:163/39169]
2024-06-25 12:52:43.817591500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-06-25 12:52:43.817971500 INFO:SerialBattery:Testing Jkbms
2024-06-25 12:52:43.969512500 INFO:SerialBattery:Connection established to Jkbms
2024-06-25 12:52:44.151765500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-06-25 12:52:44.302533500 INFO:SerialBattery:DeviceInstance = 1
2024-06-25 12:52:44.303388500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-06-25 12:52:44.304039500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-06-25 12:52:44.304901500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-06-25 12:52:44.322190500 INFO:SerialBattery:publish config values = False
2024-06-25 12:52:44.323335500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2024-06-25 12:52:44.323727500 INFO:SerialBattery:========== Settings ==========
2024-06-25 12:52:44.324148500 INFO:SerialBattery:> Connection voltage: 53.79 V | Current: 13.58 A | SoC: 82%
2024-06-25 12:52:44.324517500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-06-25 12:52:44.324883500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-06-25 12:52:44.325284500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.600 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.400 V
2024-06-25 12:52:44.325730500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A
2024-06-25 12:52:44.326119500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 55.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A (read from BMS)
2024-06-25 12:52:44.326490500 INFO:SerialBattery:> CVCM:     True
2024-06-25 12:52:44.326865500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-06-25 12:52:44.327237500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-06-25 12:52:44.327608500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-06-25 12:52:44.327978500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-06-25 12:52:44.328339500 INFO:SerialBattery:Serial Number/Unique Identifier: JKBMS1
2024-06-25 12:52:45.594036500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-06-25 12:52:45.595032500 WARNING:SerialBattery:Error while executing calcMaxChargeCurrentReferringToCellVoltage(). Using default value instead.
2024-06-25 12:52:45.595372500 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_CHARGING: [3.55, 3.5, 3.45, 3.3] • MAX_CHARGE_CURRENT_CV: [0.0, 2.5, 25.0, 50.0]
2024-06-25 12:52:45.595686500 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1100
2024-06-25 12:52:45.596255500 WARNING:SerialBattery:Error while executing calcMaxDischargeCurrentReferringToCellVoltage(). Using default value instead.
2024-06-25 12:52:45.596566500 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_DISCHARGING: [2.6, 2.8, 2.9, 3.1] • MAX_DISCHARGE_CURRENT_CV: [0.0, 8.0, 40.0, 80.0]
2024-06-25 12:52:45.596862500 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1140
2024-06-25 12:52:45.626726500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1061
[... cut ...]
2024-06-27 09:11:59.351550500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 55.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A (read from BMS)
2024-06-27 09:11:59.351809500 INFO:SerialBattery:> CVCM:     True
2024-06-27 09:11:59.352076500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-06-27 09:11:59.352339500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-06-27 09:11:59.352600500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-06-27 09:11:59.352856500 INFO:SerialBattery:Serial Number/Unique Identifier: JKBMS1
2024-06-27 09:12:34.871067500 *** CCGX booted (0) ***

mr-manuel, 2024-06-27:
It seems that your cell voltages are empty sometimes. This error logging was added some time ago, where before it wasn't shown.

I made some changes to the code. Please install the latest nightly.

ckvsoft, 2024-06-27:
I did it. I'll watch that.
Strange that I don't have such messages with 1.1

ckvsoft, 2024-06-27:
1.4 doesn't work at all. At the beginning of the log file you can still see the start of 1.1. It works

2024-06-27 13:58:04.380636500 INFO:SerialBattery:
2024-06-27 13:58:04.380855500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 13:58:04.381448500 INFO:SerialBattery:dbus-serialbattery v1.1.20240121
2024-06-27 13:58:20.391414500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-27 13:58:20.392088500 INFO:SerialBattery:Testing Jkbms
2024-06-27 13:58:20.540680500 INFO:SerialBattery:Connection established to Jkbms
2024-06-27 13:58:20.541455500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2024-06-27 13:58:20.542034500 INFO:SerialBattery:========== Settings ==========
2024-06-27 13:58:20.542466500 INFO:SerialBattery:> Connection voltage: 52.95V | Current: -11.67A | SoC: None%
2024-06-27 13:58:20.542847500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-06-27 13:58:20.543229500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-06-27 13:58:20.543617500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A
2024-06-27 13:58:20.544008500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 55.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A (read from BMS)
2024-06-27 13:58:20.544443500 INFO:SerialBattery:> CVCM:     True
2024-06-27 13:58:20.544855500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.6V | MAX CELL VOLTAGE: 3.45V
2024-06-27 13:58:20.545237500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-06-27 13:58:20.545696500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-06-27 13:58:20.546073500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-06-27 13:58:20.546439500 INFO:SerialBattery:Serial Number/Unique Identifier: JKBMS1
2024-06-27 13:58:20.708646500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-06-27 13:58:20.824850500 INFO:SerialBattery:DeviceInstance = 1
2024-06-27 13:58:20.825164500 INFO:SerialBattery:Used device instances: ['1']
2024-06-27 13:58:20.825466500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-06-27 13:58:20.841400500 INFO:SerialBattery:publish config values = True
2024-06-27 14:22:09.682853500 INFO:SerialBattery:
2024-06-27 14:22:09.683071500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 14:22:09.684012500 INFO:SerialBattery:Venus OS v3.33
2024-06-27 14:22:09.684286500 INFO:SerialBattery:dbus-serialbattery v1.4.20240627dev
2024-06-27 14:22:25.696710500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-27 14:22:25.696716500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:22:41.650174500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:163/39169]
2024-06-27 14:22:42.151472500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-06-27 14:22:42.151753500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:22:42.424145500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-06-27 14:22:42.925412500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-06-27 14:22:42.925700500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:22:43.197147500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-06-27 14:22:43.698400500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-06-27 14:22:44.585609500 INFO:SerialBattery:
2024-06-27 14:22:44.586022500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 14:22:44.588651500 INFO:SerialBattery:Venus OS v3.33
2024-06-27 14:22:44.589066500 INFO:SerialBattery:dbus-serialbattery v1.4.20240627dev
2024-06-27 14:23:00.598350500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-27 14:23:00.598714500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:23:16.549593500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:99/177]
2024-06-27 14:23:17.051315500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-06-27 14:23:17.051321500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:23:17.328388500 ERROR:SerialBattery:>>> ERROR: No reply - returning
tail: /data/log/dbus-serialbattery.ttyUSB0/current has been replaced; following end of new file
2024-06-27 14:23:18.598320500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-06-27 14:23:24.949182500 INFO:SerialBattery:
2024-06-27 14:23:24.949464500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 14:23:24.950441500 INFO:SerialBattery:Venus OS v3.33
2024-06-27 14:23:24.950713500 INFO:SerialBattery:dbus-serialbattery v1.4.20240627dev
2024-06-27 14:23:40.967293500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-27 14:23:40.967564500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:23:56.925062500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:99/177]
2024-06-27 14:23:57.426453500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-06-27 14:23:57.426744500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:23:57.694904500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-06-27 14:23:58.196145500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-06-27 14:23:58.196424500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:23:58.465459500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-06-27 14:23:58.970589500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-06-27 14:24:04.290741500 INFO:SerialBattery:
2024-06-27 14:24:04.291014500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 14:24:04.291992500 INFO:SerialBattery:Venus OS v3.33
2024-06-27 14:24:04.292269500 INFO:SerialBattery:dbus-serialbattery v1.4.20240627dev
[... cut ...]
2024-06-27 14:25:17.624524500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-06-27 14:25:22.927488500 INFO:SerialBattery:
2024-06-27 14:25:22.927775500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 14:25:22.928728500 INFO:SerialBattery:Venus OS v3.33
2024-06-27 14:25:22.929005500 INFO:SerialBattery:dbus-serialbattery v1.4.20240627dev
2024-06-27 14:25:38.945591500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-27 14:25:38.945879500 INFO:SerialBattery:Testing Jkbms

ckvsoft, 2024-06-27:
her the log with 1.1

2024-06-27 14:28:34.788497500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-27 14:28:34.789295500 INFO:SerialBattery:dbus-serialbattery v1.1.20240121
2024-06-27 14:28:50.806391500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-27 14:28:50.806396500 INFO:SerialBattery:Testing Jkbms
2024-06-27 14:28:50.954837500 INFO:SerialBattery:Connection established to Jkbms
2024-06-27 14:28:50.955163500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2024-06-27 14:28:50.955438500 INFO:SerialBattery:========== Settings ==========
2024-06-27 14:28:50.955763500 INFO:SerialBattery:> Connection voltage: 53.43V | Current: -0.0A | SoC: None%
2024-06-27 14:28:50.956053500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-06-27 14:28:50.956347500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-06-27 14:28:50.956642500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A
2024-06-27 14:28:50.957513500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 55.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A (read from BMS)
2024-06-27 14:28:50.957519500 INFO:SerialBattery:> CVCM:     True
2024-06-27 14:28:50.957522500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.6V | MAX CELL VOLTAGE: 3.45V
2024-06-27 14:28:50.957805500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-06-27 14:28:50.958113500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-06-27 14:28:50.958406500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-06-27 14:28:50.958696500 INFO:SerialBattery:Serial Number/Unique Identifier: JKBMS1
2024-06-27 14:28:51.105863500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-06-27 14:28:51.222475500 INFO:SerialBattery:DeviceInstance = 1
2024-06-27 14:28:51.223182500 INFO:SerialBattery:Used device instances: ['1']
2024-06-27 14:28:51.223602500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-06-27 14:28:51.239922500 INFO:SerialBattery:publish config values = True

mr-manuel, 2024-06-27:
Can you write me on Discord?

mr-manuel, 2024-06-27:
You tried a reboot after install, right? I would have to connect to your system, since I do not find anything that changed and can cause this.

I have access to multiple systems with different JKBMS hardware versions, but on all it is working fine.

ckvsoft, 2024-06-27:
> Can you write me on Discord?

Yes, if I don't have to set up an account somewhere else then I'm happy to

ckvsoft, 2024-06-27:
> You tried a reboot after install, right? I would have to connect to your system, since I do not find anything that changed and can cause this.
> 
> I have access to multiple systems with different JKBMS hardware versions, but on all it is working fine.

Yes. first without reboot then it reboot.
It's peculiar. but 1.1 is running flawlessly again so far

mr-manuel, 2024-06-27:
You only need Discord, else it will be very difficult to help you.

ckvsoft, 2024-06-28:
No problem. 1.1 has been running without problems since yesterday. At the moment I don't see the advantage of the 1.3 or 1.4 allos, so I'll leave it like that. I just wanted to report the error, if no one else has the error, that's fine, thanks

jjdejong, 2024-06-30:
I have the exact same error messages since I switched to 1.3 with JKBMS. With 1.2, the instance on ttyUSB0 is very unstable and regularly disconnects. The disconnections seem in sync with the errors, only they weren't reported in 1.2, it seems. No issue over Bluetooth, but I'd like to replace that with the RS485 link, which, I understand, should be faster and more reliable... if it wasn't for the errors.

mr-manuel, 2024-07-01:
1.3 shows you in the gui, if in the log are errors present. The versions before did not. Check your logs to find out whats going wrong.

mr-manuel, 2024-07-01:
It seems that you have multiple issues here.

1. A bug in the script
```
2024-07-01 02:49:53.061124500 WARNING:SerialBattery:Error while executing calcMaxChargeCurrentReferringToCellVoltage(). Using default value instead.
2024-07-01 02:49:53.062276500 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_CHARGING: [3.65, 3.6, 3.55, 3.4] • MAX_CHARGE_CURRENT_CV: [0.0, 2.5, 25.0, 50.0]
2024-07-01 02:49:53.063268500 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1100
2024-07-01 02:49:53.065129500 WARNING:SerialBattery:Error while executing calcMaxDischargeCurrentReferringToCellVoltage(). Using default value instead.
2024-07-01 02:49:53.066259500 WARNING:SerialBattery:CELL_VOLTAGES_WHILE_DISCHARGING: [2.6, 2.7, 2.8, 3.0] • MAX_DISCHARGE_CURRENT_CV: [0.0, 6.0, 30.0, 60.0]
2024-07-01 02:49:53.067317500 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #1140
2024-07-01 02:49:53.180107500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1061
```

2. Seems that multiple devices try to access the battery and then the adapter crashed.

The latest nightly should solve that. Can you install that please?

jjdejong, 2024-07-01:
OK, installed, let you know.

jjdejong, 2024-07-01:
An error occurred, but the driver reconnected quickly:

```
2024-07-01 08:36:23.716585500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-01 08:36:23.720232500 INFO:SerialBattery:Venus OS v3.33
2024-07-01 08:36:23.720702500 INFO:SerialBattery:dbus-serialbattery v1.4.20240629dev
2024-07-01 08:36:39.734275500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-07-01 08:36:39.734854500 INFO:SerialBattery:Testing Jkbms
2024-07-01 08:36:39.880461500 INFO:SerialBattery:Connection established to Jkbms
2024-07-01 08:36:40.341448500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-07-01 08:36:40.590763500 INFO:SerialBattery:DeviceInstance = 2
2024-07-01 08:36:40.592188500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-07-01 08:36:40.592656500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-07-01 08:36:40.593106500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-07-01 08:36:40.618472500 INFO:SerialBattery:publish config values = False
2024-07-01 08:36:40.620348500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-07-01 08:36:40.620856500 INFO:SerialBattery:========== Settings ==========
2024-07-01 08:36:40.621396500 INFO:SerialBattery:> Connection voltage: 50.03 V | Current: -0.0 A | SoC: 7% | SoC calc: 1%
2024-07-01 08:36:40.621780500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-07-01 08:36:40.622155500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-07-01 08:36:40.622586500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-07-01 08:36:40.623061500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-07-01 08:36:40.623486500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A (read from BMS)
2024-07-01 08:36:40.623856500 INFO:SerialBattery:> CVCM:     False
2024-07-01 08:36:40.624231500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-01 08:36:40.624628500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-01 08:36:40.625120500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-01 08:36:40.625516500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-01 08:36:40.625952500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_BD6A20S10
2024-07-01 08:56:58.497537500 ERROR:SerialBattery:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-07-01 09:00:39.759965500 Traceback (most recent call last):
2024-07-01 09:00:39.759979500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 782, in publish_battery
2024-07-01 09:00:39.759983500     result = self.battery.refresh_data()
2024-07-01 09:00:39.759985500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 69, in refresh_data
2024-07-01 09:00:39.759989500     result = self.read_status_data()
2024-07-01 09:00:39.759991500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 82, in read_status_data
2024-07-01 09:00:39.760097500     status_data = self.read_serial_data_jkbms(self.command_status)
2024-07-01 09:00:39.760101500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 343, in read_serial_data_jkbms
2024-07-01 09:00:39.760105500     data = read_serial_data(
2024-07-01 09:00:39.760107500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 501, in read_serial_data
2024-07-01 09:00:39.760110500     with serial.Serial(port, baudrate=baud, timeout=0.1) as ser:
2024-07-01 09:00:39.760133500   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
2024-07-01 09:00:39.760137500     self.open()
2024-07-01 09:00:39.760139500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 288, in open
2024-07-01 09:00:39.760142500     self._update_rts_state()
2024-07-01 09:00:39.760144500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 627, in _update_rts_state
2024-07-01 09:00:39.760274500     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
2024-07-01 09:00:39.760276500 OSError: [Errno 62] Timer expired
2024-07-01 09:00:39.937504500 /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (17382) - No such process
2024-07-01 09:01:02.738855500 INFO:SerialBattery:
2024-07-01 09:01:02.739378500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-01 09:01:02.741166500 INFO:SerialBattery:Venus OS v3.33
2024-07-01 09:01:02.741778500 INFO:SerialBattery:dbus-serialbattery v1.4.20240629dev
2024-07-01 09:01:18.758806500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-07-01 09:01:18.759642500 INFO:SerialBattery:Testing Jkbms
2024-07-01 09:01:18.916635500 INFO:SerialBattery:Connection established to Jkbms
2024-07-01 09:01:19.388134500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-07-01 09:01:19.619957500 INFO:SerialBattery:DeviceInstance = 2
2024-07-01 09:01:19.621413500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-07-01 09:01:19.621943500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-07-01 09:01:19.622409500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-07-01 09:01:19.647343500 INFO:SerialBattery:publish config values = False
2024-07-01 09:01:19.649178500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
[... cut ...]
2024-07-01 09:01:19.654186500 INFO:SerialBattery:> CVCM:     False
2024-07-01 09:01:19.654714500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-01 09:01:19.655300500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-01 09:01:19.655807500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-01 09:01:19.656256500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-01 09:01:19.656723500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_BD6A20S10
```

jjdejong, 2024-07-01:
OK, now I'm experiencing regular disconnections with the following messages repeated each time:

```
2024-07-01 09:49:24.186619500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-01 09:49:24.189046500 INFO:SerialBattery:Venus OS v3.33
2024-07-01 09:49:24.189866500 INFO:SerialBattery:dbus-serialbattery v1.4.20240629dev
2024-07-01 09:49:40.202053500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-07-01 09:49:40.202573500 INFO:SerialBattery:Testing Jkbms
2024-07-01 09:49:40.313803500 ERROR:SerialBattery:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-07-01 09:49:40.820083500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-07-01 09:49:40.820670500 INFO:SerialBattery:Testing Jkbms
2024-07-01 09:49:40.822123500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-07-01 09:49:41.323695500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-07-01 09:49:41.324378500 INFO:SerialBattery:Testing Jkbms
2024-07-01 09:49:41.480692500 INFO:SerialBattery:Connection established to Jkbms
2024-07-01 09:49:42.107273500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-07-01 09:49:42.363379500 INFO:SerialBattery:DeviceInstance = 2
2024-07-01 09:49:42.366970500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-07-01 09:49:42.368780500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-07-01 09:49:42.370479500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-07-01 09:49:42.410321500 INFO:SerialBattery:publish config values = False
2024-07-01 09:49:42.412104500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-07-01 09:49:42.412567500 INFO:SerialBattery:========== Settings ==========
2024-07-01 09:49:42.413105500 INFO:SerialBattery:> Connection voltage: 48.37 V | Current: 9.29 A | SoC: 6% | SoC calc: 1%
2024-07-01 09:49:42.413497500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-07-01 09:49:42.413874500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-07-01 09:49:42.414284500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-07-01 09:49:42.414738500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-07-01 09:49:42.415235500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A (read from BMS)
2024-07-01 09:49:42.415734500 INFO:SerialBattery:> CVCM:     False
2024-07-01 09:49:42.416194500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-01 09:49:42.416590500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-01 09:49:42.416963500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-01 09:49:42.417337500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-01 09:49:42.417719500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_BD6A20S10
2024-07-01 09:49:45.535479500 ERROR:SerialBattery:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-07-01 09:49:47.533074500 ERROR:SerialBattery:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-07-01 09:49:50.513870500 ERROR:SerialBattery:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-07-01 09:49:52.548947500 Traceback (most recent call last):
2024-07-01 09:49:52.548961500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 782, in publish_battery
2024-07-01 09:49:52.548971500     result = self.battery.refresh_data()
2024-07-01 09:49:52.548974500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 69, in refresh_data
2024-07-01 09:49:52.548977500     result = self.read_status_data()
2024-07-01 09:49:52.548979500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 82, in read_status_data
2024-07-01 09:49:52.549159500     status_data = self.read_serial_data_jkbms(self.command_status)
2024-07-01 09:49:52.549163500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 343, in read_serial_data_jkbms
2024-07-01 09:49:52.549167500     data = read_serial_data(
2024-07-01 09:49:52.549169500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 502, in read_serial_data
2024-07-01 09:49:52.549173500     return read_serialport_data(
2024-07-01 09:49:52.549200500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 553, in read_serialport_data
2024-07-01 09:49:52.549204500     toread = ser.inWaiting()
2024-07-01 09:49:52.549206500   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 590, in inWaiting
2024-07-01 09:49:52.549209500     return self.in_waiting
2024-07-01 09:49:52.549211500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 467, in in_waiting
2024-07-01 09:49:52.549214500     s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
2024-07-01 09:49:52.549277500 OSError: [Errno 5] Input/output error
2024-07-01 09:49:52.737356500 /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (22972) - No such process
2024-07-01 09:50:05.468319500 INFO:SerialBattery:
2024-07-01 09:50:05.469520500 INFO:SerialBattery:Starting dbus-serialbattery
[... cut ...]
2024-07-01 09:50:22.573359500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-07-01 09:50:22.573858500 INFO:SerialBattery:Testing Jkbms
2024-07-01 09:50:22.641035500 ERROR:SerialBattery:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-07-01 09:50:23.147260500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-07-01 09:50:23.295702500 /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (23511) - No such process
```

mr-manuel, 2024-07-01:
I would like to troubleshoot this directly on your system. Can you write me on Discord?

mr-manuel, 2024-07-02:
I tried it with different versions on your system, but you always have the same issue with the USB to serial adapter not responding correctly. Unfortunately I have no idea for that.

closed 2024-07-02 by mr-manuel