mr-manuel/venus-os_dbus-serialbattery #40: System overload

josmeijer, 2024-06-29:
### Describe the problem

I have installed serialbattery on a COLOR CONTROL GX runnin 3.33 firmware. It seems to work. (though victron support generated an error report with some glitches and I sometimes get a "no real time data for system overload"warning in VRM) but as soon as I trie to switch on/off the new Dynamic ESS I get an error message (2 way communication is off, though it's on) and can't do the trick. If I manually switch off Serialbattery throug SSH it does work.

I tried to install the latest version and though a little improvement is to be seen in processor load (see both logs attached) I still can't use Dynamic ESS together with serialbattery.

### Driver version of the currently installed driver

20230407 serial 1.3.20240624 firmware

### Venus OS device type

CCGX

### Venus OS version

3.33

### BMS type

Daly BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60
INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
/data/log/serial-starter/current
@40000000667ea7b001ef75e4 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea7b5262fa08c INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea7bf191bfc74 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea7c5116b2cd4 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea7cb00dbcb94 INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea7d4290f601c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea7da1548b524 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea7e00600377c INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea7e90843805c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea7f0025bf73c INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea7f53155342c INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea7ff336186bc INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea8052cebf63c INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea80b1cb7dc2c INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea8151d68d0f4 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea81b0e063f84 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea821100b1fd4 INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea82b1b8b818c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea83112e0a24c INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea8370844e3d4 INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea841065dd06c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea8463981e2bc INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea84c2f6798e4 INFO: Start service gps-dbus.ttyUSB0 once
@40000000667ea8562f028db4 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000667ea85c24d3c5c4 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000667ea8621e43ac4c INFO: Start service gps-dbus.ttyUSB0 once

/data/log/serial-starter/current
  GNU nano 4.9.3                                                /data/log/dbus-serialbattery.ttyUSB0/current                                                           @40000000667fbf3a057bc53c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fbf5c163b1fbc INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc0722846053c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc07727bef704 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc07928e6350c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc0892b14ffd4 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc0be3a2ab964 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc0dc28aa9754 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc12a274a8e3c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc1611128dc44 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc1c82f58819c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc1cc32fb5e64 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc1f01b8f80fc INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.3s
@40000000667fc231165696fc INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc247044c9fc4 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc27e145b057c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc28a365e6b3c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc29d03e97124 INFO:SerialBattery:refresh_data: result: False
@40000000667fc2c724231b34 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc31c0fde3ba4 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc324114ad77c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc35e291e15bc INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc36d3ab57cd4 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc3ed25a6fffc INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc414016bf5d4 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
@40000000667fc48a2409076c INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
```

### Any other information that may be helpful

https://docs.google.com/spreadsheets/d/1OPBUOT0t29FpSqNEytqgXXRI--RenBbPmFKvFKtE8Fg/edit?gid=1876670686#gid=1876670686

https://docs.google.com/spreadsheets/d/1kLEyEji3LZW806jLrBGRFzfXgG8Eaq-2pnAlvrKNUNg/edit?gid=1803095261#gid=1803095261

mr-manuel, 2024-06-29:
May I know why you did not provide the logs in the correct format like written everywhere? I just want to understand why people are not getting it how to provide correct logs. Where did you get the command from to fetch the logs?

Have you already checked the [FAQ](https://mr-manuel.github.io/venus-os_dbus-serialbattery/faq/#why-do-i-have-a-high-cpu-usage-after-installing-the-driver)?

Your links are not reachable.

josmeijer, 2024-07-02:
Ho,
I got the logs by opening the logfiles in Nano as the instructions on GitHub say.
Sure you slept well?

Verzonden met Proton Mail Android

-------- Oorspronkelijk bericht --------

josmeijer, 2024-07-02:
I fixed the links.
I did check the. FAQ bus somehow missed this one.
I'll try adjusting the parameter. (tonight, when I'll be with the system again).
But resetting is not happening as far as I can see.
Only reporting system overload from VRM (no real-time communication).
And blocking the possibility to switch dynamic ESS on/off.

Verzonden met Proton Mail Android

-------- Oorspronkelijk bericht --------

mr-manuel, 2024-07-02:
Can you send me a screenshot of it? Nowhere on GitHub it is written that you should open the logfiles with nano.

## When opening an issue

[image]

[image]

## When checking the documentation

[image]

[image]

I'm only wondering how people still do not understand to properly provide logs, after putting that much effort in the docs to make it understandable for everyone. But it seems that people are just too lazy to read nowadays, therefore I wanted to understand, if somewhere the instruction is not clear or wrong. I have to repeat this in nearly every issue and after 100 issues I'm getting tired of it.

Remove the links and post the content directly here in the issue, the links will not work forever. The Color Control GX has a very weak CPU, from what I know of. I don't know, if it can even handle that driver.

josmeijer, 2024-07-02:
It was perfectly understandable but I did not succeed following the instructions. Somehow the whole terminal froze and <ctrl> X couldn't get me out. So I used the method I used since I started with CP/M somewhere in the eighties. Open in a text editor. (I used Nano which on your GitHub pages is "recommended") Copy the text with <ctrl> c and paste it with <ctrl> v.

(I added a screenshot)

Verzonden met Proton Mail Android

-------- Oorspronkelijk bericht --------

josmeijer, 2024-07-02:
[image]

this is as much output I get if I use the instructions (by copying the line with the copy pictogram. If I paste it with the right mouse button I presume I get the exact copy of the text that you typed so what can possibly go wrong :-)

mr-manuel, 2024-07-02:
You have to exit with `CTRL + C` not `X`. This is one log. We need the second one also: https://mr-manuel.github.io/venus-os_dbus-serialbattery/troubleshoot/#datalogdbus-serialbatteryttyusbcurrent-or-datalogdbus-serialbatteryttyama0current

josmeijer, 2024-07-02:
ok, that worked. I was apparently too much stuck in CP/M, holding on to ctrl X :-)
I replaced the second logfile by the output generated by grep. 
The first one does not respond. If I open it with nano, on the other hand I get this:
@4000000066815dc22f969ff4 serstart starting
@4000000066815dc303c72f9c INFO: loading config file /etc/venus/serial-starter.conf
@4000000066815dc535859f3c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000066815dc815b53294 INFO: Create daemontools service vedirect-interface.ttyO0
@4000000066815dca2c59c08c INFO: Create daemontools service mk2-dbus.ttyO1
@4000000066815dcd0b88c04c INFO: Create daemontools service vedirect-interface.ttyO2
@4000000066815dcf0154dbc4 INFO: Start service vedirect-interface.ttyO0
@4000000066815dd101fed37c INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@4000000066815dd10d405774 INFO: Start service mk2-dbus.ttyO1
@4000000066815dd329dcba6c INFO: Start service vedirect-interface.ttyO2
@4000000066815dd735dda27c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066815df406a0c50c INFO: Start service vedirect-interface.ttyO0
@4000000066816349346f83ec INFO: Start service vedirect-interface.ttyO0
@400000006681634b113ed544 INFO: Start service vedirect-interface.ttyO2
@400000006681634c32df0c64 INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@400000006681635239989b24 INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000066816358333fe944 INFO: Create daemontools service gps-dbus.ttyUSB0
@400000006681635e38d1c4cc INFO: Start service gps-dbus.ttyUSB0 once
@4000000066816368357bbbfc INFO: Create daemontools service vedirect-interface.ttyUSB0
@400000006681636e3b84cf0c INFO: Start service vedirect-interface.ttyUSB0 once
@400000006681637437ce6274 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066816456143564ac INFO: Start service mk2-dbus.ttyO1
@40000000668165421cf4a2c4 INFO: Start service vedirect-interface.ttyO0
@40000000668165441b0e048c INFO: Start service vedirect-interface.ttyO2
@40000000668165462f7f1c6c INFO: Start service dbus-cgwacs.ttyUSB0 once
@400000006681654d2f57126c INFO: Start service gps-dbus.ttyUSB0 once

what am I doing wrong?

mr-manuel, 2024-07-02:
You are drifting away from your solution. Check the FAQ and do what the FAQ say. It should solve your problem.

[How to troubleshoot high CPU load?](https://mr-manuel.github.io/venus-os_dbus-serialbattery/faq/#how-to-troubleshoot-high-cpu-load)

[Why do I have a high CPU usage after installing the driver?](https://mr-manuel.github.io/venus-os_dbus-serialbattery/faq/#why-do-i-have-a-high-cpu-usage-after-installing-the-driver)

mr-manuel, 2024-07-06:
Do you need still help?

josmeijer, 2024-07-06:
No, it seems that though I'm cancelled by Victron level3 support, (fot using htop which is Linux in their view) I got it more or less stable by setting POLLING_INTERVAL to 2.
Thank you for your help (and patience 😁).

closed 2024-07-07 by mr-manuel