mr-manuel/venus-os_dbus-serialbattery #43: v1.4.20240702dev ERROR:dbus.service:Unable to append

tlamoureux24, 2024-07-02:
### Describe the bug

Trying the nightly build with jk-pb2a16s20p with rs485 usb adapter

### How to reproduce

Simply install v1.4.20240702dev

### Expected behavior

[image]
[image]
[image]

### Driver version of the currently installed driver

v1.4.20240702dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.33

### BMS type

JKBMS Inverter

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
Nothing
```

### Relevant log output

```shell
*** CCGX booted (0) ***
INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
INFO:SerialBattery:Connection established to Jkbms_pb
INFO:SerialBattery:DeviceInstance = 1
INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
INFO:SerialBattery:Used DeviceInstances = []
INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
INFO:SerialBattery:publish config values = False
INFO:SerialBattery:Battery JKBMS PB Model connected to dbus from /dev/ttyUSB0
INFO:SerialBattery:========== Settings ==========
INFO:SerialBattery:> Connection voltage: 52.93 V | Current: -4.612 A | SoC: 98%
INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 50.0 A (read from BMS)
INFO:SerialBattery:> CVCM:     True
INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
INFO:SerialBattery:> CHARGE FET: 1 | DISCHARGE FET: 1 | BALANCE FET: None
INFO:SerialBattery:Serial Number/Unique Identifier: 4041849088
ERROR:dbus.service:Unable to append ({'/Mgmt/ProcessName': {'Value': dbus.String('/opt/victronenergy/dbus-serialbattery/dbushelper.py', variant_level=1), 'Text': '/opt/victronenergy/dbus-serialbattery/dbushelper.py'}, '/Mgmt/ProcessVersion': {'Value': dbus.String('Python 3.8.18', variant_level=1), 'Text': 'Python 3.8.18'}, '/Mgmt/Connection': {'Value': dbus.String('Serial /dev/ttyUSB0__0x01', variant_level=1), 'Text': 'Serial /dev/ttyUSB0__0x01'}, '/DeviceInstance': {'Value': dbus.Int32(1, variant_level=1), 'Text': '1'}, '/ProductId': {'Value': dbus.Int32(47735, variant_level=1), 'Text': '0xBA77'}, '/ProductName': {'Value': dbus.String('SerialBattery(JKBMS PB Model)', variant_level=1), 'Text': 'SerialBattery(JKBMS PB Model)'}, '/FirmwareVersion': {'Value': dbus.String('1.4.20240702dev', variant_level=1), 'Text': '1.4.20240702dev'}, '/HardwareVersion': {'Value': dbus.String('15A\x00 / 15.2', variant_level=1), 'Text': '15A\x00 / 15.2'}, '/Connected': {'Value': dbus.Int32(1, variant_level=1), 'Text': '1'}, '/CustomName': {'Value': dbus.String('SerialBattery(JKBMS PB Model)', variant_level=1), 'Text': 'SerialBattery(JKBMS PB Model)'}, '/Serial': {'Value': dbus.String('4041849088', variant_level=1), 'Text': '4041849088'}, '/DeviceName': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/State': {'Value': dbus.Int32(0, variant_level=1), 'Text': '0'}, '/ErrorCode': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Info/BatteryLowVoltage': {'Value': dbus.Double(46.4, variant_level=1), 'Text': '46.4'}, '/Info/MaxChargeVoltage': {'Value': dbus.Double(55.2, variant_level=1), 'Text': '55.20V'}, '/Info/MaxChargeCurrent': {'Value': dbus.Double(40.0, variant_level=1), 'Text': '40.00A'}, '/Info/MaxDischargeCurrent': {'Value': dbus.Double(50.0, variant_level=1), 'Text': '50.00A'}, '/Info/ChargeMode': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Info/ChargeModeDebug': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Info/ChargeModeDebugFloat': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Info/ChargeModeDebugBulk': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Info/ChargeLimitation': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Info/DischargeLimitation': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/NrOfCellsPerBattery': {'Value': dbus.Int32(16, variant_level=1), 'Text': '16'}, '/System/NrOfModulesOnline': {'Value': dbus.Int32(1, variant_level=1), 'Text': '1'}, '/System/NrOfModulesOffline': {'Value': dbus.Int32(0, variant_level=1), 'Text': '0'}, '/System/NrOfModulesBlockingCharge': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/NrOfModulesBlockingDischarge': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Capacity': {'Value': dbus.Double(299.327, variant_level=1), 'Text': '299.33Ah'}, '/InstalledCapacity': {'Value': dbus.Double(304.0, variant_level=1), 'Text': '304Ah'}, '/ConsumedAmphours': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Soc': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Dc/0/Voltage': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Dc/0/Current': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Dc/0/Power': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Dc/0/Temperature': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Dc/0/MidVoltage': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Dc/0/MidVoltageDeviation': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MinCellTemperature': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MinTemperatureCellId': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MaxCellTemperature': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MaxTemperatureCellId': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MOSTemperature': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature1': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature1Name': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature2': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature2Name': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature3': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature3Name': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature4': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/Temperature4Name': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MaxCellVoltage': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MaxVoltageCellId': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MinCellVoltage': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/System/MinVoltageCellId': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/History/ChargeCycles': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/History/TotalAhDrawn': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Balancing': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Io/AllowToCharge': {'Value': dbus.Int32(0, variant_level=1), 'Text': '0'}, '/Io/AllowToDischarge': {'Value': dbus.Int32(0, variant_level=1), 'Text' [... cut ...] /Alarms/LowTemperature': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Alarms/BmsCable': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Alarms/HighInternalTemperature': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Voltages/Cell1': {'Value': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Text': '---'}, '/Balances/Cell1': {'Value': dbus.Arr
```

mr-manuel, 2024-07-03:
Thanks for reporting and providing all the needed data. Install the latest nightly `v1.4.20240703dev` and the problem should be fixed.

closed 2024-07-03 by mr-manuel

tlamoureux24, 2024-07-03:
Thanks for the quick fix, i can confirm bug solved.

grydo, 2024-07-03:
hi, Which port do you connect the cable?

tlamoureux24, 2024-07-10:
Hello, sorry, not be notified of your question .... i use the RS485 port just near the CAN port
I set dip switch for device #1 and with the bluetooth android app set the protocol for UART1 to 1 (i'm unable to connect to BMS if dip switch set to 0)
I use a USB to RJ45 RS485 adapter (the one sell with Seplos V3 BMS) and only connect A and B (no ground)
The adapter is plug in the first USB port of my cerbo gx

commit 9fe9e848 (2024-07-03): Fix #43
https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/43