mr-manuel/venus-os_dbus-serialbattery #44: jk-bms inverter not connect

grydo, 2024-07-03:
### Describe the problem

No connection bms vs venus os

### Driver version of the currently installed driver

v1.4.20240703dev

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.30 11

### BMS type

JKBMS Inverter

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
2024-07-03 22:21:09.305412500 INFO:SerialBattery:Testing Jkbms
2024-07-03 22:21:09.579215500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-03 22:21:09.581606500 INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
2024-07-03 22:21:11.283218500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2024-07-03 22:21:11.284216500 INFO:SerialBattery:Testing LltJbd
2024-07-03 22:21:11.557328500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-03 22:21:11.559196500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-07-03 22:21:12.527525500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2024-07-03 22:21:12.528782500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2024-07-03 22:21:12.803026500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-03 22:21:12.804407500 INFO:SerialBattery:Testing Seplos
```

### Any other information that may be helpful

tested all port rs485 bms

grydo, 2024-07-04:
> Is this a first time installation?   No, I use it regularly with other bms
> 
> How does you cabeling look like?  Converter rs485-usb cable bms pin 1 pin 2, cable converter pin 1 in B, pin 2 in A.

mr-manuel, 2024-07-04:
What is the exact model of the BMS?

grydo, 2024-07-04:
> What is the exact model of the BMS?

JK-PB2A16S20P

mr-manuel, 2024-07-04:
Check this https://github.com/Louisvdw/dbus-serialbattery/discussions/969

grydo, 2024-07-04:
not work rs485 only CAN?

mr-manuel, 2024-07-05:
Please comment on the discussion above, if you don't understand how to connect correctly hardware wise.

closed 2024-07-05 by mr-manuel