mr-manuel/venus-os_dbus-serialbattery #47: SoC value is not correct

stixif, 2024-07-06:
### Describe the bug

the BMS Shows 100% on Both (LltJbd)
the VenusOS 96% and 100%
[image]

### How to reproduce

?

### Expected behavior

correct SoC

### Driver version of the currently installed driver

v1.3.20240705dev

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.34

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

18

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
;LOGGING = INFO

CUSTOM_BATTERY_NAMES = /dev/ttyACM0:BattPack1,/dev/ttyACM1:BattPack2

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 130.0
MAX_BATTERY_DISCHARGE_CURRENT = 130.0

; --------- Additional settings ---------
; Specify one or more BMS types to load else leave empty to try to load all available
; Available BMS:
;     Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; Available BMS, but disabled by default (just enter one or more below and it will be enabled):
;     ANT, MNB, Sinowealth
BMS_TYPE = LltJbd
```

### Relevant log output

```shell
NR1
@40000000668917ad2b811f84 INFO:SerialBattery:Starting dbus-serialbattery
@40000000668917ad2b983d7c INFO:SerialBattery:Venus OS v3.34
@40000000668917ad2b9e9e4c INFO:SerialBattery:dbus-serialbattery v1.3.20240705dev
@40000000668917bd2c113254 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@40000000668917bd2c18334c INFO:SerialBattery:Testing LltJbd
@40000000668917be2509f0f4 INFO:SerialBattery:Connection established to LltJbd
@40000000668917bf18627844 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
@40000000668917bf2b2db7f4 INFO:SerialBattery:DeviceInstance = 1
@40000000668917bf2b3ef9ec INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
@40000000668917bf2b47374c INFO:SerialBattery:Used DeviceInstances = ['3', '1']
@40000000668917bf2b4e9604 INFO:SerialBattery:com.victronenergy.battery.ttyACM1
@40000000668917c0126651cc INFO:SerialBattery:publish config values = False
@40000000668917c0127b0e64 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM1
@40000000668917c012913e14 INFO:SerialBattery:========== Settings ==========
@40000000668917c012baf9ac INFO:SerialBattery:> Connection voltage: 61.85 V | Current: 3.26 A | SoC: 96.44%
@40000000668917c012bb20bc INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
@40000000668917c012c365ec INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000668917c012cd127c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
@40000000668917c012d8f574 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 130.0 A | MAX BATTERY DISCHARGE CURRENT: 130.0 A
@40000000668917c012e29a34 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 190.0 A | MAX BATTERY DISCHARGE CURRENT: 190.0 A (read from BMS)
@40000000668917c012f9a0bc INFO:SerialBattery:> CVCM:     True
@40000000668917c01319e28c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000668917c0132515bc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000668917c0132db8ac INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000668917c013361934 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@40000000668917c01348babc INFO:SerialBattery:Serial Number/Unique Identifier: 40_280.0Ah

NR2
@40000000668917ac234c27dc INFO:SerialBattery:Starting dbus-serialbattery
@40000000668917ac23665ecc INFO:SerialBattery:Venus OS v3.34
@40000000668917ac23707ca4 INFO:SerialBattery:dbus-serialbattery v1.3.20240705dev
@40000000668917bc248fbfb4 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@40000000668917bc248fdef4 INFO:SerialBattery:Testing LltJbd
@40000000668917bd1c03034c INFO:SerialBattery:Connection established to LltJbd
@40000000668917be008dab94 INFO:SerialBattery:Found existing battery with DeviceInstance = 3
@40000000668917be1a770834 INFO:SerialBattery:DeviceInstance = 3
@40000000668917be1a8d89ec INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_3.pid
@40000000668917be1a980584 INFO:SerialBattery:Used DeviceInstances = ['3', '1']
@40000000668917be1aa0101c INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@40000000668917bf007f0594 INFO:SerialBattery:publish config values = False
@40000000668917bf0097cd54 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@40000000668917bf009f8db4 INFO:SerialBattery:========== Settings ==========
@40000000668917bf00a86b3c INFO:SerialBattery:> Connection voltage: 61.86 V | Current: 3.58 A | SoC: 99.5%
@40000000668917bf00af7bd4 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
@40000000668917bf00b62ac4 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000668917bf00bd81ac INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
@40000000668917bf00c46f1c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 130.0 A | MAX BATTERY DISCHARGE CURRENT: 130.0 A
@40000000668917bf00cbd98c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 190.0 A | MAX BATTERY DISCHARGE CURRENT: 190.0 A (read from BMS)
@40000000668917bf00d2e63c INFO:SerialBattery:> CVCM:     True
@40000000668917bf00da6c04 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000668917bf00e14dbc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000668917bf00f3acdc INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000668917bf00f3c834 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@40000000668917bf00fb7124 INFO:SerialBattery:Serial Number/Unique Identifier: 40_279.9Ah
```

### Any other information that may be helpful

what can i do to become the soc value from the BMS

mr-manuel, 2024-07-06:
The driver can only get, what the BMS is sending. Have you the same problem with other stable releases?

stixif, 2024-07-06:
yes in all versions...
the BMS App shows 100% for both Packs

but in VenusOS i geht 96 and 100...

mr-manuel, 2024-07-06:
1. Is this only happening when the batteries are on 100%?
2. Does the SoC match, if the batteries are below 90% like 60%, 50%, 40%, 10%, 0%?
3. Please enable `GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = False` and send me some screenshots of both batteries and the parameters page. You have to scroll down. Not all data will fit on one screenshot.
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/ea4a175cca67d34dc26f7936363127673f027729/etc/dbus-serialbattery/config.default.ini#L408-L412
4. Did you restart the driver, when the batteries where both 100% in the app? If yes are the logs above from that situation? If not please reboot/reinstall and post the logs again.

I don't think there could be done much, since the driver is displaying the SoC as it is send from the BMS. No manipulation in this case.

stixif, 2024-07-07:
Hi, sorry for my late answer...
1. no... its all the time...
2. no...
3. here you have... 1. batpack1 / 2. batpack2
[image]
[image]

4. this logs are from an new reinstalled and rebootet one...

the Difference is in the Consumed AmpHours

mr-manuel, 2024-07-07:
For 3. you have to scroll down, below "Discharge Current Limit (DCL) are more fields, if you enabled it in the `config.ini` and reinstalled the driver.

stixif, 2024-07-07:
Hi, and Tanks...
For 2.
[image]
[image]
[image]

For 3. I do an Reinstall but get Not more information...
Should i deinstall and Install new?!?

mr-manuel, 2024-07-07:
What about the second battery?

Which BMS are you using? LLT, JBD or another one?

3. Have you added the value to the `config.ini`?

stixif, 2024-07-07:
for the Pictures...
In VenusOS
Bat1 = 61
Bat2 = 73

in the APP 
Bat1 = 63
Bat2 = 67

for 3.
yes i do...
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True

and i do 
/data/etc/dbus-serialbattery/reinstall-local.sh
and test a new install and reboot but i get not more...

now i´m on the Stable v1.3.20240705

stixif, 2024-07-07:
and now the sun cames out and it would not load my battery with
v1.3.20240705

with the "old" Beta it works
v1.3.20240615beta"
^^ i reinstalled now

stixif, 2024-07-07:
now i have mor Information...
if i use this config all works well
`
; Show additional info in GUI -> Serialbattery -> Parameters
; This will show additional information to better understand how the driver works
; and what values are currently set which are not shown elsewhere in the GUI
; You have to scroll down to see the additional information
;GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
`

and with this config
i geht low Voltage warning...
(so it do not load my batteries)
`

; Show additional info in GUI -> Serialbattery -> Parameters
; This will show additional information to better understand how the driver works
; and what values are currently set which are not shown elsewhere in the GUI
; You have to scroll down to see the additional information
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True

`

with both Versions...
v1.3.20240705
and
v1.3.20240615beta

so now i use v1.3.20240705 with my first config

stixif, 2024-07-07:
the JBD

mr-manuel, 2024-07-07:
Ok, thats very strange, that only you have this issue. You have also the model and hardware version?

stixif, 2024-07-07:
hmmm sorry ;)
Pack1
[image]

Pack2
[image]

mr-manuel, 2024-07-07:
Have you the possibility to connect one BMS over Bluetooth? Would be interesting to see the result there.

mr-manuel, 2024-07-07:
@idstein @Jaco-TTT @packrat969 @wherzig @sledge2019 @pmarches have you also this problem?

stixif, 2024-07-07:
i connect over BLE and have the same issu...
in the App 82% and in VenusOS 80%

Jaco-TTT, 2024-07-08:
>  @Jaco-TTT  have you also this problem?
Cannot comment, as I'm currently running without the software.
I do know the SOC from the BMS, compared to Inverter SOC versus BMV(Smartshunt) SOC always differs.

mr-manuel, 2024-07-08:
@stixif now the debugging with `GUI_PARAMETERS_SHOW_ADDITIONAL_INFO` was fixed. Install the latest nightly and post the screenshots of the parameters page (should be around 5 screenshots).

stixif, 2024-07-08:
Hi, Thanks!
Bat1:
[image]
[image]
[image]
[image]
[image]

Pack2
[image]
[image]
[image]
[image]
[image]

mr-manuel, 2024-07-08:
There are no signs of any issues. Can you request the protocol documentation from your seller?

stixif, 2024-07-10:
I will ask... but I don't have high hopes

stixif, 2024-07-12:
hi Manuel, 
I got something...
[JBD communication protocol new-RS485,RS232,UART.pdf](https://github.com/user-attachments/files/16195342/JBD.communication.protocol.new-RS485.RS232.UART.pdf)

i hope it helps...

mr-manuel, 2024-07-13:
I found something: https://github.com/Louisvdw/dbus-serialbattery/issues/769#issuecomment-1720805325

Can you check that values on your end?

Please also install the latest nightly. If you already have it installed, please reinstall it. Then post fresh logs here, they should contain some important values to understand the problem.

stixif, 2024-07-14:
yes here the same... i have total capacity and cycle capacity.
Bat1
total capacity = 280
cycle capacity = 260
[current1.txt](https://github.com/user-attachments/files/16216175/current1.txt)

Bat2
total capacity = 279,9
cycle capacity = 260
[current2.txt](https://github.com/user-attachments/files/16216174/current2.txt)

From the Log...
INFO:SerialBattery:soc: 74 - self.soc: 79.75
soc is the same like the BMS APP

mr-manuel, 2024-07-14:
@idstein do you know maybe, why the cycle capacity was used and not the total capacity?

Strange, that the app also do not use the cycle capacity, since with time and degradation the total capacity gets less.

@stixif install the latest nightly, I changed it to test in the meanwhile.

mr-manuel, 2024-07-14:
Please install and try again. The error should now be more specific.

stixif, 2024-07-14:
hmm... same again...

@400000006693e4ef0d23f9bc INFO:SerialBattery:
@400000006693e4ef0d297414 INFO:SerialBattery:Starting dbus-serialbattery
@400000006693e4ef0d6150dc INFO:SerialBattery:Venus OS v3.34
@400000006693e4ef0d68f5e4 INFO:SerialBattery:dbus-serialbattery v1.4.20240714dev
@400000006693e4ff0d9fccf4 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006693e4ff0da74ed4 INFO:SerialBattery:Testing LltJbd
@400000006693e4ff1b50be94 ERROR:SerialBattery:Exception occurred: TypeError("unsupported operand type(s) for /: 'int' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd.py line #297
@400000006693e4ff394c01c4 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@400000006693e4ff394c1d1c INFO:SerialBattery:Testing LltJbd
@400000006693e5000aebe40c ERROR:SerialBattery:Exception occurred: TypeError("unsupported operand type(s) for /: 'int' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd.py line #297
@400000006693e50028ce8e5c INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@400000006693e50028d70654 INFO:SerialBattery:Testing LltJbd
@400000006693e500366c9824 ERROR:SerialBattery:Exception occurred: TypeError("unsupported operand type(s) for /: 'int' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd.py line #297
@400000006693e50118b464ec ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM0
@400000006693e5103969f9a4 INFO:SerialBattery:
@400000006693e510397173b4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006693e510398a4efc INFO:SerialBattery:Venus OS v3.34
@400000006693e51039919644 INFO:SerialBattery:dbus-serialbattery v1.4.20240714dev

mr-manuel, 2024-07-14:
Not exactly the same :) we are getting closer. I updated the code again. Else you can write me on Discord to get access to your system. That would speed things up a bit.

mr-manuel, 2024-07-14:
It should be solved now. Found a typo.

stixif, 2024-07-14:
now it works...

but batpack1 is 90% in VenusOS and on the BMS 100%
but the consumed AmpHours are 28Ah

and batpack2 is 100% in VenusOS and BMS
consumed AmpHours 0Ah

but if i disconnect batpack2 it will not load batpack1

mr-manuel, 2024-07-14:
Consumed Amp hours in the app or the driver?

stixif, 2024-07-14:
in the driver...
but it looks like the same in the App (280AH - Remaining 252AH)

mr-manuel, 2024-07-14:
The driver uses the consumed amp hours to have a higher resolution on the SOC. Maybe there is a wrong setting in your BMS? You could try a factory reset of the BMS and see the results.

If the battery is full, is the `remaining capacity` same as the `capacity` or the `cycle capacity`?

I add again the logging, so we see whats happening.

mr-manuel, 2024-07-16:
@stixif can you install from the [`troubleshooting-llt-jbs-soc`](https://github.com/mr-manuel/venus-os_dbus-serialbattery/tree/troubleshooting-llt-jbs-soc) branch and provide fresh logs?

See [Install a specific branch](https://mr-manuel.github.io/venus-os_dbus-serialbattery/general/install#specific-branch) if you don't know how to install a specific branch.

stixif, 2024-07-17:
@mr-manuel i do...

i download it...
https://github.com/mr-manuel/venus-os_dbus-serialbattery/archive/refs/heads/troubleshooting-llt-jbs-soc.tar.gz

and rename it and copy to "/var/volatile/tmp/venus-data.tar.gz"

and install with option 8 of the script

[current2.txt](https://github.com/user-attachments/files/16268039/current2.txt)
[current1.txt](https://github.com/user-attachments/files/16268040/current1.txt)

mr-manuel, 2024-07-18:
From the logfiles the version was not installed somehow. Could you write me on Discord so that I could check your system?

stixif, 2024-07-18:
@mr-manuel 
??? what is discord... i dont know sorry
or can you tell me how i can install your version?

mr-manuel, 2024-07-18:
> See [Install a specific branch](https://mr-manuel.github.io/venus-os_dbus-serialbattery/general/install#specific-branch) if you don't know how to install a specific branch.

try this

stixif, 2024-07-19:
@mr-manuel 
sorry...
now I got it! ;)
(i create the venus-data.tar.gz by myself... i never done this before...)
[current2.txt](https://github.com/user-attachments/files/16312387/current2.txt)
[current1.txt](https://github.com/user-attachments/files/16312388/current1.txt)

mr-manuel, 2024-07-19:
current1:
```
INFO:SerialBattery:soc (BMS): 99 - self.soc (calc): 99.5 - soc_2 (calc): 107.12 - capacity: 27990 - self.cycle_capacity: 26000.0 - capacity_remain: 27850
```

current2:
```
INFO:SerialBattery:soc (BMS): 100 - self.soc (calc): 89.55 - soc_2 (calc): 96.44 - capacity: 28000 - self.cycle_capacity: 26000.0 - capacity_remain: 25075
```

`soc`: comes directly from the BMS
`self.soc`: ` round(100 * capacity_remain / capacity, 2)`
`soc_2`: `round(100 * capacity_remain / self.cycle_capacity, 2)`
`capacity`: comes directly from the BMS
`self.cycle_capacity`: comes directly from the BMS
`capacity_remain`: comes directly from the BMS

So the problem here is, that your BMS shows inconsistent data. The SOC is 100%, but the remaining capacity has not reached at least the `cycle_capacity` in the second case. In the fist case, it's even exceeding this value.

Please ask your seller why that is the case and why this values don't match. Maybe it's a settings issue.

mr-manuel, 2024-07-19:
SOC is set to BMS SOC with https://github.com/mr-manuel/venus-os_dbus-serialbattery/commit/a7b0c4bad4085cacb5a9bd78283cc53796191c68.

closed 2024-07-19 by mr-manuel

stixif, 2024-07-19:
@mr-manuel 
thanks a lot... i will ask...

now i installed the newest snapshot...
but it hangs in the first few minutes the first batpack after a few  minutes all works well...
[image]
[image]

stixif, 2024-07-19:
here 2 logs on the time it stucks
[@40000000669ac346222b1fac.txt](https://github.com/user-attachments/files/16316577/%4040000000669ac346222b1fac.txt)
[@40000000669ac3290e9dbcc4.txt](https://github.com/user-attachments/files/16316578/%4040000000669ac3290e9dbcc4.txt)

mr-manuel, 2024-07-20:
Since this is a different problem open a new issue and provide correct logs. Check the documentation on how to provide correct logs. For this issue the timings were not needed, but for your current problem they are.