mr-manuel/venus-os_dbus-serialbattery #48: Driver v1.3.20240705 GUI_PARAMETERS_SHOW_ADDITIONAL_INFO doesn't work

tlamoureux24, 2024-07-08:
### Describe the bug

Activate the parameter in config.ini change nothing in GUI, no details in GUI -> Serialbattery -> Parameters
Activate the parameter in config.ini result in no charge mode and no CVL in driver

### How to reproduce

Add GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True in config.ini and restart driver

### Expected behavior

Expect details in driver parameters
Expect charge mode
Expect CVL
[image]

### Driver version of the currently installed driver

v1.3.20240705

### Venus OS device type

Cerbo GX

### Venus OS version

3.33

### BMS type

JKBMS Inverter

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

MAX_BATTERY_CHARGE_CURRENT = 30.0
MAX_BATTERY_DISCHARGE_CURRENT = 50.0

BMS_TYPE = Jkbms_pb

GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
```

### Relevant log output

```shell
INFO:SerialBattery:
INFO:SerialBattery:Starting dbus-serialbattery
INFO:SerialBattery:Venus OS v3.33
INFO:SerialBattery:dbus-serialbattery v1.3.20240705
INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
INFO:SerialBattery:Connection established to Jkbms_pb
INFO:SerialBattery:Found existing battery with DeviceInstance = 1
INFO:SerialBattery:DeviceInstance = 1
INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
INFO:SerialBattery:Used DeviceInstances = ['1']
INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
INFO:SerialBattery:publish config values = False
INFO:SerialBattery:Battery JKBMS PB Model connected to dbus from /dev/ttyUSB0
INFO:SerialBattery:========== Settings ==========
INFO:SerialBattery:> Connection voltage: 53.491 V | Current: 8.131 A | SoC: 86%
INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 30.0 A | MAX BATTERY DISCHARGE CURRENT: 50.0 A
INFO:SerialBattery:> CVCM:     True
INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
INFO:SerialBattery:> CHARGE FET: 1 | DISCHARGE FET: 1 | BALANCE FET: None
INFO:SerialBattery:Serial Number/Unique Identifier: 4041849088
```

### Any other information that may be helpful

Generate a low battery alam even if driver seems to work properly
All solved by commenting GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True in config.ini and restart driver

mr-manuel, 2024-07-08:
Thanks for reporting. Could you please install the latest nightly and check the logs? I added some troubleshooting for that.

tlamoureux24, 2024-07-08:
Thanks for the quick reply.
Installed driver v1.3.20240708dev

config.ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

MAX_BATTERY_CHARGE_CURRENT = 30.0
MAX_BATTERY_DISCHARGE_CURRENT = 50.0

BMS_TYPE = Jkbms_pb

GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True

log output
INFO:SerialBattery:
INFO:SerialBattery:Starting dbus-serialbattery
INFO:SerialBattery:Venus OS v3.33
INFO:SerialBattery:dbus-serialbattery v1.3.20240708dev
INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
INFO:SerialBattery:Connection established to Jkbms_pb
INFO:SerialBattery:Found existing battery with DeviceInstance = 1
INFO:SerialBattery:DeviceInstance = 1
INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
INFO:SerialBattery:Used DeviceInstances = ['1']
INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
INFO:SerialBattery:publish config values = False
INFO:SerialBattery:Battery JKBMS PB Model connected to dbus from /dev/ttyUSB0
INFO:SerialBattery:========== Settings ==========
INFO:SerialBattery:> Connection voltage: 54.208 V | Current: 0.387 A | SoC: 100%
INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 30.0 A | MAX BATTERY DISCHARGE CURRENT: 50.0 A
INFO:SerialBattery:> CVCM:     True
INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
INFO:SerialBattery:> CHARGE FET: 1 | DISCHARGE FET: 1 | BALANCE FET: None
INFO:SerialBattery:Serial Number/Unique Identifier: 4041849088
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #680

cerbo gx alarm internal calc error
[image]

No parameter's detail in venus

Let me know if you need more information

mr-manuel, 2024-07-08:
Can you please install the same version again? I updated the code.

tlamoureux24, 2024-07-08:
Same as above only the battery.py error line change.

log output
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704
ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #704

mr-manuel, 2024-07-08:
Unfortunately I'm not able to reproduce this issue, therefore we have to continue like this. Can you reinstall again? Should now be good :)

tlamoureux24, 2024-07-08:
I can confirm, now all good and got parameters detail !!!
Thanks for the support :)

tlamoureux24, 2024-07-08:
My pleasure, if you need a beta tester i can do test for you !