mr-manuel/venus-os_dbus-serialbattery #50: JK-PB2A16S20P Add temperature sensors 3 and 4

tlamoureux24, 2024-07-09:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

Add battery temperature sensors number 3 and 4 in addition to numbers 1 and 2 and mosfet temperature.

KoljaWindeler, 2024-07-10:
Hey @mr-manuel, how motivating are command-tickets like that for you? 
Also: how much are people paying you for your work :) ?

I can take care of this though

tlamoureux24, 2024-07-10:
> Hey @mr-manuel, how motivating are command-tickets like that for you? Also: how much are people paying you for your work :) ?
> 
> I can take care of this though

hello, I don't know if this is ironic, humorous or something else ..... sorry I'm French and I use deepl ......
I posted this enhancement request with the proper form I think because I think it can improve the driver for everyone.
If my request is badly formulated I apologize.
Unless I'm mistaken, temperature values are used in some driver calculations, and what I've noticed on my battery, which is a 16-cell vertical pack, is that the batteries at the top are sometimes 2 to 3 degrees hotter than those at the bottom, so this can have some influence.
Sorry for the inconvenience, my request can be closed.

KoljaWindeler, 2024-07-10:
well maybe I should react a little easier, I'm maintaining some other projects and receive quite some "why is this not working, fix it!" messages .. and I guess I also speak for other open source contributor that we all do this work at our free time so a little "hey guys, thanks for the work, it would be awesome if you could add XYZ" would be nice ..

but anyway .. I already have temp3 and temp4 working and it is a good idea to add it .. will send out an update soon

tlamoureux24, 2024-07-10:
You're right, I should have phrased it more politely and I apologize for that, and you're also right, opensource developers are often not thanked enough for their work.
I don't know how to code, but I think I've contributed a little to my small level of knowledge by reporting bugs with all the information I can and testing everything I'm asked to.
I've already had the opportunity to thank mr-manuel several times for his work and his time (please see https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/43 and https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/48 )

In any case, thank you for your time and your work on this and other projects.
Have a nice day

KoljaWindeler, 2024-07-10:
should be working .. 
[image]
and 
[image]

pull request submitted https://github.com/mr-manuel/venus-os_dbus-serialbattery/pull/53, please reopen if you have an issue
Kolja

closed 2024-07-10 by mr-manuel

tlamoureux24, 2024-07-10:
Thank you very much @KoljaWindeler  and @mr-manuel for the work and the time paste on this enhancement.