mr-manuel/venus-os_dbus-serialbattery #51: VRM Internal Calculation Error

coca7, 2024-07-09:
### Describe the problem

Getting an internal calculation error in the VRM.  Serial battery driver is reporting errors

### Driver version of the currently installed driver

v1.3.20240708dev

### Driver version of the last known working driver

OLD Cerbo GX before it was RMA'd, not sure the version

### Venus OS device type

Cerbo GX

### Venus OS version

3.33

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

4

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 150.0
 MAX_BATTERY_DISCHARGE_CURRENT = 160.0
 SOC_RESET_VOLTAGE = 3.620
 CVL_ICONTROLLER_MODE   = True
 BMS_TYPE = LltJbd
```

### Relevant log output

```shell
Yesterday, from battery on USB0:
2024-07-09 00:28:20.948612500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 00:28:20.950426500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29
2024-07-09 00:48:37.618993500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 00:48:37.620169500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 16 expected 7

Today on USB0:
2024-07-09 00:27:39.851326500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_230.01Ah
2024-07-09 00:28:20.948612500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 00:28:20.950426500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29
2024-07-09 00:48:37.618993500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 00:48:37.620169500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 16 expected 7
2024-07-09 02:33:28.183325500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 02:33:28.184676500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 52
2024-07-09 03:18:28.943190500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 03:18:28.944067500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 48
2024-07-09 03:41:39.887632500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 03:41:39.888492500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29
2024-07-09 04:16:12.753131500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 04:16:12.754674500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 15 expected 7
2024-07-09 04:19:44.043892500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
[... 48 lines cut ...]
2024-07-09 02:39:47.136777500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/57]
2024-07-09 03:19:42.352338500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:32/110]
2024-07-09 05:03:18.139296500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/260]
2024-07-09 05:20:23.513970500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/55]
2024-07-09 05:52:47.124995500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/261]
2024-07-09 06:35:08.895435500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:14/18]
2024-07-09 06:46:01.749376500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/53]
2024-07-09 08:37:46.290849500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-09 08:37:46.292097500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29
2024-07-09 08:51:41.330208500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/56]
2024-07-09 09:00:14.053989500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:32/202]
2024-07-09 09:00:31.261995500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:10/18]
2024-07-09 09:13:27.308925500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/261]
2024-07-09 09:50:17.559429500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:11/252]
2024-07-09 10:45:43.010595500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/56]
2024-07-09 10:52:59.630022500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/261]
2024-07-09 11:09:57.019052500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/261]
2024-07-09 11:13:36.297822500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:32/202]
```

### Any other information that may be helpful

I was having no issues before my Cerbo GX was replaced.  As a result of waiting for a replacement Cerbo Gx and running hte batteries with only a smarthsunt they got out of balance.  Now there is one of three batteries remains out of balance (I think it is USB0).  I will rebalance it manually.  The other thing that changed after I reinstalled the Cerbo is that instead of using BatteryAggregator, I switched to Dbus-Aggregate-Batteries by dr Gigavolt.  There are no errors in the Aggregate batteries log, the VRM  error mentions SerialBattery not Aggregate batteries:  

> Battery Monitor - SerialBattery(LLT/JBD) [1] Error: Internal calculation error

mr-manuel, 2024-07-10:
With the latest stable release I added, that if there are errors in the logs, then an error is shown in the GUI, else the user won't get notified about the errors unless he checks the logs. It could be, that the errors were already present before, but you never checked the logs.

Install the stable release `v1.2.20240408` and check the logs. See [Specific version/troubleshooting option](https://mr-manuel.github.io/venus-os_dbus-serialbattery/general/install#specific-versiontroubleshooting-option) for how to achieve that.

coca7, 2024-07-10:
The errors were not present before because I did check the logs before I replaced my Cerbo GX.  However, the cells are out of balance since the replacement.  I will check the logs again after I rebalance the cells manually.

coca7, 2024-07-10:
v1.3.20240705 was installed.

USB0:

> 2024-07-10 18:17:40.188815500 INFO:SerialBattery:========== Settings ==========
2024-07-10 18:17:40.189831500 INFO:SerialBattery:> Connection voltage: 13.54 V | Current: 1.03 A | SoC: 65.71%
2024-07-10 18:17:40.192837500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2024-07-10 18:17:40.194380500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-07-10 18:17:40.195872500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-07-10 18:17:40.201765500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A
2024-07-10 18:17:40.202655500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 160.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A (read from BMS)
2024-07-10 18:17:40.203500500 INFO:SerialBattery:> CVCM:     True
2024-07-10 18:17:40.204220500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-10 18:17:40.204946500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-10 18:17:40.205539500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-10 18:17:40.206197500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-10 18:17:40.206980500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_230.01Ah
2024-07-10 18:17:57.274024500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/80]
2024-07-10 18:18:03.879120500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-10 18:18:06.450222500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character  0xDD
2024-07-10 18:18:06.450233500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 217

**Restarted the Driver**

> 2024-07-10 18:22:08.484428500 INFO:SerialBattery:
2024-07-10 18:22:08.485241500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-10 18:22:08.487750500 INFO:SerialBattery:Venus OS v3.33
2024-07-10 18:22:08.488593500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
2024-07-10 18:22:24.495742500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-07-10 18:22:24.496470500 INFO:SerialBattery:Testing LltJbd
2024-07-10 18:22:25.029165500 INFO:SerialBattery:Connection established to LltJbd
2024-07-10 18:22:29.991000500 INFO:SerialBattery:Found existing battery with DeviceInstance = 4
2024-07-10 18:22:30.464929500 INFO:SerialBattery:DeviceInstance = 4
2024-07-10 18:22:30.466974500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_4.pid
2024-07-10 18:22:30.467834500 INFO:SerialBattery:Used DeviceInstances = ['2', '3', '4', '1']
2024-07-10 18:22:30.468945500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-07-10 18:22:30.802203500 INFO:SerialBattery:publish config values = False
2024-07-10 18:22:30.804592500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB0
2024-07-10 18:22:30.805691500 INFO:SerialBattery:========== Settings ==========
2024-07-10 18:22:30.807017500 INFO:SerialBattery:> Connection voltage: 13.57 V | Current: 1.2 A | SoC: 65.74%
2024-07-10 18:22:30.808095500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2024-07-10 18:22:30.809706500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-07-10 18:22:30.811252500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-07-10 18:22:30.812600500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A
2024-07-10 18:22:30.813659500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 160.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A (read from BMS)
2024-07-10 18:22:30.815108500 INFO:SerialBattery:> CVCM:     True
2024-07-10 18:22:30.816293500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-10 18:22:30.817293500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-10 18:22:30.818266500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-10 18:22:30.819412500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-10 18:22:30.820558500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_230.01Ah

USB1:

> 2024-07-10 18:17:51.483700500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_230.0Ah
> 2024-07-10 18:18:13.357435500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:24/80]
> 2024-07-10 18:22:09.877901500 INFO:SerialBattery:
> 2024-07-10 18:22:09.879042500 INFO:SerialBattery:Starting dbus-serialbattery
> 2024-07-10 18:22:09.881538500 INFO:SerialBattery:Venus OS v3.33
> 2024-07-10 18:22:09.882559500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
> 2024-07-10 18:22:25.898373500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
> 2024-07-10 18:22:25.899119500 INFO:SerialBattery:Testing LltJbd
> 2024-07-10 18:22:26.720690500 INFO:SerialBattery:Connection established to LltJbd
> 2024-07-10 18:22:33.898593500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
> 2024-07-10 18:22:34.423336500 INFO:SerialBattery:DeviceInstance = 1
> 2024-07-10 18:22:34.425581500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
> 2024-07-10 18:22:34.426723500 INFO:SerialBattery:Used DeviceInstances = ['2', '3', '4', '1']
> 2024-07-10 18:22:34.428039500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
> 2024-07-10 18:22:34.935187500 INFO:SerialBattery:publish config values = False
[... cut ...]
2024-07-10 18:22:36.453881500 INFO:SerialBattery:> CVCM:     True
2024-07-10 18:22:36.456529500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-10 18:22:36.458545500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-10 18:22:36.459572500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-10 18:22:36.462268500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-10 18:22:36.463495500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_229.0Ah

mr-manuel, 2024-07-10:
The problem is gone now?

coca7, 2024-07-10:
The VRM error is gone, but the errors with the BMS in the logs are still present.  

2024-07-10 18:38:17.120873500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:32/91]

2024-07-10 18:22:34.948813500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_230.0Ah
2024-07-10 18:42:49.640877500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2024-07-10 18:42:49.642282500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 71

2024-07-10 18:22:36.463495500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_229.0Ah
2024-07-10 18:38:29.694159500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:32/114]
2024-07-10 18:59:02.370959500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/261]

coca7, 2024-07-11:
Maybe.  The data cables are not shielded.  I do find it notable that the battery that throws the most errors is the one out of balance.  I have an active balancer on the way to help top balance the battery.

coca7, 2024-07-11:
Now that I think about it.  I think I might have recently zip tied the data cables to one of larger battery cables.  I will report back after I have an opportunity to examine it physically.

mr-manuel, 2024-07-14:
Any updates? Else I would close this.

coca7, 2024-07-14:
No, because I have not had the opportunity to inspect the cables physically.  Also the active balancer did not come in yet.

mr-manuel, 2024-07-14:
Also see [What can I do, if the BMS communication is unstable?](https://mr-manuel.github.io/venus-os_dbus-serialbattery/faq/#what-can-i-do-if-the-bms-communication-is-unstable).

closed 2024-07-14 by mr-manuel

coca7, 2024-08-01:
I finally got my active balancer to fix my problem battery, and I examined my data cables and the BMS's.  I have some thoughts.  1) I don't think the data cables were the problem.  They are near the DC power cables that go from the batter to the DC bus bar, but they are not touching or zip tied as I thought they were.  2) I noted that Dbus-Aggregate-batterises wasn't producing an error but kept limiting the current.  I looked into it and a second battery developed a single high cell.   I examined the BMS software itself and I noted a bunch of over volts.  So...I suspect the internal calculation error was a result of high cells reaching 3.45 volts before the other cells.  Dbus-Aggregate batteries fixed the good battery.  I am using an active balancer to restore the top balance of the battery that has a cell replaced.