mr-manuel/venus-os_dbus-serialbattery #60: dbus-serialbattery starts but dbusmonitor skips because no device instance

BipedalPrimate, 2024-07-27:
### Describe the problem

Dbus-serialbattery seems to start correctly, appears in remote console, communication to Jkbms occurs, but appears skipped because no device instance.  

[image]

[image]

[image]

[image]

### Driver version of the currently installed driver

v1.3.20240705

### Venus OS device type

Raspberry Pi

### Venus OS version

Venus OS v3.13

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 125.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used for:
;         - Calculating the min/max battery voltage
;         - Trigger the SoC reset when SoC calculation is enabled
; Example:
;     16 cells * 3.45 V/cell = 55.2 V max charge voltage.
;     16 cells * 2.90 V/cell = 46.4 V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

[... 505 lines cut ...]
root@raspberrypi4:~# dbus-spy

com.victronenergy.battery.ttyUSB2
Cell/7/Volts                                                               3.336
Cell/8/Volts                                                               3.336
Cell/9/Volts                                                               3.336
Cell/Diff                                                                  0.005
Cell/Sum                                                                   53.38
Connected                                                                      1
ConsumedAmphours                                                           87.99
CurrentAvg                                                                  -0.2
CustomName                                                  SerialBattery(Jkbms)
Dc/0/Current                                                                -0.2
Dc/0/MidVoltage                                                                -
Dc/0/MidVoltageDeviation                                                       -
Dc/0/Power                                                                -10.67
Dc/0/Temperature                                                              20
Dc/0/Voltage                                                               53.37
**DeviceInstance                                                                 1**
DeviceName                                                                     -
ErrorCode                                                                      -
FirmwareVersion                                                     1.3.20240705
HardwareVersion                                        JKBMS 16 cells (20240101)
Info/BatteryLowVoltage                                                      46.4
Info/ChargeLimitation                                 Max Battery Charge Current

mr-manuel, 2024-07-27:
Please try the nightly build.

BipedalPrimate, 2024-07-27:
dbus-serialbattery v1.4.20240721dev installed,  

Some differences in outcomes.
No longer appears in remote console.
[image]

Log file indicates "Battery JKBMS connected to dbus from /dev/ttyUSB2" but creates a different dbus entry com.victronenergy.battery.ttyUSB2__0

2024-07-27 07:15:16.513085500 *** CCGX booted (0) ***
2024-07-27 07:15:52.474661500 INFO:SerialBattery:
2024-07-27 07:15:52.474888500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-27 07:15:52.475862500 INFO:SerialBattery:Venus OS v3.13
2024-07-27 07:15:52.476140500 INFO:SerialBattery:dbus-serialbattery v1.4.20240721dev
2024-07-27 07:16:24.407564500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-07-27 07:16:24.407857500 INFO:SerialBattery:Testing Jkbms
2024-07-27 07:16:24.469538500 INFO:SerialBattery:Connection established to Jkbms
2024-07-27 07:16:24.765916500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-07-27 07:16:24.883842500 INFO:SerialBattery:DeviceInstance = 1
2024-07-27 07:16:24.884618500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-07-27 07:16:24.884910500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-07-27 07:16:24.885214500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB2__0
2024-07-27 07:16:24.911019500 INFO:SerialBattery:publish config values = True
2024-07-27 07:16:24.927184500 INFO:SerialBattery:Polling data every 1.000 s
2024-07-27 07:16:24.929252500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB2
2024-07-27 07:16:24.936973500 INFO:SerialBattery:========== Settings ==========
2024-07-27 07:16:24.937465500 INFO:SerialBattery:> Connection voltage: 53.38 V | Current: -0.0 A | SoC: 71% | SoC calc: 71%
2024-07-27 07:16:24.937962500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-07-27 07:16:24.938390500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2024-07-27 07:16:24.938811500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-07-27 07:16:24.939582500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 125.0 A
2024-07-27 07:16:24.939999500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 75.0 A | MAX BATTERY DISCHARGE CURRENT: 125.0 A (read from BMS)
2024-07-27 07:16:24.940465500 INFO:SerialBattery:> CVCM:     True
2024-07-27 07:16:24.940857500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-27 07:16:24.941189500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-27 07:16:24.941560500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-27 07:16:24.942681500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-27 07:16:24.942983500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A24S15P
2024-07-27 07:16:24.982493500 INFO:SerialBattery:Monitoring external current using: com.victronenergy.battery.ttyUSB3/Dc/0/Current
2024-07-27 07:16:24.993584500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2024-07-27 07:16:24.995208500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
2024-07-27 07:16:25.004470500 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 289
2024-07-27 07:16:25.005618500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2__0, scanning and storing items
2024-07-27 07:16:50.006690500 INFO:dbusmonitor:GetItems failed, trying legacy methods
2024-07-27 07:17:15.008488500 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2__0 was skipped because it has no device instance
2024-07-27 07:17:15.009125500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====

tail -F -n 100 /data/log/serial-starter/current | tai64nlocal
2024-07-27 07:15:20.291556500 INFO: Create daemontools service dbus-cgwacs.ttyUSB0
2024-07-27 07:15:22.176498500 INFO: Create daemontools service mk2-dbus.ttyUSB1
2024-07-27 07:15:24.418796500 INFO: Create daemontools service dbus-imt-si-rs485tc.ttyUSB2
2024-07-27 07:15:25.840983500 INFO: Create daemontools service vedirect-interface.ttyUSB3
2024-07-27 07:15:26.429415500 INFO: Start service dbus-cgwacs.ttyUSB0 once
2024-07-27 07:15:28.294732500 INFO: Start service mk2-dbus.ttyUSB1
2024-07-27 07:15:30.504379500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-07-27 07:15:31.899979500 INFO: Start service vedirect-interface.ttyUSB3 once
2024-07-27 07:15:33.563620500 INFO: Create daemontools service dbus-modbus-client.serial.ttyUSB2
2024-07-27 07:15:39.616347500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-07-27 07:15:47.121963500 INFO: Create daemontools service dbus-serialbattery.ttyUSB2
2024-07-27 07:16:09.087162500 INFO: Start service dbus-serialbattery.ttyUSB2 once

head /data/var/lib/serial-starte
==> /data/var/lib/serial-starter/ttyUSB0 <==
cgwacs

==> /data/var/lib/serial-starter/ttyUSB1 <==
mkx

==> /data/var/lib/serial-starter/ttyUSB2 <==
sbattery

[... 3 lines cut ...]
dbus-spy

com.victronenergy.battery.ttyUSB2__0
Cell/6/Volts                                                               3.339
Cell/7/Volts                                                               3.336
Cell/8/Volts                                                               3.336
Cell/9/Volts                                                               3.334
Cell/Diff                                                                  0.006
[... cut ...]
Dc/0/Voltage                                                               53.37
DeviceInstance                                                                 1
DeviceName                                                                     -
ErrorCode                                                                      -
FirmwareVersion                                                  1.4.20240721dev
HardwareVersion                                JKBMS 11.XW S11.25 16S (20240101)
Info/BatteryLowVoltage                                                      46.4

mr-manuel, 2024-07-27:
Try to remove this settings, since they are for external current sensors like a SmartShunt:

```ini
EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = com.victronenergy.battery.ttyUSB3
EXTERNAL_CURRENT_SENSOR_DBUS_PATH = /Dc/0/Current
```

BipedalPrimate, 2024-07-27:
A SmartShunt is connected on ttyUSB3 via VEDirect.

Should I revert to v1.3.20240705 to remove the external current sensor device/path?

head /data/var/lib/serial-starter/*
==> /data/var/lib/serial-starter/ttyUSB0 <==
cgwacs

==> /data/var/lib/serial-starter/ttyUSB1 <==
mkx

==> /data/var/lib/serial-starter/ttyUSB2 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB3 <==
vedirect

mr-manuel, 2024-07-27:
Ok, then it should be correct. If I remember correctly someone else had the same problem as you. If you remove this setting the battery is shown correctly?

BipedalPrimate, 2024-07-27:
Removing the external current sensor setting on v1.4.20240721dev has worked.

Battery is shown correctly.

dbus-spy still shows com.victronenergy.battery.ttyUSB2__0  instead of com.victronenergy.battery.ttyUSB2

VRM is also displaying correctly.

mr-manuel, 2024-07-27:
Then it seems something related to fetching the data from a SmartShunt. Can you write me on Discord?

BipedalPrimate, 2024-07-27:
Yes.  
Done.

BipedalPrimate, 2024-10-06:
Have you mounted your SmartShunt in your system?
Did it work correctly with dbus_serialbattery?

mr-manuel, 2024-10-06:
It is fixed with `v1.5.20241006dev`.

closed 2024-10-06 by mr-manuel