mr-manuel/venus-os_dbus-serialbattery #63: Internal calculation error

frbuceta, 2024-08-02:
### Describe the bug

When the battery is reaching 100% of its charge, then the driver gives this error.

> This screenshot shows that it is at 71% but this is not true. The smartshunt says it is at 99%.
[image]

### How to reproduce

I don't know exactly what is going on but it gives it automatically when it is reaching 100% of its load.

### Expected behavior

Do not give this error.

### Driver version of the currently installed driver

v1.4.20240721dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.50~11

### BMS type

JKBMS (Heltec BMS)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 115.0
MAX_BATTERY_DISCHARGE_CURRENT = 230.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.570
```

### Relevant log output

```shell
2024-08-02 14:01:44.730730500 python: can't open file '/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py': [Errno 2] No such file or directory
2024-08-02 14:02:17.192223500 INFO:SerialBattery:
2024-08-02 14:02:17.211029500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-02 14:02:17.217619500 INFO:SerialBattery:Venus OS v3.50~11
2024-08-02 14:02:17.231404500 INFO:SerialBattery:dbus-serialbattery v1.4.20240721dev
2024-08-02 14:02:33.250878500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-02 14:02:33.251952500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-08-02 14:02:33.400744500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:33.402029500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-08-02 14:02:33.554458500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:33.555420500 INFO:SerialBattery:Testing Ecs
2024-08-02 14:02:33.641451500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:33.642560500 INFO:SerialBattery:Testing EG4_Lifepower at address "\x01"
2024-08-02 14:02:33.956174500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:34.257163500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:34.563215500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:34.564711500 INFO:SerialBattery:Testing EG4_LL at address "\x01"
2024-08-02 14:02:34.871013500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:34.872656500 INFO:SerialBattery:Testing HeltecModbus at address "\x01"
2024-08-02 14:02:38.684925500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:38.684934500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-08-02 14:02:47.723069500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-02 14:02:47.723079500 INFO:SerialBattery:Testing Jkbms
2024-08-02 14:02:47.793541500 INFO:SerialBattery:Connection established to Jkbms
2024-08-02 14:02:51.015879500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-08-02 14:02:51.371065500 INFO:SerialBattery:DeviceInstance = 1
2024-08-02 14:02:51.372727500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-08-02 14:02:51.373409500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-08-02 14:02:51.374346500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1__0
2024-08-02 14:02:51.408283500 INFO:SerialBattery:publish config values = False
2024-08-02 14:02:51.414908500 INFO:SerialBattery:Polling data every 1.000 s
2024-08-02 14:02:51.419419500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB1
2024-08-02 14:02:51.420294500 INFO:SerialBattery:========== Settings ==========
2024-08-02 14:02:51.422223500 INFO:SerialBattery:> Connection voltage: 55.52 V | Current: -1.86 A | SoC: 71%
2024-08-02 14:02:51.422233500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-08-02 14:02:51.422870500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-02 14:02:51.424032500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-02 14:02:51.455591500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 115.0 A | MAX BATTERY DISCHARGE CURRENT: 230.0 A
2024-08-02 14:02:51.457711500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A (read from BMS)
2024-08-02 14:02:51.457722500 INFO:SerialBattery:> CVCM:     True
2024-08-02 14:02:51.458068500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-02 14:02:51.458879500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-02 14:02:51.459701500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-02 14:02:51.468062500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-02 14:02:51.468922500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A20S20P
2024-08-02 14:03:08.436977500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:85/40993]
2024-08-02 14:03:08.442204500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #609
2024-08-02 14:03:08.443417500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-02 14:03:08.444824500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-02 14:03:08.557871500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1131
```

mr-manuel, 2024-08-04:
The error is related to a wrong reading from the BMS.

As the log say `If you don't see this warning very often, you can ignore it.`. You see it very often?

frbuceta, 2024-08-04:
It has never appeared until I updated a few days ago, and it appears every day several times a day.

I have managed to avoid configuring the calculated SOC and to ignore the BMS.

Some bug must have been introduced so that now it starts giving an error when before I never got this error.

mr-manuel, 2024-08-04:
Additional error logging was added, therefore the error could have been there for a long time.

RickD880, 2024-08-04:
I also have this fault after updating to latest stable release.
Daly BMS with UART-USB connection. 

@4000000066b012be362a2c3c DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012be36e5c794 DEBUG:SerialBattery:Cells:[1]3.355V [2]3.357V [3]3.355V [4]3.354V
@4000000066b012bf1d6425a4 DEBUG:SerialBattery:read_sentence \x91: no sentence start received
@4000000066b012bf1d76907c DEBUG:SerialBattery:request_data: bad reply no. 0
@4000000066b012bf1d805c4c DEBUG:SerialBattery:No data received in read_cell_voltage_range_data()
@4000000066b012bf367a6364 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066b012bf36915e34 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012bf378a50ac DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012bf37bbf79c DEBUG:SerialBattery:Cells:[1]3.356V [2]3.357V [3]3.354V [4]3.354V
@4000000066b012c035642104 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c0367fe58c DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c036e1accc DEBUG:SerialBattery:Cells:[1]3.355V [2]3.357V [3]3.356V [4]3.354V
@4000000066b012c133d2ab94 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c1352db36c DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c135954af4 DEBUG:SerialBattery:Cells:[1]3.354V [2]3.357V [3]3.356V [4]3.354V
@4000000066b012c2327f92fc DEBUG:SerialBattery:read_sentence \x95: no sentence start received
@4000000066b012c2328a87ac DEBUG:SerialBattery:request_data: bad reply no. 0
@4000000066b012c234009d4c ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c2353b72f4 DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c235494604 DEBUG:SerialBattery:Cells:[1]3.354V [2]3.357V [3]3.356V [4]3.354V
@4000000066b012c333962f34 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c3349b75ec DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c334ed00ec DEBUG:SerialBattery:Cells:[1]3.355V [2]3.357V [3]3.355V [4]3.354V
@4000000066b012c4356d8744 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c436c4f53c DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c436e323cc DEBUG:SerialBattery:Cells:[1]3.355V [2]3.357V [3]3.356V [4]3.354V
@4000000066b012c51d62c614 DEBUG:SerialBattery:read_sentence \x91: no sentence start received
@4000000066b012c51d6dda04 DEBUG:SerialBattery:request_data: bad reply no. 0
@4000000066b012c51d77cce4 DEBUG:SerialBattery:No data received in read_cell_voltage_range_data()
@4000000066b012c5368a3244 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066b012c536a10604 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c537e7b6d4 DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c53802208c DEBUG:SerialBattery:Cells:[1]3.356V [2]3.357V [3]3.354V [4]3.354V
@4000000066b012c634d5238c ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c637204c0c DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c637392b3c DEBUG:SerialBattery:Cells:[1]3.355V [2]3.357V [3]3.355V [4]3.354V
@4000000066b012c733d73f74 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c7345fc4ac DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c7347cbabc DEBUG:SerialBattery:Cells:[1]3.356V [2]3.357V [3]3.355V [4]3.354V
@4000000066b012c834d51fa4 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c83576552c DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c835865ea4 DEBUG:SerialBattery:Cells:[1]3.354V [2]3.357V [3]3.355V [4]3.354V
@4000000066b012c933e0e434 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/vi>
@4000000066b012c93510b974 DEBUG:SerialBattery:logged to dbus [95.8]
@4000000066b012c93524743c DEBUG:SerialBattery:Cells:[1]3.354V [2]3.357V [3]3.355V [4]3.354V
@4000000066b012ca32a95664 DEBUG:SerialBattery:read_sentence \x95: no sentence start received

coca7, 2024-08-05:
I think everything is working.  When I revisted my logs I realized I misunderstood the time stamps.  The errors were at times when something lost power on my system.  Please mark as closed

mr-manuel, 2024-08-06:
This behaviour is changed with `v1.4.20240806dev`. Please install it and it will resolve your problem.

mr-manuel, 2024-08-08:
@RickD880 this error tells only, that there are errors in the logs. Please open a new issue and compile all needed informations.

closed 2024-08-08 by mr-manuel