mr-manuel/venus-os_dbus-serialbattery #71: ALARM #119 Settings invalid

seamaster101, 2024-08-25:
### Describe the bug

I run 4X300Ahr 12V batteries with 4X JKBMS in parallel. they are connected to the RPI running Venus OS via 4 X RS485 to USB. 
it is been running fine with no issues with the occasional disconnect of all batteries in about week time. Than was the reason I updated to v1.3.20240705. Immediately after update I started to get this error (bug name). i tryed upgrading to v1.4.20240824dev with same results. 

### How to reproduce

It shows in the Notifications (one for each battery) and they stays active.
[image]

### Expected behavior

Not to have this alarm or make it mean something.

### Driver version of the currently installed driver

v1.4.20240824dev (started with  v1.3.20240705 )

### Driver version of the last known working driver

Not sure which one I had before upgraded to  v1.3.20240705 

### Venus OS device type

Raspberry Pi 3

### Venus OS version

Venus OS v3.50~13

### BMS type

JKBMS (Heltec BMS)

### Cell count

4

### Battery count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
GNU nano 4.9.3               config.ini                Modified
[DEFAULT]

; If you want to add custom values/settings, then check the values>
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the >
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0
;LOGGING = ERROR
MAX_CELL_VOLTAGE   = 3.550
MIN_CELL_VOLTAGE   = 2.95
FLOAT_CELL_VOLTAGE = 3.375
SOC_RESET_VOLTAGE = 3.520
;TEMPERATURES_WHILE_CHARGING      =    0,    2,    5,   10,   15, >
;MAX_CHARGE_CURRENT_T_FRACTION    = 0.00, 0.00, 0.20, 0.40, 0.80, >

;TEMPERATURES_WHILE_DISCHARGING   =  -20,    0,    5,   10,   15, >
;MAX_DISCHARGE_CURRENT_T_FRACTION = 0.00, 0.05, 0.15, 0.40, 1.00, >
BATTERY_CELL_DATA_FORMAT = 3
;TIME_TO_SOC_POINTS = 50,25,0
[... 90 lines cut ...]
2024-08-25 01:45:56.824961500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:45:56.827004500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 01:46:36.170427500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:46:36.201346500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 01:47:14.913722500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:47:14.917250500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
==> /data/log/dbus-serialbattery.ttyUSB4/current <==
2024-08-25 01:44:35.930799500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:44:35.934022500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 01:45:18.794265500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:45:18.797734500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 01:45:58.196705500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:45:58.199787500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 01:46:37.199520500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:46:37.201908500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 01:47:16.069008500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 01:47:16.074697500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 02:12:14.478349500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 02:12:14.484030500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 02:12:37.038189500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_8.pid
==> /data/log/dbus-serialbattery.ttyUSB5/current <==
2024-08-25 03:41:35.233437500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-25 03:42:12.295468500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 03:42:12.298345500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824de
```

mr-manuel, 2024-08-25:
Please provide the logs as described in the docs. This way unfortunately they are useless. You messed up with the commands.

Use this, as in the docs:

```
tail -F -n 100  /data/log/dbus-serialbattery.ttyUSB*/current | tai64nlocal
```

With grep you are truncating all other relevant log lines in this case.

mr-manuel, 2024-08-28:
Anyway, it should be solvable, if you check the docs.

[#119 Settings invalid](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#119-settings-invalid)

closed 2024-08-28 by mr-manuel