mr-manuel/venus-os_dbus-serialbattery #72: Battery Monitor - SerialBattery(EG4 Lifepower) [3] - Error: Internal calculation error

Zeezichtje, 2024-08-26:
### Describe the bug

I am running 6 EG4 batteries in my server rack. One battery (48v) is connected to CERBO CX. Everything worked fine till I migrated from 1.0.20240102beta to latest release "v1.3.20240705".

Version v1.3.20240705 results in: No battery connection at /dev/ttyUSB0

Upgrading to version v1.3.20240615beta results in my battery found. All values seems to be correct. However, Cerbo shows an error: Battery Monitor - SerialBattery(EG4 Lifepower) [3] - Error: Internal calculation error

### How to reproduce

Install  v1.3.20240615beta via the installer and reboot CERBO CX.

### Expected behavior

The battery should immediately be recognized since it was working before.

### Driver version of the currently installed driver

v1.3.20240615beta

### Driver version of the last known working driver

1.0.20240102beta

### Venus OS device type

Cerbo GX

### Venus OS version

v3.41

### BMS type

EG4 LiFePower (Revov, Tian Power)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.
; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 300.0
;MAX_BATTERY_DISCHARGE_CURRENT = 250.0

BMS_TYPE = Lifepower

MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 0.8,  1,  1, 0.6,  0
```

### Relevant log output

```shell
Version v1.3.20240705
tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal 
-> results in no data

tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-08-26 01:15:43.597648500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-26 01:15:43.600042500 INFO:SerialBattery:Venus OS v3.41
2024-08-26 01:15:43.600686500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
2024-08-26 01:15:59.613580500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-26 01:16:00.115161500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-08-26 01:16:00.616705500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-08-26 01:16:01.118550500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0

Version v1.3.20240615beta:
tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal 
-> results in no data

tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-08-26 01:27:28.987658500 INFO:SerialBattery:DeviceInstance = 3
2024-08-26 01:27:28.989598500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_3.pid
2024-08-26 01:27:28.990460500 INFO:SerialBattery:Used DeviceInstances = ['1', '2', '3']
2024-08-26 01:27:28.991289500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-08-26 01:27:29.033328500 INFO:SerialBattery:Hardware Version:B
2024-08-26 01:27:29.100744500 INFO:SerialBattery:Firmware Version:3QT-YS00-16SV100A-V3.40
2024-08-26 01:27:29.132474500 INFO:SerialBattery:publish config values = False
2024-08-26 01:27:29.134800500 INFO:SerialBattery:Battery EG4 Lifepower connected to dbus from /dev/ttyUSB0
2024-08-26 01:27:29.135484500 INFO:SerialBattery:========== Settings ==========
2024-08-26 01:27:29.136404500 INFO:SerialBattery:> Connection voltage: 52.39 V | Current: -10.16 A | SoC: 59.2%
2024-08-26 01:27:29.137016500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-08-26 01:27:29.137639500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-26 01:27:29.138294500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-26 01:27:29.138888500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 300.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-08-26 01:27:29.139927500 INFO:SerialBattery:> CVCM:     True
2024-08-26 01:27:29.140766500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-26 01:27:29.141430500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-26 01:27:29.142024500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-26 01:27:29.142610500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-26 01:27:29.143305500 INFO:SerialBattery:Serial Number/Unique Identifier: B_98.7Ah
```

Zeezichtje, 2024-08-26:
My bad. The error was only visible in the VRM Portal. After deleting the Batterymonitor and rebooting Cerbo CX the error was marked as solved.

mr-manuel, 2024-08-26:
Reading the [changelog](https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/CHANGELOG.md) before updating is always a good idea. There you will find the solution to your problem.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/17458f527d38a835e1d715ba686d71994013c387/CHANGELOG.md?plain=1#L13-L15

closed 2024-08-26 by mr-manuel