mr-manuel/venus-os_dbus-serialbattery #73: Strange behavior with Seplos Battery

KoljaWindeler, 2024-08-26:
### Describe the problem

Hi Mr-Manuel, 
i've added a third battery to my setup yesterday, this time a Seplos.
Everything worked well with serialbattery but there are some odd effects that I can't figure out.

It is very possible that this has nothing to do with serialbattery but who knows:
1. Battery is found and datacommunicatio is working fine 
[image]
2. But the menu at Settings -> System Settings -> Battery measurements  is NOT showing the new battery
[image]
I've update the driver, rebooted the system .. nothing .. ok whatever .. not important 
3. my Homeassistant setup was also NOT receiving any MQTT updates from the new battery. 
... which was strange, because I had a JKBMS Inverter BMS connected to the system before and didn't change the setup since.

dbus-spy is showing all the as victronenergy.battery.* 
[image]

and I'm subscribing to all c06*****1fe4/battery/# topics but I'm not getting any update from battery/3/# as I get it from battery/2/# 

any idea why? likely not connected to this driver or? 
Thanks Kolja

### Driver version of the currently installed driver

v1.3.20240705

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.30

### BMS type

Seplos

### Cell count

16

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
default
```

### Relevant log output

```shell
@4000000066cc69cc29238844 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@4000000066cc69cc292b63fc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000066cc69cc293406ec INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
@4000000066cc69cc293c28f4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A
@4000000066cc69cc29442004 INFO:SerialBattery:> CVCM:     True
@4000000066cc69cc294c1afc INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000066cc69cc29541dc4 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000066cc69cc295bfd64 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@4000000066cc69cc2964179c INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
@4000000066cc69cc296d33a4 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
```

KoljaWindeler, 2024-08-26:
1. nope, but good idea .. will try later today
2. yes :) 
3. yep - no change
4. 2 / 1 / 5 (seplos) I've also check all other devices and they all have different ID's

KoljaWindeler, 2024-08-26:
ok .. strange .. 1. worked .. seplos only ,.. after that also all other batteries worked .. no idea what happened here but works now.
thanks for your idea :)

KoljaWindeler, 2024-08-27:
Actually .. the Battery monitor still doesn't work .. no idea why but I can't get my Seplos battery in the menu.
Everything else is working fine now .. This is really not important for me, it is just strange :D 
I have a JKBMS incoming, will see if that shows up or if there is some kind of limitations to only show the first two or so

KoljaWindeler, 2024-11-26:
hi, I'd like to bring this up again .. as the new gui is also not showing the seplos batteries .. 

to summarize: my setup has 4 jkbms_pb, all connected via one rs485 but and 2 seplos batteries (both on the second rs485 bus).
All batteries show up in the old UI (/gui-v1/) device overview:
[image]

in the old ui I can NOT change the visibility of the seplos batteries in the sub menu -> settings -> system setup -> battery measurements 
[image]
there is no scrolling down any further

I've also tried with only the seplos batteries connected (one) .. same result .. no seplos battery available.
I've also tried with the nightly build .. also no change 

please check the IDs below
[image]

it is not a critical issue because everything important works but it is sad that the batteries are not showing up online and on the new GUI 
[image]

any idea how to debug this? I've checked with dbus-spy and the only "interesting" part was the serial that is "_dev_ttyUSB1__0x01" for the seplos .. is that a possibility?

mr-manuel, 2024-11-26:
Can you write me on Discord so I can check your system?

KoljaWindeler, 2024-11-26:
Sure, let me know how .. I feel old now .. not a discord guy ;)

mr-manuel, 2024-11-26:
You find the link in my profile. Anyway I was able to reproduce it. Somehow.

Checking the code of the QML page this relies on this dbus path: `com.victronenergy.system/AvailableBatteries`

This dbus path is populated by https://github.com/victronenergy/dbus-systemcalc-py/blob/17965bb50e21b2218a26a4768385896caab2983f/delegates/batterydata.py#L276-L304

<details><summary>The data of my testsetup:</summary>

```
root@einstein:~# dbus -y com.victronenergy.system /AvailableBatteries GetValue
('{"com.victronenergy.battery/3": {"name": "SerialBattery(JKBMS PB CAN)", '
 '"channel": null, "type": "battery"}}')

root@einstein:~# dbus -y com.victronenergy.system /AvailableBatteryMeasurements GetValue
{'com_victronenergy_battery_1/Dc/0': 'SerialBattery(JKBMS) on Serial '
                                     '/dev/ttyUSB0',
 'com_victronenergy_battery_3/Dc/0': 'SerialBattery(JKBMS CAN V2) on CAN can0 '
                                     'Device address 0',
 'default': 'Automatic',
 'nobattery': 'No battery monitor'}

root@einstein:~# dbus -y com.victronenergy.system /AvailableBatteryServices GetValue
('{"default": "Automatic", "nobattery": "No battery monitor", '
 '"com.victronenergy.battery/3": "SerialBattery(JKBMS CAN V2) on CAN can0 '
 'Device address 0", "com.victronenergy.battery/1": "SerialBattery(JKBMS) on '
 'Serial /dev/ttyUSB0"}')

root@einstein:~# dbus -y com.victronenergy.system /AvailableBmsServices GetValue
[{'instance': 3, 'name': 'SerialBattery(JKBMS PB CAN)'},
 {'instance': 1, 'name': 'SerialBattery(JKBMS)'}]
 
root@einstein:~# dbus -y com.victronenergy.system /Batteries GetValue
[{'active_battery_service': False,
  'bmsstate': 14,
  'current': 0.0,
  'id': 'com.victronenergy.battery.can0',
  'instance': 3,
  'name': 'SerialBattery(JKBMS PB CAN)',
  'power': 0.0,
  'soc': 87,
  'state': 0,
  'temperature': 1.0,
  'voltage': 13.1},
 {'active_battery_service': True,
  'bmsstate': 9,
  'current': -0.0,
  'id': 'com.victronenergy.battery.ttyUSB0',
  'instance': 1,
  'name': 'SerialBattery(JKBMS)',
  'power': -0.0,
  'soc': 87,
  'state': 0,
  'temperature': 22.0,
  'voltage': 13.17}]
```

</details>

<details><summary>The data of your setup:</summary>

```
dbus -y com.victronenergy.system /AvailableBatteries GetValue
('{"com.victronenergy.vebus/291": {"name": "MultiPlus-II 48/5000/70-50", '
 '"channel": null, "type": "vebus"}, "com.victronenergy.battery/17": {"name": '
 '"SB 4 JKBMS", "channel": null, "type": "battery"}, '
 '"com.victronenergy.battery/18": {"name": "SB 1 JKBMS ", "channel": null, '
 '"type": "battery"}, "com.victronenergy.battery/19": {"name": "SB 6 JKBMS", '
 '"channel": null, "type": "battery"}, "com.victronenergy.battery/20": '
 '{"name": "SB 5 JKBMS", "channel": null, "type": "battery"}, '
 '"com.victronenergy.battery/0": {"name": "AggregateBatteries", "channel": '
[... 85 lines cut ...]
  'current': -1.53,
  'id': 'com.victronenergy.battery.ttyUSB0__0x04',
  'instance': 20,
  'name': 'SB 5 JKBMS',
  'power': -81.02,
  'soc': 88,
  'state': 2,
  'temperature': 18.65,
  'timetogo': 265580,
  'voltage': 53.1},
 {'active_battery_service': True,
  'current': -16.5,
  'id': 'com.victronenergy.battery.aggregate',
  'instance': 0,
  'name': 'AggregateBatteries',
  'power': -875.9575000000001,
  'soc': 86.38324769452421,
  'state': 2,
  'temperature': 19.041666666666668,
  'voltage': 53.08833333333334}]
```

</details>

I have to investigate further...

mr-manuel, 2024-12-05:
While troubleshooting the problem solved itself. Since then we were not able to reproduce it again. I will close this issue again.

closed 2024-12-05 by mr-manuel