mr-manuel/venus-os_dbus-serialbattery #75: Jkbms-pb temparture sensor mapping

cmock, 2024-08-31:
### Describe the bug

The mapping between the labels on the physical temperature sensors and the IDs reported by dbus-serialbattery is wrong.

On the BMS balancer wiring loom (in my case, PB2A16S20P)  the sensor cables are labeled "1" to "4", while in their bluetooth app, they are shown as 1, 2, 4 and 5.

In addition, the reading from the "MOS" sensor is mirrored in "Temperature3".

I have two of these BMSes, they behave identically, so it's not a wiring mixup.

### How to reproduce

Do the "finger test": take each sensor between your fingers and check which value in `dbus-spy` and the JK App goes up.

What I found is the following mapping:

| Physical | JK App | dbus |
|-|-|-|
| 1 | 2 | 2 |
| 2 | 1 | 1 |
| 3 | 5 | - |
| 4 | 4 | 4 |
| MOS | MOS | MOS, 3 |

### Expected behavior

I would like the reported temperatures to match the labels on the cables, and I'd like to see all four of them.

### Driver version of the currently installed driver

v1.4.20240826dev

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.41

### BMS type

JKBMS Inverter

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
MAX_BATTERY_CHARGE_CURRENT    = 140.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0
MIN_CELL_VOLTAGE   = 2.800
MAX_CELL_VOLTAGE   = 3.500
FLOAT_CELL_VOLTAGE = 3.280

MODBUS_ADDRESSES = 0x01, 0x02
BMS_TYPE = Jkbms_pb

AUTO_RESET_SOC = False
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True

CCCM_CV_ENABLE = False
DCCM_CV_ENABLE = False
CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False

CCCM_T_ENABLE = True
;; aus dem LF280K-datenblatt 4.2.1
TEMPERATURES_WHILE_CHARGING =   0,   5,    10,  15,  20,  25,  45,  50,  55,   60
MAX_CHARGE_CURRENT_T_FRACTION = 0.0, 0.12, 0.3, 0.5, 0.8, 1.0, 0.8, 0.5, 0.25, 0

;; datenblatt 4.3.1
DCCM_T_ENABLE = True
TEMPERATURES_WHILE_DISCHARGING   =  -30, -20, -10, -5,  0,   5,   45,  50,  55,  60
MAX_DISCHARGE_CURRENT_T_FRACTION =  0.0, 0.2, 0.4, 0.5, 0.5, 1.0, 1.0, 0.5, 0.5, 0.0
```

### Relevant log output

```shell
-
```

### Any other information that may be helpful

The BMSes are actually connected via a Waveshare isolated RS485/CAN Hat, not via USB.

mr-manuel, 2024-09-23:
Since I got no response [here](https://github.com/Louisvdw/dbus-serialbattery/discussions/969#discussioncomment-10510501) I will make the changes as you proposed.

mr-manuel, 2024-09-23:
Could you please also post a picture of the labeling?

cmock, 2024-09-23:
sure, but I'm only at that installation at the end of the week...

KoljaWindeler, 2024-09-26:
hi, sorry for the delay .. 
i haven't had a JKBMS on hand because I sold one of my batteries .. but now I have another JKBMS.... 
This is what I see in the app:
[image]
and this is what i see at the serial battery
[image]
[image]

I'd say it looks alright, doesn't it?

KoljaWindeler, 2024-09-26:
hmm I've also doublechecked the datasheet:

[image]
[image]

those are the offsets in the message and this is my parsing
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/1446ffff5aa2c249e95af8b1c4b8ba1e835cfd70/etc/dbus-serialbattery/bms/jkbms_pb.py#L190

you have to add on offset of 6 byte to translate between the two adresses ... 156 in the datasheet +6 byte offset (header) = 162 in the  code .. could be wrong in the app :)

mr-manuel, 2024-09-26:
If you check with `dbus-spy` or MQTT, you should see all temperatures, so no guessing :)

KoljaWindeler, 2024-09-26:
jep, dbus-spy showed the same .. 
[image]
looks alright to me .. (not that I care very much about number 1 vs number 2 :D )

hiliuk, 2024-11-04:
I can confirm the similar happening with me:  In dbus the MOSFET temperature is shown as both `System/MOSTemperature` AND as `System/Temperature3`

Therefore one actual temperature sensor data is missing in dbus. The temperature sensor that is missing from dbus is labeled as `Battery T5` in JKBMS app.

Below screenshots are taken at the same time, therefore the temperature readings match between the screenshots.

[image]

[image]

dbus-serialbattery version: v1.4.20240928
JKBMS firmware version: 15.33

KoljaWindeler, 2024-11-09:
ok, I can see that you're right and the datasheet from JK BMS is wrong .. old .. :)

mr-manuel, 2024-11-09:
This was fixed by @KoljaWindeler and is available with `v1.5.20241109dev`.

closed 2024-11-09 by mr-manuel