mr-manuel/venus-os_dbus-serialbattery #77: SOC Stuck at 100%

Honusnap, 2024-09-21:
### Describe the bug

Since Yesterday the displayed SOC stay at 100% while my shunt say otherwise => Soc calculus problem
The oinly thing i see strange in the log is the : 2024-09-21 08:20:21.993289500 INFO:SerialBattery:> Connection voltage: 53.79 V | Current: 7.5 A | SoC: 4% | SoC calc: 100%

SOC 4% ? Hell no it isn't at 4%, wna why SOC calc = 100% when conditions are not met for a 100% ?

### How to reproduce

Nothing really to do .. it just happend without visible condition others then using the driver.

### Expected behavior

SOC decrease at night.

### Driver version of the currently installed driver

1.320240705

### Venus OS device type

Cerbo GX

### Venus OS version

3.30

### BMS type

JKBMS Inverter

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 120.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0

; --------- Cell Voltages ---------
; Description:
; Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage) : 3.450
MAX_CELL_VOLTAGE   = 3.420
; Float voltage (can be seen as resting voltage) : 3.375
FLOAT_CELL_VOLTAGE = 3.400

; --------- SOC reset voltage ---------
[... 437 lines cut ...]
2024-09-21 08:20:09.387835500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E9:0C:E6
2024-09-21 08:20:16.427846500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20230331)
2024-09-21 08:20:16.429309500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-09-21 08:20:20.981006500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-09-21 08:20:21.761797500 INFO:SerialBattery:DeviceInstance = 1
2024-09-21 08:20:21.770063500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-09-21 08:20:21.770865500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-09-21 08:20:21.771789500 INFO:SerialBattery:com.victronenergy.battery.ble_c8478ce90ce6
2024-09-21 08:20:21.773124500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20230331)
2024-09-21 08:20:21.878030500 INFO:SerialBattery:publish config values = True
2024-09-21 08:20:21.991856500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c8478ce90ce6
2024-09-21 08:20:21.992529500 INFO:SerialBattery:========== Settings ==========
2024-09-21 08:20:21.993289500 INFO:SerialBattery:> Connection voltage: 53.79 V | Current: 7.5 A | SoC: 4% | SoC calc: 100%
2024-09-21 08:20:21.993969500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
2024-09-21 08:20:21.994621500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-09-21 08:20:21.995383500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.420 V| FLOAT CELL VOLTAGE: 3.400 V
2024-09-21 08:20:21.996072500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0 A | MAX BATTERY DISCHARGE CURRENT: 150.0 A
2024-09-21 08:20:22.014725500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 140.0 A (read from BMS)
2024-09-21 08:20:22.016129500 INFO:SerialBattery:> CVCM:     False
2024-09-21 08:20:22.017159500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-09-21 08:20:22.017857500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-09-21 08:20:22.036483500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-09-21 08:20:22.044089500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-09-21 08:20:22.044933500 INFO:SerialBattery:Serial Number/Unique Identifier: 2092416547
```

Honusnap, 2024-09-21:
An other thing i see is the "Cell populated" info .. where does it come from.. ? 32 cells  ... ?

mr-manuel, 2024-09-21:
What are your cell voltages? 4% SOC is reported from the BMS.

Honusnap, 2024-09-21:
3.4V cell diff 0.007V
Does the Calculated SOC depend on the SOC reported by the BMS ? Cause the BMS SOC maybe wrong, since i use SerialBattery for along time, the BMS reset 100% maybe have not been met for along time.

mr-manuel, 2024-09-21:
Please check this https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#why-is-my-battery-not-switching-to-floatbulk

Honusnap, 2024-09-21:
> Please check this https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#why-is-my-battery-not-switching-to-floatbulk

My charger is switching from float to bulk and vice versa without any problem.
What part of your link do you think i should look at .. cause i can't see anything that would help to solve my problem here.

mr-manuel, 2024-09-21:
You should enable and check the additional fields on the parameters page. There you find why the SPC os set to 100 %.

Honusnap, 2024-09-21:
> You should enable and check the additional fields on the parameters page. There you find why the SPC os set to 100 %.

"on the parameters page", you mean in the config.ini ?
What is SPC ?

mr-manuel, 2024-09-21:
Just follow the link. SPC is a typo of SOC.

Honusnap, 2024-09-21:
Useless directions, no problem... it's buggy .. will stay this way.

mr-manuel, 2024-09-21:
If you would follow the instructions from the link above and watch the short video, you would see, that there are a lot of useful informations that would help to get this problems sorted. If you don't want to follow the inscrutions, then I can't force you to do so.

[image]

[image]

[image]

[image]

There are over 9.000 installations out there, if only you have a problem it's very likely due to a very special configuration.

mr-manuel, 2024-09-21:
You are also using a `config.default.ini` which does not match your driver version. Please install the latest nightly build, it will be released as the next stable shortly.

mr-manuel, 2024-09-21:
Also, it the calculated SOC is reset by the driver, you should see something like this:

```
2024-09-02 20:30:28.265473500 INFO:SerialBattery:Starting dbus-serialbattery
2024-09-02 20:30:28.267335500 INFO:SerialBattery:Venus OS v3.50~13
2024-09-02 20:30:28.267789500 INFO:SerialBattery:dbus-serialbattery v1.4.20240902dev
2024-09-02 20:30:44.282126500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-09-02 20:30:44.283154500 INFO:SerialBattery:Testing Jkbms
2024-09-02 20:30:44.354394500 INFO:SerialBattery:Connection established to Jkbms
2024-09-02 20:30:44.536180500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-09-02 20:30:44.883293500 INFO:SerialBattery:DeviceInstance = 1
2024-09-02 20:30:44.893239500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-09-02 20:30:44.894224500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-09-02 20:30:44.894737500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0__0
2024-09-02 20:30:44.922789500 INFO:SerialBattery:publish config values = False
2024-09-02 20:30:44.928203500 INFO:SerialBattery:Polling data every 1.000 s
2024-09-02 20:30:44.931091500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-09-02 20:30:44.932634500 INFO:SerialBattery:========== Settings ==========
2024-09-02 20:30:44.933836500 INFO:SerialBattery:> Connection voltage: 59.92 V | Current: 0.58 A | SoC: 76% | SoC calc: 81%
2024-09-02 20:30:44.934826500 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
2024-09-02 20:30:44.935750500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-09-02 20:30:44.937031500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.500 V| FLOAT CELL VOLTAGE: 3.410 V
2024-09-02 20:30:44.938331500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0 A | MAX BATTERY DISCHARGE CURRENT: 120.0 A
2024-09-02 20:30:44.939539500 INFO:SerialBattery:> CVCM:     True
2024-09-02 20:30:44.940855500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-09-02 20:30:44.942001500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-09-02 20:30:44.943074500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-09-02 20:30:44.944424500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-09-02 20:30:44.945625500 INFO:SerialBattery:Serial Number/Unique Identifier: JK-B2A24S15P
2024-09-03 11:59:16.915916500 INFO:SerialBattery:SOC set to 100%
2024-09-04 12:12:21.992282500 INFO:SerialBattery:SOC set to 100%
2024-09-07 14:42:58.116679500 INFO:SerialBattery:SOC set to 100%
```

closed 2024-09-28 by mr-manuel