mr-manuel/venus-os_dbus-serialbattery #79: Weekly connection lost to all four JBD BMS

regu-tech, 2024-09-26:
### Describe the bug

Hello,

i have the problem with the driver, that about one time in the week the connection to all of my
4 BMS (Jiabaida AP20S003s 20S 300A) gets lost. 

[current.log](https://github.com/user-attachments/files/17155269/current.log)

I don't know how to reproduce the error, it appears every week one time. So maybe it depends on the
errors in the log 

`ERROR: Invalid response packet. Expected begin packet character 0xDD
@4000000066f533410ab756ec WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29
@4000000066f53620099f091c ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@4000000066f5362009b223bc WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29
@4000000066f542541e4aff4c ERROR:SerialBattery:>>> ERROR: No reply - returning [len:32/77]
@4000000066f542541f6719c4 WARNING:SerialBattery:Polling took too long. Set to 17.000 s
@4000000066f54715116bdc9c ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@4000000066f54715117db6ec WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 29`

Maybe this would be helpful, if other informations are needed, i can also provide them.

best regards
Steffen

### How to reproduce

Can't reproduce it, it comes every week one or two times.

### Expected behavior

[image]

### Driver version of the currently installed driver

v1.4.20240902dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.42

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

16

### Battery count

4

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
[... 448 lines cut ...]
; Use min/max cell voltage, CVL, CCL and DCL from the BMS
SEPLOS_USE_BMS_VALUES = False

; --------- Voltage drop ---------
; If you have a voltage drop between the BMS and the charger because of wire size or length
; then you can specify the voltage drop here. The driver will then add the voltage drop
; to the calculated CVL to compensate.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 V and the battery is now charged until the charger reaches 55.20 V.
;     The BMS now measures 55.05 V since there is a voltage drop of 0.15 V on the cable.
;     Since the dbus-serialbattery reads the voltage of 55.05 V from the BMS the max voltage
;     of 55.20 V is never reached and max voltage is kept forever.
;     By setting the VOLTAGE_DROP to 0.15 V the voltage on the charger is increased and the
;     target voltage on the BMS is reached.
VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
see below
```

mr-manuel, 2024-09-27:
Unfortunately you did not read carefully the instructions when opening an issue and the logs are in the wrong format.

Anyway, This seems to be an issue of cabeling interference (is your cable shielded and not near the battery power cable?), a cheap USB hub or a faulty BMS. Please provide correct logs to see the timings beween the messages.

regu-tech, 2024-09-27:
Hello, sorry for the wrong log format. I have added now the converted log in my first post.

The usb cables are in parallel to the batterie cables (they lie on them) for approx. 40cm, could this be a problem? 

best regards
Steffen

mr-manuel, 2024-09-27:
The USB cables should be shielded, if you did not get very cheap ones. That could be a problem. Try to use a spare USB cable and check, if the same error occurs.

What about the serial cables from the USB to UART/TTL adapters to the BMS? Sometimes also the serial cables are too short (< 20 cm), which causes strange issues. Use shielded not to short (> 30 cm) serial cables which are not in the near of the battery power cables.

The only way to hopefully solve this is to change, try and check the cabeling. In this case it's not a driver fault.

regu-tech, 2024-09-28:
The USB cables are not new, so i don't know how good the are. I have ordered 4 new ones, double shieldet for 9€ each with 2m length and a ferrit core. I will also put them away from the battery cables.

The serial cables from USB to UART are approx. 40cm long, but they lie also for 20cm parallel to the balancer wires. I put them also away from them and hope that the connection losts are fixed then. 

Thanks for your help, it is good to know that we can exclude an error on driver side. 

best regards
Steffen

mr-manuel, 2024-10-06:
You do net mention, if your serial cable is shielded or not. On one side the shield has also to be grounded.

closed 2024-10-11 by mr-manuel