mr-manuel/venus-os_dbus-serialbattery #81: 1.4+ version does no longer support Seplos V3?

endurance1968, 2024-09-29:
### Describe the problem

I tried 1.4 and 1.5 stream today and got

```
2024-09-29 05:39:50.324117500 INFO:SerialBattery:Testing Seplos at address "\x00"
2024-09-29 05:39:50.325701500 DEBUG:SerialBattery:read status data
2024-09-29 05:39:50.327135500 DEBUG:SerialBattery:read serial data seplos
2024-09-29 05:39:50.331719500 DEBUG:SerialBattery:wrote 0 bytes to serial port /dev/ttyUSB0, command=b''
2024-09-29 05:39:51.333615500 DEBUG:SerialBattery:short read, data=b''
2024-09-29 05:39:51.340557500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-09-29 05:39:51.341277500 INFO:SerialBattery:Testing Seplosv3
2024-09-29 05:39:51.352055500 INFO:SerialBattery:|- on slave address 0
2024-09-29 05:39:51.772212500 DEBUG:SerialBattery:Seplos v3 testing failed (No communication with the instrument (no answer)) 1/3 for /dev/ttyUSB0(0)
2024-09-29 05:39:52.204039500 DEBUG:SerialBattery:Seplos v3 testing failed (No communication with the instrument (no answer)) 2/3 for /dev/ttyUSB0(0)
2024-09-29 05:39:52.214962500 INFO:SerialBattery:|- on slave address 1
```

switched back to 1.3beta there it works

```
2024-09-29 05:45:38.863734500 INFO:SerialBattery:Venus OS v3.42
2024-09-29 05:45:38.864965500 INFO:SerialBattery:dbus-serialbattery v1.3.20240615beta
2024-09-29 05:45:54.881620500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-09-29 05:45:54.881629500 INFO:SerialBattery:Testing Seplos
2024-09-29 05:45:54.881632500 DEBUG:SerialBattery:read status data
2024-09-29 05:45:54.881634500 DEBUG:SerialBattery:read serial data seplos
2024-09-29 05:45:54.899142500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB0, command=b'~20004642E00201FD36\r'
2024-09-29 05:45:56.044030500 DEBUG:SerialBattery:returning info data of length 150, info_length is 150 : b'0000100CD60CD60CD60CD60CD60CD60CD60CD70CD60CD70CD60CD80CD70CD70CD60CD8060B630B630B630B620B880B60004B148A3FD90A772402177724004303E8148BFFF900002487235A'
2024-09-29 05:45:56.047975500 DEBUG:SerialBattery:Temp cell1=18.4°C
2024-09-29 05:45:56.049322500 DEBUG:SerialBattery:Temp cell2=18.4°C
2024-09-29 05:45:56.050624500 DEBUG:SerialBattery:Temp cell3=18.4°C
2024-09-29 05:45:56.051848500 DEBUG:SerialBattery:Temp cell4=18.3°C
```

### Driver version of the currently installed driver

v1.4.20240928

### Driver version of the last known working driver

v1.3.20240615beta

### Venus OS device type

Cerbo GX

### Venus OS version

3.42

### BMS type

Seplos v3

### Cell count

16

### Battery count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

LOGGING = DEBUG

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.45
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.4

BMS_TYPE = Seplos

SOC_LOW_WARNING = 15
SOC_LOW_ALARM = 10

POLL_INTERVAL = 10
```

### Relevant log output

```shell
2024-09-29 05:39:50.324117500 INFO:SerialBattery:Testing Seplos at address "\x00"
2024-09-29 05:39:50.325701500 DEBUG:SerialBattery:read status data
2024-09-29 05:39:50.327135500 DEBUG:SerialBattery:read serial data seplos
2024-09-29 05:39:50.331719500 DEBUG:SerialBattery:wrote 0 bytes to serial port /dev/ttyUSB0, command=b''
2024-09-29 05:39:51.333615500 DEBUG:SerialBattery:short read, data=b''
2024-09-29 05:39:51.340557500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-09-29 05:39:51.341277500 INFO:SerialBattery:Testing Seplosv3
2024-09-29 05:39:51.352055500 INFO:SerialBattery:|- on slave address 0
2024-09-29 05:39:51.772212500 DEBUG:SerialBattery:Seplos v3 testing failed (No communication with the instrument (no answer)) 1/3 for /dev/ttyUSB0(0)
2024-09-29 05:39:52.204039500 DEBUG:SerialBattery:Seplos v3 testing failed (No communication with the instrument (no answer)) 2/3 for /dev/ttyUSB0(0)
2024-09-29 05:39:52.214962500 INFO:SerialBattery:|- on slave address 1
```
```

endurance1968, 2024-09-29:
Maybe I need to change config for new drivers?

mr-manuel, 2024-09-29:
Hello, the driver should still work. Probably no one tested the latest version in the last month. The issue is probably because of the optimizations, since before the Seplos driver had some useless checks.

Can you write me on Discord? So I could connect to your system and fix.

mr-manuel, 2024-10-05:
Regression was introduced with https://github.com/mr-manuel/venus-os_dbus-serialbattery/commit/38ebf1b8028a9787c2112ccabb5034dba8aefe80. I republish the latest stable with this fix.

commit 10236a3e (2024-10-05): Seplos BMS - Fix #81 regression
https://github.com/mr-manuel/venus-os_dbus-serialbattery/pull/81
closed 2024-10-05 by mr-manuel