mr-manuel/venus-os_dbus-serialbattery #82: Persistent "Low Battery" alarm

josmeijer, 2024-09-30:
### Describe the problem

I have a persistent and repeating "low battery"alarm. I have no clue where it comes from, as I see no registration of any voltige even near the warning and alarm tresholds in the grahic logs of both D-bus serial and the Victron Easysolar voltages
[image]
[image]

### Driver version of the currently installed driver

1.3.20240624

### Venus OS device type

CCGX

### Venus OS version

v3.40~39

### BMS type

Daly BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
POLL_INTERVAL = 2
INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal

^C
root@ccgx:~# tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
^C
root@ccgx:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-09-30 00:56:40.691286500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 00:57:10.738374500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 00:59:56.909822500 INFO:SerialBattery:refresh_data: result: False
2024-09-30 01:03:41.219881500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:11:05.630800500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:11:39.663606500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:15:05.878175500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:21:30.274263500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:21:40.211244500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:22:00.289980500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:27:06.640169500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 01:35:37.125276500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
[... 64 lines cut ...]
2024-09-30 04:16:43.691034500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:19:43.918421500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:30:30.617548500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:33:58.923501500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:35:33.001704500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:46:35.733912500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:47:09.748469500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 04:49:03.910487500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:00:50.693232500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:31:24.657466500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:32:02.708461500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:32:46.752345500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:33:46.913844500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:34:20.909877500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:34:46.923915500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:49:23.921412500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 05:53:10.699916500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.5s
2024-09-30 05:53:32.999018500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.4s
2024-09-30 05:53:40.830226500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.3s
2024-09-30 05:53:56.736048500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.3s
2024-09-30 05:55:10.078212500 INFO:SerialBattery:  |- refresh_data: read_soc_data - result: True - runtime: 0.2s
2024-09-30 05:55:11.319667500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.3s
2024-09-30 05:55:56.467463500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
2024-09-30 06:16:19.680903500 INFO:SerialBattery:  |- refresh_data: read_cells_volts - result: True - runtime: 0.2s
```

mr-manuel, 2024-09-30:
Please post also a screenshot of the alam message itself.

mr-manuel, 2024-09-30:
As you can see in the screenshot you added above, the alarm is not coming from the battery driver, but from the EasySolar. Check the config of your device with the VE.Config software. For further help please ask the Victron Community.

closed 2024-09-30 by mr-manuel

josmeijer, 2024-09-30:
> Closed [#82](https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/82) as completed.

I assumed that the Easysolar gets its voltage information from the BMS (via Dbus) . Correctly, but I overlooked the fact that ESS is working with different treshold voltages (and the "ESS bot" defaulted to a rather high cut-off voltage)
I will discuss this indeed in the Victron community

Verzonden met [Proton Mail](https://proton.me/mail/home) beveiligde e-mail.