mr-manuel/venus-os_dbus-serialbattery #93: Daly WNT issue? Batteries in Parallel not displaying in Venus OS

NickRoberts174, 2024-10-16:
### Describe the bug

My Equipment
Victron Cerbo GX
Victron MPPT 250|100 Solar Charge Controller
Victron Multiplus-II 48/5000/70
Battery 1 - 280AH x18s LiFePo4, Daly 150A BMS, Parallel protection board, 1A active Balancer and WNT interface board
Battery 2 - 280AH x18s LiFePo4, Daly 150A BMS, Parallel protection board, 1A active Balancer and WNT interface board
The Batteries are connected in series.

dbus serial battery driver version 1.4.20240902dev (nightly release)

The Daly WNT board supports RS485, RS232 and CAN 500kbps connections, The WNT boards are addressed via dip switches. 

The issue I have is when connecting both batteries to the dbus serial driver (via RS485) at once. I have 2x USB to serial converters connected to the cerbo (https://www.amazon.co.uk/dp/B081NBCJRS?ref=ppx_yo2ov_dt_b_fed_asin_title) , they both work individually, but if you plug both in at the same time, one battery will show up, while the other does not. The one which shows up sometimes goes blank and then changes to display the other battery, this often happens after a reboot.

Despite many different configurations and conditions, The driver does not display both batteries at the same time, which is what I am hoping to achieve.

Solutions I have already tried

- Alternative settings of the dip switches (all off, swapped addresses, set the same etc) most reliable setting is to have both WNT boards to be set with dip switch 1 ON.
- I have tried each RS485-USB device separately, both work individually and read correctly, but not together.
- I have tried one RS485 adaptor with both batteries connected
- I have tried one RS485 adaptor with the second battery daisy chained on the spare RS485 (this shows both batteries figures but they alternate one battery entry on venus)
- I have tried 2 RS485 adaptors connected into a USB hub and then into one USB hub on the Cerbo GX

### How to reproduce

Attempt to connect Daly WNT boards to cerbo GX using RS485 - USB adaptor and this Driver in any of these formats; 

[image]
[image]
[image]
[image]
[image]
[image]

### Expected behavior

Expect 2 separate entries for each battery. Sometimes on reboots, there is a second entry which shows 'disconnected' it will never reconnect, on a reboot, it may work, but then the other battery will not. the data displayed is not locked to an individual battery either, it appears to be random for which data it will show. 

[image]
[image]

### Driver version of the currently installed driver

v1.4.20240928

### Venus OS device type

Cerbo GX

### Venus OS version

v3.42

### BMS type

Daly BMS

### Cell count

2x 18s (in parallel) 

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 75.0
; MAX_BATTERY_DISCHARGE_CURRENT = 150.0
```

### Relevant log output

```shell
Logs attached in additional info (maximum is 65536 characters exceeded here)
```

### Any other information that may be helpful

[Logs (combined).pdf](https://github.com/user-attachments/files/17402465/Logs.combined.pdf)

mr-manuel, 2024-10-18:
As the error message in the log states, you have two batteries with the same identical identifier. You need to change that identifier with the Daly PC software.

NickRoberts174, 2024-10-18:
They are addressed differently in the Daly PC software - using the PC master tool, under the engineering settings Battery one is addressed Board 1, battery 2 is addressed Board 2. I have tried changing the slave numbers too but it doesn't seem to have any effect? 

[image]

Is there another way of giving them unique addresses? I have also previously tried setting line 449 of the config.default.ini  file to TRUE, hoping I could use the USB as the unique identifier and had no luck there.

[image]

mr-manuel, 2024-10-18:
Please check the documentation: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/next/general/connect/

mr-manuel, 2024-10-18:
Would be great, if you could provide a screenshot of the whole window of this software, so I can update it in the documentation.

closed 2024-10-30 by mr-manuel