mr-manuel/venus-os_dbus-serialbattery #94: JBD UP series

Rubens974, 2024-10-19:
### Describe the problem

Hi,
@4000000067132ad70855b09c INFO:SerialBattery:Testing LltJbd
@4000000067132ad719a5355c ERROR:SerialBattery:>>> ERROR: No reply - returning
My BMS JBD UP series is not recognized.
Have I missed something?

Regards
Rubens

### Driver version of the currently installed driver

v1.420240923

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.303

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

15

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
BMS_TYPE =  LltJbd
; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
@40000000671328c1161c9754 ERROR:SerialBattery:Exception occurred: SerialException(2, "could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'") of type <class 'serial.serialutil.SerialException'> in /opt/victronenergy/dbus-serialbattery/bms/seplos.py line #108
@40000000671328c116e67704 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000671328c116e68a8c INFO:SerialBattery:Testing Seplosv3
@40000000671328c116e69644 ERROR:SerialBattery:Non blocking exception occurred: SerialException(2, "could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'") of type <class 'serial.serialutil.SerialException'> in /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py line #168
@40000000671328c134336934 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
@40000000671328c20835b904 /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (23944) - No such process
^C
root@raspberrypi4:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current
@40000000671328c1161c9754 ERROR:SerialBattery:Exception occurred: SerialException(2, "could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'") of type <class 'serial.serialutil.SerialException'> in /opt/victronenergy/dbus-serialbattery/bms/seplos.py line #108
@40000000671328c116e67704 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000671328c116e68a8c INFO:SerialBattery:Testing Seplosv3
@40000000671328c116e69644 ERROR:SerialBattery:Non blocking exception occurred: SerialException(2, "could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'") of type <class 'serial.serialutil.SerialException'> in /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py line #168
@40000000671328c134336934 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
@40000000671328c20835b904 /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (23944) - No such process
```

Rubens974, 2024-10-19:
With Venus 3.40 and dbus-serialbattery v1.5.20241016dev

@4000000067132b0e17e028e4 INFO:SerialBattery:Starting dbus-serialbattery
@4000000067132b0e17f70c44 INFO:SerialBattery:Venus OS v3.40
@4000000067132b0e18017454 INFO:SerialBattery:dbus-serialbattery v1.5.20241016dev
@4000000067132b1e18ec2a44 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@4000000067132b1e18ec459c INFO:SerialBattery:Testing Daly at address "\x40"
@4000000067132b1e213cacb4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b1e2143b194 INFO:SerialBattery:Testing Daly at address "\x80"
@4000000067132b1e29644294 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b1e296b0124 INFO:SerialBattery:Testing Daren485 at address "\x01"
@4000000067132b1f06a8fa9c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b1f06aef5dc INFO:SerialBattery:Testing Ecs
@4000000067132b1f0ad2030c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b1f0adf64d4 INFO:SerialBattery:Testing EG4_Lifepower at address "\x01"
@4000000067132b1f1c4ee514 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b1f1c5891a4 INFO:SerialBattery:Testing EG4_LL at address "\x01"
@4000000067132b1f2d1ada24 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b1f2d269ddc INFO:SerialBattery:Testing HeltecModbus at address "\x01"
@4000000067132b231f2c4efc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b231f93bb8c INFO:SerialBattery:Testing HLPdataBMS4S
@4000000067132b2c20a11894 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2c20a9c354 INFO:SerialBattery:Testing Jkbms
@4000000067132b2c335aef3c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2c33923b7c INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
@4000000067132b2d0ab3383c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2d0af5bf7c INFO:SerialBattery:Testing LltJbd
@4000000067132b2d1e3ca76c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2d1e7a6804 INFO:SerialBattery:Testing Renogy at address "\x30"
@4000000067132b2d30d56d3c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2d312467fc INFO:SerialBattery:Testing Renogy at address "\xf7"
@4000000067132b2e06a9b234 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2e06ddd1f4 INFO:SerialBattery:Testing Seplos at address "\x00"
@4000000067132b2f08176f1c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067132b2f082935e4 INFO:SerialBattery:Testing Seplosv3
@4000000067132b2f091a2594 INFO:SerialBattery:|- on slave address 0
@4000000067132b2f3b604394 INFO:SerialBattery:|- on slave address 1
@4000000067132b303217f78c INFO:SerialBattery:|- on slave address 2
@4000000067132b3128af5a14 INFO:SerialBattery:|- on slave address 3
@4000000067132b321f6bec24 INFO:SerialBattery:|- on slave address 4
@4000000067132b3316bd5f7c INFO:SerialBattery:|- on slave address 5
@4000000067132b340d3a2584 INFO:SerialBattery:|- on slave address 6
@4000000067132b3503bb779c INFO:SerialBattery:|- on slave address 7
@4000000067132b35368ac6b4 INFO:SerialBattery:|- on slave address 8
@4000000067132b362d9211ac INFO:SerialBattery:|- on slave address 9
@4000000067132b37254cca3c INFO:SerialBattery:|- on slave address 10
@4000000067132b381d7be3c4 INFO:SerialBattery:|- on slave address 11
@4000000067132b39150fd68c INFO:SerialBattery:|- on slave address 12
@4000000067132b3a0bfd5bdc INFO:SerialBattery:|- on slave address 13
@4000000067132b3b033363ac INFO:SerialBattery:|- on slave address 14
@4000000067132b3b36034734 INFO:SerialBattery:|- on slave address 15
@4000000067132b3c2c18eae4 ERROR:SerialBattery:>>> No reply - returning
@4000000067132b3d0e6ed10c INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds

Rubens974, 2024-10-19:
My BMS:
JBD-UP16S010-L16S-200A-B-R-C

mr-manuel, 2024-10-19:
From your logs you have an unstable USB connection or a faulty USB to serial adapter.

Rubens974, 2024-10-19:
Thank you.

I've changed into a new USB to serial adapter.

root@raspberrypi4:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC**
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I just want to make sure that my JBD-UP16S010-L16S-200A-B-R-C is in the list of compatible BMSs.

And here is my log after a reboot.

@400000006713cab235c88054 *** CCGX booted (0) ***
@400000006713cad029eafec4 INFO:SerialBattery:
@400000006713cad029ef09ec INFO:SerialBattery:Starting dbus-serialbattery
@400000006713cad029ff9c1c INFO:SerialBattery:Venus OS v3.40
@400000006713cad02a047264 INFO:SerialBattery:dbus-serialbattery v1.5.20241016dev
@400000006713caf608882694 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006713caf6088845d4 INFO:SerialBattery:Testing Daly at address "\x40"
@400000006713caf611b36904 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713caf611ba46d4 INFO:SerialBattery:Testing Daly at address "\x80"
@400000006713caf61b323a14 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713caf61b37a0e4 INFO:SerialBattery:Testing Daren485 at address "\x01"
@400000006713caf634a276e4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713caf634a28a6c INFO:SerialBattery:Testing Ecs
@400000006713caf638d2a374 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713caf638d2becc INFO:SerialBattery:Testing EG4_Lifepower at address "\x01"
@400000006713caf70e7d7af4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713caf70e7d8e7c INFO:SerialBattery:Testing EG4_LL at address "\x01"
@400000006713caf71f47493c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713caf71f52555c INFO:SerialBattery:Testing HeltecModbus at address "\x01"
@400000006713cafb1148ef34 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cafb114d7b44 INFO:SerialBattery:Testing HLPdataBMS4S
@400000006713cb04127234c4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb041277fd3c INFO:SerialBattery:Testing Jkbms
@400000006713cb0423dc59ec ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb04241b3b94 INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
@400000006713cb0435d570d4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb04361592cc INFO:SerialBattery:Testing LltJbd
@400000006713cb050c67607c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb050cc088b4 INFO:SerialBattery:Testing Renogy at address "\x30"
@400000006713cb051e428754 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb051ea494e4 INFO:SerialBattery:Testing Renogy at address "\xf7"
@400000006713cb053106e164 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb0531500bdc INFO:SerialBattery:Testing Seplos at address "\x00"
@400000006713cb06324b9e34 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb06325352dc INFO:SerialBattery:Testing Seplosv3
@400000006713cb0633603e9c INFO:SerialBattery:|- on slave address 0
@400000006713cb072a34199c INFO:SerialBattery:|- on slave address 1
@400000006713cb082197effc INFO:SerialBattery:|- on slave address 2
@400000006713cb0918222f3c INFO:SerialBattery:|- on slave address 3
@400000006713cb0a0f0a3a34 INFO:SerialBattery:|- on slave address 4
@400000006713cb0b064e9214 INFO:SerialBattery:|- on slave address 5
@400000006713cb0b38d15b54 INFO:SerialBattery:|- on slave address 6
@400000006713cb0c30035fcc INFO:SerialBattery:|- on slave address 7
@400000006713cb0d27528164 INFO:SerialBattery:|- on slave address 8
@400000006713cb0e1e78e934 INFO:SerialBattery:|- on slave address 9
@400000006713cb0f15be0c34 INFO:SerialBattery:|- on slave address 10
@400000006713cb100d1e1204 INFO:SerialBattery:|- on slave address 11
@400000006713cb110446815c INFO:SerialBattery:|- on slave address 12
@400000006713cb1136e6dcec INFO:SerialBattery:|- on slave address 13
@400000006713cb122e09dda4 INFO:SerialBattery:|- on slave address 14
@400000006713cb132510848c INFO:SerialBattery:|- on slave address 15
@400000006713cb141b098fec ERROR:SerialBattery:>>> No reply - returning
@400000006713cb1438e8dedc INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@400000006713cb1438f00acc INFO:SerialBattery:Testing Daly at address "\x40"
@400000006713cb15059b2c4c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cb1505a1f694 INFO:SerialBattery:Testing Daly at address "\x80"
[... 212 lines cut ...]
@400000006713cbfc09034dec INFO:SerialBattery:Testing Renogy at address "\x30"
@400000006713cbfc1adcf9dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cbfc1b031b94 INFO:SerialBattery:Testing Renogy at address "\xf7"
@400000006713cbfc2c9569fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cbfc2cbd58a4 INFO:SerialBattery:Testing Seplos at address "\x00"
@400000006713cbfd2d6f958c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006713cbfd2d6facfc INFO:SerialBattery:Testing Seplosv3
@400000006713cbfd2ea36654 INFO:SerialBattery:|- on slave address 0
@400000006713cbfe2708f544 INFO:SerialBattery:|- on slave address 1
@400000006713cbff1f196724 INFO:SerialBattery:|- on slave address 2
@400000006713cc0015f2186c INFO:SerialBattery:|- on slave address 3
@400000006713cc010d05465c INFO:SerialBattery:|- on slave address 4
@400000006713cc02038490ec INFO:SerialBattery:|- on slave address 5
[... cut ...]
@400000006713cc070b0a9f3c INFO:SerialBattery:|- on slave address 11
@400000006713cc0801c623c4 INFO:SerialBattery:|- on slave address 12
@400000006713cc083440d2cc INFO:SerialBattery:|- on slave address 13
@400000006713cc092adb002c INFO:SerialBattery:|- on slave address 14
@400000006713cc0a21c8eb0c INFO:SerialBattery:|- on slave address 15
@400000006713cc0b17ca436c ERROR:SerialBattery:>>> No reply - returning
@400000006713cc0b35d5f98c ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0

mr-manuel, 2024-10-20:
Unfortunately there is no BMS model compatibily list. It could be that your cabeling is wrong or your BMS is not supported.

mr-manuel, 2024-11-05:
Ask your seller for the protocol documentation. This allows me to compare it with the existing driver and see if this should work or not.