mr-manuel/venus-os_dbus-serialbattery #100: Remove round function for soc_calc_capacity_remain

jojo18222, 2024-10-30:
### Describe the bug

In File battery.py Line 470 the calculatated soc_calc_capacity_remain is rounded to 3 decimal places. This should removed because small current values have no affect and so the soc_calc_capacity_remain and Soc value doesn't change.

### How to reproduce

SOC_CALCULATION   = True
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 0, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, -0.1, 300
Low Soc is reached, BMS current is 0A, calculatated Soc stays forever at the low Soc value.

### Expected behavior

Soc should slowly reduced.

### Driver version of the currently installed driver

1.4.20240902dev

### Venus OS device type

Raspberry Pi 4

### Venus OS version

3.50

### BMS type

Daly BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
SOC_CALCULATION   = True
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 0, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, -0.1, 300
```

### Relevant log output

```shell
-
```

mr-manuel, 2024-10-31:
Currents smaller than three decimals are so low that your measurement reading is probably not accurate enough to measure it. Even a SmartShunt has a maximum accuracy of 0.1 A. I don't think that a Daly BMS is that accurate. Probably your correction table should be updated.

jojo18222, 2024-10-31:
Correct, the Daly BMS current starts at 0.4A I think. 
I have connected Venus os raspi with a DC/DC converter directly to the battery. The clamp meter shows round about 0.1A.
So in the correction table I set if BMS shows 0A, the real value is -0.1A.
With an older version of serial driver, the Soc was reduced slowly if the multiplus was in idle, for example at low Soc situation.
Now, with the rounded value, Soc stays forever at a Soc value and I think this is not how it should work.

mr-manuel, 2024-11-04:
Which older version? I don‘t think that this is the issue. You should calibrate the current measurement on your Daly BMS, if this is possible with Daly. I tested with a simulated current of 0.01 A and it worked correctly. The current measurement for that current is simply too small.

jojo18222, 2024-11-04:
Here is the round function not implemented:
https://github.com/Louisvdw/dbus-serialbattery/blob/fd0fc1a193427df6ca0fadd141aa57826f761819/etc/dbus-serialbattery/battery.py#L280
Ok, I don't have further arguments. I can't calibrate the daly bms, because the current of the raspi is to low (5 Watts).
And if soc_calc_capacity_remain is rounded and the Multiplus 2 is in idle or off, the soc never get lower although the raspi consumes 5 Watts. After some days or longer, the real soc is way lower then the calculated soc.
I don't understand the advantages of the soc_calc_capacity_remain is rounded, because a line before, the soc_calc_capacity_remain is calculated without rounding and later, the soc value is rounded with 2 digits anyway.

mr-manuel, 2024-11-04:
Could you try to change the rounding to 6 decimals? Some values are rounded, to filter out stupid values like `45.0000000000001` or `29.99999999999998` that happen sometimes.

mr-manuel, 2024-11-04:
Now I had more time to check the code. Indeed it does not make any sense. It was rounded since the first PR. Currently I have a bigger change in the pipeline, therefore it will take some time, but the rounding will be removed in the next version. Until then I leave the issue open.

closed 2024-11-06 by mr-manuel

jojo18222, 2024-11-07:
Hey, thanks for the change. Sorry to bother you again. You also changed the digital places for soc_calc from 2 to 3.
But this has no effect because in dbushelper.py soc_calc is rounded back to 2 decimal places.
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/81d743ba207bd4fdfc696707412f11bdab33d3b8/etc/dbus-serialbattery/dbushelper.py#L816

mr-manuel, 2024-11-07:
Yes I changed that also. This is only relevant for a more pricise saving of the calculated SOC to the dbus. Useful for batteries with >1.000 Ah.

commit e3283373 (2024-11-06): Fixes #100
https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/100
commit 81d743ba (2024-11-06): Fixes #100
https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/100