mr-manuel/venus-os_dbus-serialbattery #110: DeviceInstance 2 instead of 1

zigomatichub, 2024-11-29:
### Describe the bug

After upgrading from v1.2.20240408 to v1.5.20241119beta, I saw a change from instance 1 to instance 2. I found this becasue data was not sent to the same topic. Previously, it was sent to N/blabla/battery/1 and now N/blabla/battery/2.

But from now, I have only one battery and one instance. I checked release changelog and I don't see any change but I also don't find why it's moved from instance 1 to 2 and how to solve this. 

**Some steps already done but not helping to resolve the issue**
- redo the config based on default to get the new parameters of the new version.
  - Discard in config.ini the /dev/ttyUSB0 used by MK3 USB connection
- disable driver ( with command from documentation).
- remove the device from the victron UI.

             --- reboot OS ---

Re-install ( with command from documentation).

Nothing in the victron UI.
In dbus-serial, I only see *** CCGX booted (0) ***
Based on timing, it's not starting and seems used by dbus-cgwacs
```
root@venus:/data/log# ls -ltr | grep USB
drwx------    2 root     root          4096 Nov 23 22:54 mk2_dbus.ttyUSB0
drwx------    2 root     root          4096 Nov 23 22:54 gps-dbus.ttyUSB1
drwx------    2 root     root          4096 Nov 23 22:55 vedirect.ttyUSB1
drwx------    2 root     root          4096 Nov 23 22:55 dbus-modbus-client.ttyUSB1
drwx------    2 root     root          4096 Nov 23 22:55 dbus-imt-si-rs485tc.ttyUSB1
drwx------    2 root     root          4096 Nov 29 08:12 dbus-serialbattery.ttyUSB1
drwx------    2 root     root          4096 Nov 29 10:11 dbus-fzsonick-48tl.ttyUSB1
drwx------    2 root     root          4096 Nov 29 10:11 dbus-cgwacs.ttyUSB1
```

             --- reboot OS ---

Started successfuly but still show in logs Reconnected to previously identified battery with DeviceInstance: 2
```
*** CCGX booted (0) ***
INFO:SerialBattery:
INFO:SerialBattery:Starting dbus-serialbattery
INFO:SerialBattery:Venus OS v3.52
INFO:SerialBattery:dbus-serialbattery v1.5.20241119beta
INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
INFO:SerialBattery:Testing Seplos at address "\x00"
INFO:SerialBattery:-- Connection established to Seplos
INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 2
INFO:SerialBattery:Use DeviceInstance: 2
INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB1
INFO:SerialBattery:Polling interval: 5.000 s
INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
INFO:SerialBattery:========== Settings ==========
INFO:SerialBattery:> Connection voltage: 53.31 V | Current: 0.0 A | SoC: 80.7%
INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
INFO:SerialBattery:> CVCM:     True
INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: None
INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16S_280.0Ah
```

Extra logs:
```
serial-starter
INFO: loading config file /etc/venus/serial-starter.conf
INFO: Create daemontools service mk2-dbus.ttyUSB0
INFO: Create daemontools service dbus-cgwacs.ttyUSB1
INFO: Start service mk2-dbus.ttyUSB0
INFO: Start service dbus-cgwacs.ttyUSB1 once
INFO: Create daemontools service dbus-fzsonick-48tl.ttyUSB1
INFO: Start service dbus-fzsonick-48tl.ttyUSB1 once
INFO: Create daemontools service dbus-imt-si-rs485tc.ttyUSB1
INFO: Start service dbus-imt-si-rs485tc.ttyUSB1 once
INFO: Create daemontools service dbus-modbus-client.serial.ttyUSB1
INFO: Start service dbus-modbus-client.serial.ttyUSB1 once
INFO: Create daemontools service gps-dbus.ttyUSB1
INFO: Start service gps-dbus.ttyUSB1 once
INFO: Create daemontools service vedirect-interface.ttyUSB1
INFO: Start service vedirect-interface.ttyUSB1 once
INFO: Start service dbus-cgwacs.ttyUSB1 once
INFO: Start service dbus-fzsonick-48tl.ttyUSB1 once
INFO: Start service dbus-imt-si-rs485tc.ttyUSB1 once
INFO: Start service dbus-modbus-client.serial.ttyUSB1 once
INFO: Start service gps-dbus.ttyUSB1 once
INFO: Start service vedirect-interface.ttyUSB1 once
INFO: Start service dbus-cgwacs.ttyUSB1 once
INFO: Start service dbus-fzsonick-48tl.ttyUSB1 once
```

Remark: 
I also see a weird information in the device information:
[image]

### How to reproduce

I don't know.

### Expected behavior

I expect to have one instance instead of 2

### Driver version of the currently installed driver

v1.5.20241119beta

### Driver version of the last known working driver

v1.2.20240408

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.52

### BMS type

Seplos

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
root@venus:/data/etc/dbus-serialbattery# cat config.ini | grep -v ';' | grep .
[DEFAULT]
LOGGING = INFO
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0
MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375
SOC_RESET_VOLTAGE = 3.520
SOC_RESET_AFTER_DAYS =
SOC_CALCULATION   = False
SOC_RESET_CURRENT = 7
SOC_RESET_TIME    = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300
BLUETOOTH_BMS =
BLUETOOTH_USE_USB = False
CAN_PORT =
CAN_SPEED = 250
MODBUS_ADDRESSES =
BLOCK_ON_DISCONNECT = False
BLOCK_ON_DISCONNECT_TIMEOUT_MINUTES = 20
BLOCK_ON_DISCONNECT_VOLTAGE_MIN = 3.25
BLOCK_ON_DISCONNECT_VOLTAGE_MAX = 3.35
LINEAR_LIMITATION_ENABLE = True
LINEAR_RECALCULATION_EVERY = 60
LINEAR_RECALCULATION_ON_PERC_CHANGE = 33
EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE =
EXTERNAL_CURRENT_SENSOR_DBUS_PATH =
CVCM_ENABLE = True
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL        = 0.025
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.013
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT        = 0.080
MAX_VOLTAGE_TIME_SEC = 900
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 85
CVL_ICONTROLLER_MODE   = True
CVL_ICONTROLLER_FACTOR = 0.2
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.04, 0.10, 0.8, 1
CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.10
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True
TEMPERATURES_WHILE_CHARGING      =    0,    2,    5,   10,   14,   18,   35,   40,   55
MAX_CHARGE_CURRENT_T_FRACTION    = 0.00, 0.10, 0.20, 0.40, 0.50, 1.00, 1.00, 0.40, 0.00
TEMPERATURES_WHILE_DISCHARGING   =  -20,    0,    5,   10,   14,   18,   35,   45,   55
MAX_DISCHARGE_CURRENT_T_FRACTION = 0.00, 0.10, 0.20, 0.40, 0.50, 1.00, 1.00, 0.40, 0.00
CCCM_SOC_ENABLE = True
DCCM_SOC_ENABLE = True
SOC_WHILE_CHARGING                 =  100,   98,   95,   90,   85
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.08, 0.15, 0.55, 0.85, 1.00
SOC_WHILE_DISCHARGING              =    0,    5,   10,   15,   20
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.00, 0.10, 0.20, 0.50, 1.00
CHARGE_CURRENT_RECOVERY_THRESHOLD_PERCENT = 0.015
DISCHARGE_CURRENT_RECOVERY_THRESHOLD_PERCENT = 0.015
TIME_TO_GO_ENABLE = True
TIME_TO_SOC_POINTS =
TIME_TO_SOC_VALUE_TYPE = 1
TIME_TO_SOC_RECALCULATE_EVERY = 60
TIME_TO_SOC_INC_FROM = False
BMS_TYPE = Seplos
EXCLUDED_DEVICES = /dev/ttyUSB0
POLL_INTERVAL =
PUBLISH_CONFIG_VALUES = True
BATTERY_CELL_DATA_FORMAT = 1
MIDPOINT_ENABLE = False
TEMP_BATTERY = 0
TEMP_1_NAME = Temp 1
TEMP_2_NAME = Temp 2
TEMP_3_NAME = Temp 3
TEMP_4_NAME = Temp 4
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
TELEMETRY = True
VOLTAGE_DROP = 0.00
AUTO_RESET_SOC = True
USE_PORT_AS_UNIQUE_ID = False
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10
BATTERY_CAPACITY = 50
INVERT_CURRENT_MEASUREMENT = 1
JKBMS_CAN_CELL_COUNT = 1
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15
SEPLOS_USE_BMS_VALUES = False
```

### Relevant log output

```shell
Added in description as there are some steps by steps logs
```

mr-manuel, 2024-11-29:
The unqie identifier changed. Use `dbus-spy` to check the dbus-data under `com.victronenergy.settings/Settings/Devices/serialbattery*`. There you should see two different serialbattery devices.

To fix this fastly disable the driver by running `bash /data/etc/dbus-serialbattery/disable.sh`, then open `dbus-spy`, navigate to `com.victronenergy.settings/Settings/Devices/serialbattery_*/ClassAndVrmInstance`. By pressing enter you can change the value.

On the entry with `battery:1` change it to `battery:3`. On the entry with `battery:2` change it to `battery:1`.

Run `bash /data/etc/dbus-serialbattery/reinstall-local.sh` to start the driver again.

mr-manuel, 2024-11-29:
Please post a screenshot of the dbus-entries, so we see how the unique ID changed.

zigomatichub, 2024-11-29:
```
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/AllowMaxVoltage                                1
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/ClassAndVrmInstance                    battery:2
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/CustomName                 SerialBattery(Seplos)
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/LastSeen                              1732875910
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/MaxVoltageStartTime
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/SocCalc                                     81.4
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/SocResetLastReached                            0
Settings/Devices/serialbattery_SeplosBMS16S_280_0Ah/UniqueIdentifier            SeplosBMS16S_280_0Ah

Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/AllowMaxVoltage                            1
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/ClassAndVrmInstance                battery:1
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/CustomName             SerialBattery(Seplos)
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/LastSeen                          1732402606
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/MaxVoltageStartTime
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/SocCalc                                 79.5
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/SocResetLastReached                        0
Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/UniqueIdentifier    SeplosBMS16cells_280_0Ah
```

mr-manuel, 2024-11-29:
Thanks. Yes I changed indeed something in the driver, so that all names are consistant. But should be solved for you now, by changing it manually. The other entry get's deleted after a while.

closed 2024-11-29 by mr-manuel

zigomatichub, 2024-11-29:
Thank you @mr-manuel 

Yes, the swap make the things

I tried to remove it but still present ;)

FYI
```
root@venus:/data/log/dbus-serialbattery.ttyUSB1# dbus -y com.victronenergy.settings /Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah
Interface org.freedesktop.DBus.Introspectable:
 String Introspect()

Interface com.victronenergy.Settings:
 Int32 AddSetting(String group, String name, Variant defaultValue, String itemType, Variant minimum, Variant maximum)
 Int32 AddSilentSetting(String group, String name, Variant defaultValue, String itemType, Variant minimum, Variant maximum)
 Dictionary {s, v}[] AddSettings(Dictionary {s, v}[] definition)
 Int32[] RemoveSettings(String[] settings)

Interface com.victronenergy.BusItem:
 Variant GetValue()
 Variant GetText()
 Int32 SetDefault()

root@venus:/data/log/dbus-serialbattery.ttyUSB1#dbus -y com.victronenergy.settings /Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/AllowMaxVoltage RemoveSettings "[/Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/AllowMaxVoltage]"
```
I tried a lot of commands but didn't work ...

Remark:
 I don't know if it's important but I see after restart this. 

```
2024-11-29 14:02:13.625166500 Traceback (most recent call last):
2024-11-29 14:02:13.625171500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 500, in read
2024-11-29 14:02:13.625175500     raise SerialException(
2024-11-29 14:02:13.625178500 serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-11-29 14:02:13.625183500
2024-11-29 14:02:13.625185500 During handling of the above exception, another exception occurred:
2024-11-29 14:02:13.625188500
2024-11-29 14:02:13.625277500 Traceback (most recent call last):
2024-11-29 14:02:13.625280500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 698, in publish_battery
2024-11-29 14:02:13.625284500     result = self.battery.refresh_data()
2024-11-29 14:02:13.625287500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 146, in refresh_data
2024-11-29 14:02:13.625291500     result_status = self.read_status_data()
2024-11-29 14:02:13.625294500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 202, in read_status_data
2024-11-29 14:02:13.625323500     data = self.read_serial_data_seplos(self.encode_cmd(self.address, cid2=0x42, info=b"01"))
2024-11-29 14:02:13.625327500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 283, in read_serial_data_seplos
2024-11-29 14:02:13.625331500     data = ser.readline()
2024-11-29 14:02:13.625334500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 509, in read
2024-11-29 14:02:13.625394500     raise SerialException('read failed: {}'.format(e))
2024-11-29 14:02:13.625398500 serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-11-29 14:02:15.900537500 INFO:SerialBattery:
2024-11-29 14:02:15.900817500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-29 14:02:15.901812500 INFO:SerialBattery:Venus OS v3.52
2024-11-29 14:02:15.902093500 INFO:SerialBattery:dbus-serialbattery v1.5.20241119beta
2024-11-29 14:02:31.904977500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-11-29 14:02:31.905336500 INFO:SerialBattery:Testing Seplos at address "\x00"
2024-11-29 14:02:32.033138500 ERROR:SerialBattery:Exception occurred: SerialException('read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)') of type <class 'serial.serialutil.SerialException'> in /opt/victronenergy/dbus-serialbattery/bms/seplos.py line #104
2024-11-29 14:02:32.033413500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-29 14:02:32.534450500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-11-29 14:02:32.534807500 INFO:SerialBattery:Testing Seplos at address "\x00"
2024-11-29 14:02:33.653188500 WARNING:SerialBattery:checksum error
2024-11-29 14:02:33.653705500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-29 14:02:34.154858500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-11-29 14:02:34.155237500 INFO:SerialBattery:Testing Seplos at address "\x00"
2024-11-29 14:02:35.205430500 WARNING:SerialBattery:command returned with error code b'05'
2024-11-29 14:02:35.205938500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-29 14:02:35.707014500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
2024-11-29 14:03:24.097001500 INFO:SerialBattery:
2024-11-29 14:03:24.097296500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-29 14:03:24.098462500 INFO:SerialBattery:Venus OS v3.52
2024-11-29 14:03:24.098773500 INFO:SerialBattery:dbus-serialbattery v1.5.20241119beta
2024-11-29 14:03:40.115376500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-11-29 14:03:40.115729500 INFO:SerialBattery:Testing Seplos at address "\x00"
2024-11-29 14:03:43.473590500 INFO:SerialBattery:-- Connection established to Seplos
2024-11-29 14:03:43.681187500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2024-11-29 14:03:43.817042500 INFO:SerialBattery:Use DeviceInstance: 1
2024-11-29 14:03:43.817947500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB1
2024-11-29 14:03:43.888315500 INFO:SerialBattery:Polling interval: 5.000 s
[... cut ...]
2024-11-29 14:03:43.888416500 INFO:SerialBattery:> CVCM:     True
2024-11-29 14:03:43.888419500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-11-29 14:03:43.888422500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-11-29 14:03:43.888425500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-11-29 14:03:43.888454500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: None
2024-11-29 14:03:43.888458500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16S_280.0Ah

```