mr-manuel/venus-os_dbus-serialbattery #120: JBD SP14S004 not working on BLE

warwickchapman, 2024-12-06:
### Describe the problem

Can't get it to work over BLE.  Works over serial.
￼￼
[image]

### Driver version of the currently installed driver

dbus-serialbattery v1.5.20241202

### Driver version of the last known working driver

?

### Venus OS device type

Cerbo GX

### Venus OS version

Venus OS v3.52 running on Cerbo GX

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

BLUETOOTH_BMS = LltJbd_Ble A5:C2:37:03:B4:F1
```

### Relevant log output

```shell
2024-12-06 11:42:37.139538500 Successful disconnected
2024-12-06 11:42:42.262498500 Device A5:C2:37:03:B4:F1 (public)
2024-12-06 11:42:42.262506500 	Alias: SP14S004L13S20A
2024-12-06 11:42:42.262508500 	Paired: no
2024-12-06 11:42:42.262509500 	Trusted: no
2024-12-06 11:42:42.262511500 	Blocked: no
2024-12-06 11:42:42.262512500 	Connected: no
2024-12-06 11:42:42.262514500 	LegacyPairing: no
2024-12-06 11:42:42.262515500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-12-06 11:42:42.262518500 	RSSI: -52
2024-12-06 11:42:42.263404500
2024-12-06 11:42:45.724135500 INFO:SerialBattery:
2024-12-06 11:42:45.725075500 INFO:SerialBattery:Starting dbus-serialbattery
2024-12-06 11:42:45.727917500 INFO:SerialBattery:Venus OS v3.52 running on Cerbo GX
2024-12-06 11:42:45.728643500 INFO:SerialBattery:dbus-serialbattery v1.5.20241202
2024-12-06 11:42:46.184648500 INFO:SerialBattery:Init of LltJbd_Ble at A5:C2:37:03:B4:F1
2024-12-06 11:42:46.185513500 INFO:SerialBattery:Test of LltJbd_Ble at A5:C2:37:03:B4:F1
2024-12-06 11:42:51.201695500 ERROR:SerialBattery:No BMS found at A5:C2:37:03:B4:F1
2024-12-06 11:42:51.202490500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-12-06 11:42:53.511252500 INFO:SerialBattery:BLE client disconnected
2024-12-06 11:42:53.960910500
2024-12-06 11:42:53.961268500
2024-12-06 11:42:53.961598500 INFO:Preparing Bluetooth for connection to BMS
2024-12-06 11:42:53.961862500 INFO:Bluetooth details
2024-12-06 11:42:54.018568500 [CHG] Device A5:C2:37:03:B4:F1 RSSI: -52
2024-12-06 11:42:54.018579500 [CHG] Device A5:C2:37:03:B4:F1 RSSI: -52
2024-12-06 11:42:54.135877500 Attempting to disconnect from A5:C2:37:03:B4:F1
```
```

### Any other information that may be helpful

Ps. Works fine over USB:

```
2024-12-06 11:52:48.873419500 INFO:SerialBattery:
2024-12-06 11:52:48.874256500 INFO:SerialBattery:Starting dbus-serialbattery
2024-12-06 11:52:48.877497500 INFO:SerialBattery:Venus OS v3.52 running on Cerbo GX
2024-12-06 11:52:48.878365500 INFO:SerialBattery:dbus-serialbattery v1.5.20241202
2024-12-06 11:53:04.895549500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-12-06 11:53:04.896437500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-12-06 11:53:05.041900500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:05.042837500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-12-06 11:53:05.185848500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:05.186788500 INFO:SerialBattery:Testing Daren485 at address "\x01"
2024-12-06 11:53:05.611884500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:05.612672500 INFO:SerialBattery:Testing Ecs
2024-12-06 11:53:05.672169500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:05.673044500 INFO:SerialBattery:Testing EG4_Lifepower at address "\x01"
2024-12-06 11:53:05.943628500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:06.213858500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:06.215208500 INFO:SerialBattery:Testing EG4_LL at address "\x01"
2024-12-06 11:53:06.485617500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:06.486786500 INFO:SerialBattery:Testing HeltecModbus at address "\x01"
2024-12-06 11:53:10.335371500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:10.336755500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-12-06 11:53:11.465205500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:11.466040500 INFO:SerialBattery:Testing Jkbms
2024-12-06 11:53:11.754540500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:11.772966500 INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
2024-12-06 11:53:12.043985500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-06 11:53:12.064003500 INFO:SerialBattery:Testing LltJbd at address "\x00"
2024-12-06 11:53:12.675162500 INFO:SerialBattery:-- Connection established to LltJbd
2024-12-06 11:53:16.223084500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
2024-12-06 11:53:16.487986500 INFO:SerialBattery:Use DeviceInstance: 1
2024-12-06 11:53:16.490042500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyACM0
2024-12-06 11:53:16.526816500 INFO:SerialBattery:Polling interval: 1.000 s
2024-12-06 11:53:16.528570500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
2024-12-06 11:53:16.529498500 INFO:SerialBattery:========== Settings ==========
2024-12-06 11:53:16.530503500 INFO:SerialBattery:> Connection voltage: 27.8 V | Current: 0.0 A | SoC: 100%
2024-12-06 11:53:16.531434500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2024-12-06 11:53:16.533423500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-12-06 11:53:16.536716500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-12-06 11:53:16.536726500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-12-06 11:53:16.536731500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 25.0 A | MAX BATTERY DISCHARGE CURRENT: 25.0 A (read from BMS)
2024-12-06 11:53:16.537250500 INFO:SerialBattery:> CVCM:     True
2024-12-06 11:53:16.544713500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-12-06 11:53:16.547316500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-12-06 11:53:16.549574500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-12-06 11:53:16.550563500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-12-06 11:53:16.551688500 INFO:SerialBattery:Serial Number/Unique Identifier: 20_100.0Ah
```

warwickchapman, 2024-12-06:
@idstein @Marvo2011

IamSierraCharlie, 2024-12-22:
Best advice is not to use it - at least not in a production environment.  The driver that the Bluetooth relies on is very flaky (in particular the Broadcom driver).   I discussed briefly with mr-manuel in the past 36 hours about it - it was never stable.  Many people report that they can get  a BMS working on the Cerbo with Bluetooth, but it randomly drops and never recovers unless you reboot - typically segmentation faults or "connection caused software abort" via the BLEAK module.

Marvo2011, 2024-12-23:
Bluetooth is very complex, I got my JBD stable with my patch on a raspberry pi and an external usb device. I don't now if JBD has swapt the protocol with this new device. Has somwere more informations for this device ?

warwickchapman, 2024-12-23:
@Marvo2011 May I look at your patch?

@IamSierraCharlie it looks like it finds the device, but does not recognize it in the logs. I will try a Bluetooth dongle and see if that makes any difference.

Honusnap, 2025-01-21:
> Best advice is not to use it - at least not in a production environment. The driver that the Bluetooth relies on is very flaky (in particular the Broadcom driver). I discussed briefly with mr-manuel [...]

Was using 1.2xxx and never got one drop in mote then 6 months, i call it stable. Then i upgraded to 1.4... and could not have more then 20minaa before reconnect and drop .. reconnect ... and drop.
Something has been added/changed that fucked it all ... and putting the responsability on Bleak is not fair, there is clearly something that was ok .. and that is now badly done. The guy prefer blaming bleak then looking for it own mistakes, it's sad but not surprising.

chadodes, 2025-03-07:
@warwickchapman I have a Vatrer Battery. In the X app it says the Manufacture of the BMS is ZJDY.
It comes up as "DWCE00535J-007" when I run a scan in bluetoothctrl.

I figured this information might be relevant to you as I noticed our BT MAC#'s are almost identical, only the last 4 are different.

Based off of @Honusnap information about his experience with bleak, I decided to give the old build a try.
I must say it works almost flawlessly. I used build: 1.2.20240408

May I suggest you give it a go?

From the install menu, select the option for a specific version and then paste in this url:
https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/download/v1.2.20240408/venus-data.tar.gz 

This was the only way I could get the battery to show up without having to dismantle it in order to get to the BMS UART.
The driver type I used was: LltJbd_Ble

Hope This helps.

warwickchapman, 2025-03-08:
Great. Thank you!

I’ll try it out. The battery in question is already deployed in the field with a USB interface but will try on next. 

@mr-manuel any idea where the regression took place?

mr-manuel, 2025-03-10:
I'm not sure, if you all talk about the same BMS.

### JKBMS --> NOT RELEVANT HERE

[Honusnap](https://github.com/Honusnap) is very likely talking about a JKBMS where the driver has completely a different logic. We switched from active polling to active callback, which for some time solved issues that were present when using active polling. Now there are issues with the active callback and I switched back to active polling. In the `ble-testing` branch this can now be selected in the config file. Anyway this has NOTHING to do with the LLT/JBD BLE driver.

### LLT/JBD BMS

Here are the other changes:

- [dbus-serialbattery/bms/lltjbd.py](https://github.com/mr-manuel/venus-os_dbus-serialbattery/compare/v1.2.20240408..v1.6.20250131#diff-5f8eb74263798c35aa24ac3df4181766042521be0df4b68be1bbfab001f55c19)
- [dbus-serialbattery/bms/lltjbd_ble.py](https://github.com/mr-manuel/venus-os_dbus-serialbattery/compare/v1.2.20240408..v1.6.20250131#diff-85c61bcabdaae67d662bf517a8257b752823f82f901188c6058f89c4e545d953)
- [dbus-serialbattery/dbus-serialbattery.py](https://github.com/mr-manuel/venus-os_dbus-serialbattery/compare/v1.2.20240408..v1.6.20250131#diff-5979dc9642c469007179aeda41a1672147e7dbf176a465ae47b4bcb7beab6612)

Could be something address related, since the JBD BMS does not support addresses. Maybe this is the show stopper:

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/01c01b88d785912c3eccf6c5ef96569db57f3285/etc/dbus-serialbattery/bms/lltjbd.py#L274-L276

I implemented a small modification in the `ble-testing` branch. You could try to install that and see if it works.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/02e8a4950dee893b88c11e68d1b937b82d1e7fcf/dbus-serialbattery/bms/lltjbd.py#L275-L278

To test this changes you have to install the `ble-testing` branch. Run the `install.sh` script, select `specific branch (specific feature testing)` and then `ble-testing`.

warwickchapman, 2025-03-10:
I will try to test this in the next wek or two.

NunoRgP, 2025-03-12:
I have two JBD BMSs running Venus OS 3.54 on a Raspberry Pi with an external Bluetooth dongle.

I installed the ble-testing branch, and immediately experienced the fastest Bluetooth connection of any version I've ever tested. Both BMSs connected right after the installation was completed.

With version v1.3.20240705 and lower (I couldn’t install higher versions as they didn’t work with BLE), only one BMS would connect after some time, and the other would follow up later. When I checked the logs, I could see the connection failed a couple of times but would eventually succeed. Sometimes, I even had to reboot a couple of times for both BMSs to connect.

Below is a very uneventful log (a first!) 👍. I will report back tomorrow about stability, but so far, so good!

```
2025-03-12 17:29:00.470381500 INFO:SerialBattery:Starting dbus-serialbattery
2025-03-12 17:29:00.472607500 INFO:SerialBattery:Venus OS v3.54 running on Rasperry Pi 4 Model B Rev 1.5                                                                                                                                                    
2025-03-12 17:29:00.473250500 INFO:SerialBattery:dbus-serialbattery v2.0.20250310-ble-testing                                                                                                                                                             
2025-03-12 17:29:01.223120500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:50:52:41                                                                                                                                                             
2025-03-12 17:29:01.232854500 INFO:SerialBattery:|- Try to connect to LltJbd_Ble  at A4:C1:37:50:52:41                                                                                                                                                            
2025-03-12 17:29:04.222435500 INFO:SerialBattery:|- Device connected, check if it's really a LLT/JBD BMS                                                                                                                                                 
2025-03-12 17:29:07.975660500 INFO:SerialBattery:-- Connection established to LltJbd_Ble                                                                                                                                                            
2025-03-12 17:29:08.718129500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1                                                                                                                                                             
2025-03-12 17:29:08.988136500 INFO:SerialBattery:Use DeviceInstance: 1
2025-03-12 17:29:08.989373500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ble_a4c137505241                                                                                                                                                             
2025-03-12 17:29:09.016156500 INFO:SerialBattery:Polling interval: 1.000 s
2025-03-12 17:29:09.029014500 INFO:SerialBattery:Battery LLT/JBD BLE connected to dbus from ble_a4c137505241                                                                                                                                                             
2025-03-12 17:29:09.029411500 INFO:SerialBattery:========== Settings ==========
2025-03-12 17:29:09.029848500 INFO:SerialBattery:> Connection voltage: 52.62 V |   Current: 1.34 A | SoC: 51%                                                                                                                                                           
2025-03-12 17:29:09.030183500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16                                                                                                                                                            
2025-03-12 17:29:09.030497500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-03-12 17:29:09.031622500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V                                                                                                                                                             
2025-03-12 17:29:09.032461500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A                                                                                                                                                             
2025-03-12 17:29:09.032467500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 110.0 A | MAX BATTERY DISCHARGE CURRENT: 110.0 A (read from BMS)                                                                                                                                                             
2025-03-12 17:29:09.034910500 INFO:SerialBattery:> CVCM:     True
2025-03-12 17:29:09.035422500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True                                                                                                                                                             
2025-03-12 17:29:09.035808500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True                                                                                                                                                             
[... cut ...]
2025-03-12 17:29:09.036544500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGEFET: True | BALANCE FET: True                                                                                                                                                              
2025-03-12 17:29:09.036944500 INFO:SerialBattery:Serial Number/Unique Identifier: a4c137505241                                                                                                                                                             
```

mr-manuel, 2025-03-12:
Nice to hear also good feedback once in a while, thanks!

closed 2025-03-23 by mr-manuel