mr-manuel/venus-os_dbus-serialbattery #123: Allow "SOC reset voltage" to operate independently of SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT

BipedalPrimate, 2024-12-13:
### Is your feature request related to a problem? Please describe.

"SOC reset voltage" seems to be conditioned on SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT and SOC_RESET_AFTER_DAYS.  It appears to be designed to reset the BMS SOC to 100% to counter the drift that BMS SOC measurements incur..

There is use case to allow a full charge after SOC_RESET_AFTER_DAYS at Float Voltage regardless of SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT.  

At Float Voltage and SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 20%, SOC may oscillate between 85%-40% for extended periods during summer months. Having the ability to perform a Bulk charge SOC_RESET_AFTER_DAYS allows the user to fully charge battery & re-balance cells whether SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT has been reached or not.

If usual Bulk Voltage is 3.45v, SOC_CALCULATION = True and SOC_RESET_VOLTAGE of 3.4501, a reset of external current sensor SOC (if present) & dbus-serialbattery internal SOC to 100% can occur.

In this scenario, the BMS SOC is irrelevant because dbus-serialbattery calculates SOC.

### Describe the solution you'd like

Condition "SOC reset voltage"  on only SOC_RESET_AFTER_DAYS and exclude SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT from consideration.

Existing users of "SOC reset voltage" may see a slight change in behaviour if their SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT is low but in this case, I suspect it will be a beneficial change.

mr-manuel, 2025-01-02:
`SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT` has nothing to do with `SOC_RESET_AFTER_DAYS`. I renamed `SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT` now to `SWITCH_TO_BULK_SOC_THRESHOLD` for easier understanding.

The mechanism for `SOC_RESET_AFTER_DAYS` just changes `MAX_CELL_VOLTAGE` with `SOC_RESET_VOLTAGE`, if the days are exceeded. This means, if the system is still in float mode and `FLOAT_CELL_VOLTAGE` is applied, it has no effect as you see.

Indeed on systems with low comsumption during night and low `SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT` it can be that it's not switched to bulk anymore.

I changed this now.

closed 2025-01-02 by mr-manuel

ibraheemhlaiyil, 2025-01-28:
Further to this, I think there is some missing safety logic. If the battery remains unbalanced, battery will be charged at SOC_RESET_VOLTAGE forever, since the battery high voltage alarm will not be activated on the BMS (testing on Seplos v3). This will cause damage to the cells if battery is not discharged. There should be another timer that stops the usage of SOC_RESET_VOLTAGE after some time when the current voltage is > MAX_CELL_VOLTAGE.

I also think if the driver had tail current support this would also solve this problem instead of a timer. Tail current support would also provide better transition from bulk to float instead of a fixed max time at max voltage. (Imagine how this changes if a battery is unbalanced vs balanced).

Do you have any ideas? I was going to fork and write these changes wanted to get your thoughts first.

mr-manuel, 2025-01-28:
Please open a new issue for that, so we can discuss it there.