mr-manuel/venus-os_dbus-serialbattery #126: Oddities in the log file

pkeberly, 2024-12-13:
### Describe the problem

The driver worked yesterday on the bench.  Today I noticed some errors in the log file after a reboot.  I was editing config.ini end of day yesterday so it might be pilot error.

### Driver version of the currently installed driver

1.5.20241202

### Venus OS device type

Cerbo GX

### Venus OS version

V3.6~8

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- Battery Current Limits ---------
MAX_BATTERY_CHARGE_CURRENT = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description:
;     Minimum and maximum cell voltages used for:
;         - Calculating the minimum and maximum battery voltage
;         - Triggering the SoC reset when SoC calculation is enabled
; Example:
;     8 cells * 3.52 V/cell = 28.16 V max charge voltage (default was 3.45 V)
;     8 cells * 2.90 V/cell = 23.20 V min discharge voltage
; Minimum voltage (can be seen as cut-off voltage)
MIN_CELL_VOLTAGE   = 2.900
; Maximum voltage (can be seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.520
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SoC Calculation ---------
; +++ This section is independent and unrelated to the "SoC reset voltage" section above. +++
; This is used to calculate the SoC in the driver instead of using the SoC reported by the BMS.
; Description:
[... 253 lines cut ...]
@40000000675c68dc11674ca4 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@40000000675c68dc1179bb64 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 10
@40000000675c68dc11946b6c INFO:SerialBattery:cell_voltages_good: False - min: 3.219 > 3.25 - max: 3.223 < 3.35
@40000000675c68f21c4f565c ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@40000000675c68f21c5f159c WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 192
@40000000675c68f21d094bd4 INFO:SerialBattery:cell_voltages_good: False - min: 3.219 > 3.25 - max: 3.223 < 3.35
@40000000675c68f528e3088c ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 50 expected 45
@40000000675c68f528f4fe34 INFO:SerialBattery:cell_voltages_good: False - min: 3.219 > 3.25 - max: 3.223 < 3.35
@40000000675c68f7206bd7c4 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@40000000675c68f7207a2004 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 25 expected 7
@40000000675c68f7208fbb44 INFO:SerialBattery:cell_voltages_good: False - min: 3.219 > 3.25 - max: 3.223 < 3.35
@40000000675c69082b1c2fac ERROR:SerialBattery:>>> ERROR: No reply - returning [len:22/22]
@40000000675c69082b65cb6c ERROR:SerialBattery:>>> ERROR: Battery does not respond, init/reset values <<<
@40000000675c69082c38bf54 Traceback (most recent call last):
@40000000675c69082c38e664   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 776, in publish_battery
@40000000675c69082c38f604     self.battery.manage_charge_voltage()
@40000000675c69082c38fdd4   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 395, in manage_charge_voltage
@40000000675c69082c390d74     self.soc_calculation()
@40000000675c69082c391544   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 465, in soc_calculation
@40000000675c69082c3ac6dc     self.soc_calc_capacity_remain + self.current_corrected * (current_time - self.soc_calc_capacity_remain_lasttime) / 3600
@40000000675c69082c3ade4c TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
@40000000675c693b2b2d90e4 INFO:SerialBattery:
@40000000675c693b2b3b1da4 INFO:SerialBattery:Starting dbus-serialbattery
@40000000675c693b2ba71e14 INFO:SerialBattery:Venus OS v3.60~8 running on Cerbo GX
```

mr-manuel, 2024-12-13:
This was a bug and I fixed it. Please reinstall the latest stable version.

pkeberly, 2024-12-14:
That was your latest version.  I tried your latest beta this morning and some oddities still exist.
I have not yet tried to revert to earlier versions of Venus.  If you feel the driver is safe to continue bench testing with, we can ignore it for now.   It appears to be a typo in a method name?? "remove_settings**s**"

<code>
2024-12-14 15:50:22.376229500 INFO:SerialBattery:
2024-12-14 15:50:22.376946500 INFO:SerialBattery:Starting dbus-serialbattery
2024-12-14 15:50:22.379073500 INFO:SerialBattery:Venus OS v3.60~8
2024-12-14 15:50:22.379695500 INFO:SerialBattery:dbus-serialbattery v1.5.20241119beta
2024-12-14 15:50:38.386993500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-12-14 15:50:38.388146500 INFO:SerialBattery:Testing LltJbd at address "\x00"
2024-12-14 15:50:39.785224500 INFO:SerialBattery:-- Connection established to LltJbd
2024-12-14 15:50:42.807118500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 3
2024-12-14 15:50:42.840121500 ERROR:SerialBattery:Failed to remove setting: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
2024-12-14 15:50:42.840131500   File "/usr/lib/python3.8/site-packages/dbus/service.py", line 658, in _message_cb
2024-12-14 15:50:42.840135500     (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
2024-12-14 15:50:42.840139500   File "/usr/lib/python3.8/site-packages/dbus/service.py", line 248, in _method_lookup
2024-12-14 15:50:42.840143500     raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
2024-12-14 15:50:42.840245500 dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: remove_settingss is not a valid method of interface com.victronenergy.Settings
2024-12-14 15:50:42.840252500 
2024-12-14 15:50:42.850818500 INFO:SerialBattery:Remove /Settings/Devices/serialbattery_ttyUSB0 from dbus. Old entry. Delete result: None
2024-12-14 15:50:42.883049500 ERROR:SerialBattery:Failed to remove setting: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
2024-12-14 15:50:42.883062500   File "/usr/lib/python3.8/site-packages/dbus/service.py", line 658, in _message_cb
2024-12-14 15:50:42.883066500     (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
2024-12-14 15:50:42.883070500   File "/usr/lib/python3.8/site-packages/dbus/service.py", line 248, in _method_lookup
2024-12-14 15:50:42.883073500     raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
2024-12-14 15:50:42.883188500 dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: remove_settingss is not a valid method of interface com.victronenergy.Settings
2024-12-14 15:50:42.883196500 
2024-12-14 15:50:42.893952500 INFO:SerialBattery:Remove /Settings/Devices/serialbattery_ttyUSB1 from dbus. Old entry. Delete result: None
2024-12-14 15:50:43.274110500 INFO:SerialBattery:Use DeviceInstance: 3
2024-12-14 15:50:43.276021500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB0
2024-12-14 15:50:43.343386500 INFO:SerialBattery:Polling interval: 1.000 s
2024-12-14 15:50:43.343395500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB0
2024-12-14 15:50:43.343535500 INFO:SerialBattery:========== Settings ==========
2024-12-14 15:50:43.344647500 INFO:SerialBattery:> Connection voltage: 25.85 V | Current: 0.0 A | SoC: 14% | SoC calc: 15%
2024-12-14 15:50:43.345572500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2024-12-14 15:50:43.346489500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-12-14 15:50:43.347516500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.520 V| FLOAT CELL VOLTAGE: 3.375 V
2024-12-14 15:50:43.351672500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A
2024-12-14 15:50:43.351684500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 300.0 A | MAX BATTERY DISCHARGE CURRENT: 300.0 A (read from BMS)
2024-12-14 15:50:43.351689500 INFO:SerialBattery:> CVCM:     True
2024-12-14 15:50:43.351692500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-12-14 15:50:43.351904500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-12-14 15:50:43.353058500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-12-14 15:50:43.355521500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-12-14 15:50:43.360661500 INFO:SerialBattery:Serial Number/Unique Identifier: 39_304.0Ah

</code>

mr-manuel, 2024-12-14:
> That was your latest version.

Yes, but I rebuild it with the fix.

Yes it‘s a typo. Should be `RemoveSettings`. As written above, I fixed the `v1.5.20241202` build. If you do a online reinstall, then it‘s fixed using the tar file.

Else the nightly works well.

mr-manuel, 2024-12-15:
I released a new version [`v1.5.20241215`](https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.5.20241215) with the fix now.

The SoC is very inaccurate, therefore the `CCL` and `DCL` by SoC are disabled by default. 

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/416f9e5fcac8049ae37c64209dc653044b396e4e/dbus-serialbattery/config.default.ini#L342-L353

Check the cell voltages instead of the SoC, which is enabled by default:

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/416f9e5fcac8049ae37c64209dc653044b396e4e/dbus-serialbattery/config.default.ini#L301-L312

`CCCM_*_ENABLE` and `DCCM_*_ENABLE` are the only options that limit `CCL` and `DCL` in a soft way, without the BMS has to interrupt in the hard way by turning off charging/discharging.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/416f9e5fcac8049ae37c64209dc653044b396e4e/dbus-serialbattery/config.default.ini#L18-L24

pkeberly, 2024-12-16:
I tried the latest beta & latest stable this morning.  Both produce an error, it looks like something is missing. What did I do wrong? 

ERROR: overlay-fs app not found. Please install it manually.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/416f9e5fcac8049ae37c64209dc653044b396e4e/dbus-serialbattery/config.default.ini#L342-L353

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/416f9e5fcac8049ae37c64209dc653044b396e4e/dbus-serialbattery/config.default.ini#L301-L312

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/416f9e5fcac8049ae37c64209dc653044b396e4e/dbus-serialbattery/config.default.ini#L18-L24

mr-manuel, 2024-12-16:
Just stick with `v2.x.x`, if you already installed it once.

pkeberly, 2024-12-16:
My apologizes but I don't see v2.x.x, and haven't installed that far yet.  I did select select the latest stable version and I still get the error. I even went back to a stable version of Venus to rule that out.

I have a second Cerbo-S GX that I tried to install the latest on, and it also gets "overlay-fs app not found" error.
I'm not sure what that error means or where to begin looking for problems.  

SSH install using Putty/Win11 with the command copied from the instructions.  This worked so well before, it seems to have went bad when the code moved from etc to apps?

Putty output from the Cerbo-S, 1202 driver was installed but never had the USB device plugged in.  

** Currently installed version: v1.5.20241202 **   

1) stable release "v1.5.20241215" (stable, most up to date)
2) beta build "v1.5.20241119beta" (no errors after 72 h runtime, long time testing needed)
3) nightly build "v2.0.20241211dev" (newest features and fixes, bugs possible)
4) specific branch (specific feature testing)
5) specific version
6) local tar file
7) quit

Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 1
> Selected: stable release "v1.5.20241215" (stable, most up to date)

Config.ini backed up to /data/etc/dbus-serialbattery_config.ini.backup
Config.ini restored to /data/apps/dbus-serialbattery/config.ini
ERROR: overlay-fs app not found. Please install it manually.
***@***.***:~#

It appears the serial starter isn't happy about something.  I found this entry several places in the serial starter log file
ERROR: no service template for dbus-serialbattery

Thanks, 
Paul

marcelrv, 2024-12-16:
Seeing the same here... trying to upgrade from earlier version.
note: running latest 3.6~beta

pkeberly, 2024-12-17:
Thanks marcelrv, I suspected it might have been pilot error on my end.

I tried a few combinations tonight to see what would work.
v1.5.20241215driver with 3.6 beta 
v1.4.20240928 driver with 3.6 beta
v1.5.20241215 driver with 3.52 production
factory reset Cerbo, v1.4.20240928 driver with 3.6 beta
factory reset Cerbo, v1.4.20240928 driver with 3.52 production
reflashed the Cerbo image to 3.50 with v1.4.20240928 driver

All combinations ended with: ERROR: overlay-fs app not found . Please install it manually.
I see the current install of v1.4.20240928 is located in **data/apps** NOT **data/etc** as it was previously.   It looks like the v2 path changes have made it into earlier versions too :(

mr-manuel, 2024-12-17:
The error was fixed.

closed 2024-12-17 by mr-manuel

marcelrv, 2024-12-17:
Thx! Indeed, fixed for me with latest update.