mr-manuel/venus-os_dbus-serialbattery #129: Io/AllowToCharge is 0 for EG4_Lifepower

ytechie, 2024-12-17:
### Describe the bug

Io/AllowToCharge is always 0 when using the EG4 LifePower BMS. The CCL is fine, but the battery aggregator I'm using looks at the AllowToCharge parameter and decides to change the total CCL to 0.

### How to reproduce

Use the EG4 LifePower BMS
Use dbus-spy and scroll down to Io/AllowToCharge

### Expected behavior

If a BMS doesn't support this feature, fall back to calculating via the CCL, or hardcode to 1.

### Driver version of the currently installed driver

v1.5.20241202

### Venus OS device type

Cerbo GX

### Venus OS version

v3.52

### BMS type

EG4 LiFePower (Revov, Tian Power)

### Cell count

8

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

LOGGING = INFO 

; LifePower battery specs - from the manual
; 1. Bulk/Absorption: 28.1V (+/- .2V)
; 2. Float: 27V (+/- .2V)
; 3. Low DC cutoff: 23.5-22V* (depending on load - start high, lower if needed)
; 4. Battery Charge Temperature Range: 32°F - 122°F 5. Battery Discharge Temperature Range: -4°F - 131°F
; Discharging current: 100a
; Charging current: 100a max, 30-50 recommended
; Charging range: 32°F to 122°F (0°C to 50°C)
; Discharging range: -4°F to 131°F (-20°C to 55°C)

MIN_CELL_VOLTAGE   = 2.800
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.550
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

PUBLISH_CONFIG_VALUES = True

MAX_BATTERY_CHARGE_CURRENT = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

BMS_TYPE = EG4_Lifepower
MODBUS_ADDRESSES = 0x01, 0x02
BATTERY_ADDRESSES = 0x01, 0x02

; Show additional info in GUI -> Serialbattery -> Parameters.
; This will show additional information to better understand how the driver works
; and what values are currently set which are not shown elsewhere in the GUI.
; You have to scroll down to see the additional information.
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True

; -- Unique ID settings
; Some BMS units do not have a unique ID and cannot be assigned one. In this case,
; you can use the USB port as the unique ID.
; Note: VRM IDs and custom names may not be saved/restored correctly when using this option.
USE_PORT_AS_UNIQUE_ID = True

SOC_CALCULATION = False

LINEAR_LIMITATION_ENABLE = False

; Disable charge voltage limitation
CVCM_ENABLE = True

; --------- SOC limitation (affecting CCL/DCL) ---------
; Description:
;     Maximal charge / discharge current will be increased / decreased depending on State of Charge
;     Since the SoC is not as accurate as the cell voltage, this option is disabled by default
; Example:
;     The SoC limit will be monitored to control the currents.
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = True
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
SOC_WHILE_CHARGING                 =   98,   95,   90,   80
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.10, 0.30, 0.50, 1.00

; Winter charge limits
;SOC_WHILE_CHARGING                 =   98,   80,   70
;MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.00, 0.00, 0.50

SOC_WHILE_DISCHARGING              =    5,   10,   12,   15
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.10, 0.20, 0.50, 1.00

; --------- Temperature limitation (affecting CCL/DCL) ---------
; Description:
;     Maximal charge / discharge current will be in-/decreased depending on temperature
; Example:
;     The temperature limit will be monitored to control the currents. If there are two temperature senors,
;     then the worst case will be calculated and the more secure lower current will be set.
; Charge current control management referring to temperature enable (True/False).
CCCM_T_ENABLE = True
; Charge current control management referring to temperature enable (True/False).
DCCM_T_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
TEMPERATURES_WHILE_CHARGING	 =    -1,    0,    1,   10,   15,   20,   35,   40,   50
MAX_CHARGE_CURRENT_T_FRACTION    = 0.00, 0.00, 1.00, 1.00, 1.00, 1.00, 1.00, 1.00, 0.00

TEMPERATURES_WHILE_DISCHARGING   =  -20,   -10,    5,   10,   15,   45,   55
MAX_DISCHARGE_CURRENT_T_FRACTION = 0.00, 1.00, 1.00, 1.00, 1.00, 1.00, 0.00
```

### Relevant log output

```shell
.
```

mr-manuel, 2024-12-17:
Please post a screenshot of the parameters page of the battery when `Allow to discharge` is `false`.

ytechie, 2024-12-17:
[image]

[image]

The EG4 Lifepower battery never seems to set the "allow to" parameters. If I hardcode an initial value of `True`, it seems to stay true and everything works. I think the fix may be to set this value in that BMS code itself. I'm not quite confident enough yet with the code base to send a pull request with that change.

I have always had this issue, but DVCC only looks at the CVL and CCL values, not the values under info. If I use one of my batteries with DVCC, everything works fine. It's just the battery aggregator breaking because [it's using those values](https://github.com/pulquero/BatteryAggregator/blob/b0766223741f97cfefa53adb01f3121edff3a1f6/battery_service.py#L638).

ytechie, 2024-12-17:
Adding this code to the `eg4_lifepower.py` `init` seems to fix it. If you think that's a reasonable fix, I can send a pull request.

```python
class EG4_Lifepower(Battery):
    def __init__(self, port, baud, address):
        # <removed code for brevity>
        self.charge_fet = True
        self.discharge_fet = True
```

mr-manuel, 2024-12-17:
This is fixed with `v2.0.20241217dev`.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/6639fbc79e5ca9f346c1763cf8d8155edb7fe492/CHANGELOG.md?plain=1#L45