mr-manuel/venus-os_dbus-serialbattery #130: ERROR: overlay-fs app not found. Please install it manually.

endurance1968, 2024-12-17:
### Describe the bug

Installation issue. Some docker compatibility?

### How to reproduce

run installer of actual verion and try to install version v1.5.20241215

### Expected behavior

installation runs without error but it shows ERROR: overlay-fs app not found. Please install it manually.

### Driver version of the currently installed driver

1.5.20241202

### Driver version of the last known working driver

1.5.20241202

### Venus OS device type

Cerbo GX

### Venus OS version

3.52

### BMS type

Seplos

### Cell count

16

### Battery count

6

### Connection type

Serial USB adapter to RS485

### Config file

```ini
NA
```

### Relevant log output

```shell
** Currently installed version: v1.5.20241202 **

1) stable release "v1.5.20241215" (stable, most up to date)                                 3) nightly build "v2.0.20241211dev" (newest features and fixes, bugs possible)              5) specific version                                                                         7) quit
2) beta build "v1.5.20241119beta" (no errors after 72 h runtime, long time testing needed)  4) specific branch (specific feature testing)                                               6) local tar file

Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 1
> Selected: stable release "v1.5.20241215" (stable, most up to date)

Config.ini backed up to /data/etc/dbus-serialbattery_config.ini.backup
Config.ini restored to /data/apps/dbus-serialbattery/config.ini
ERROR: overlay-fs app not found. Please install it manually.
```

### Any other information that may be helpful

NA

endurance1968, 2024-12-17:
Hmm seems that it was reported [here ](https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/126#issuecomment-2545850281) as well. But how to sort it out?

mr-manuel, 2024-12-17:
This was fixed now.

closed 2024-12-17 by mr-manuel