mr-manuel/venus-os_dbus-serialbattery #131: Update process, script not able to found venus-gui-v2.wasm.sha256

zigomatichub, 2024-12-17:
### Describe the bug

When updating to v1.5.20241215, install script report one issue related to V2 gui

### How to reproduce

Update to v1.5.20241215

### Expected behavior

I'm reporting the problem that I found.

### Driver version of the currently installed driver

v1.5.20241215

### Driver version of the last known working driver

NA

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.52

### BMS type

Seplos

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
NA
```

### Relevant log output

```shell
> Selected: stable release "v1.5.20241215" (stable, most up to date)
...

Installing QML files for GUI V1...
done.
cat: can't open '/data/etc/dbus-serialbattery/ext/venus-os_dbus-serialbattery_gui-v2/venus-gui-v2.wasm.sha256': No such file or directory

Installing GUIv2 web version...
ERROR: GUIv2 web version not found.
...
```

mr-manuel, 2024-12-17:
Paste the output of `ls -l /data/etc/dbus-serialbattery/ext` here.

Then try to execute

```bash
rm -rf /data/etc/dbus-serialbattery/ext/venus-os_dbus-serialbattery_gui-v2
```

and then execute the install script again.

Anyway. GitHub seems to have currently some problems, since the half of the pages do not load.

mr-manuel, 2024-12-17:
Is solved now.

closed 2024-12-17 by mr-manuel

zigomatichub, 2024-12-18:
@mr-manuel  
Hello thank you for your answer and I tried today but it's same:

wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/master/dbus-serialbattery/install.sh

then execute

*** Welcome to the dbus-serialbattery installer from mr-manuel! ***

Fetch available version numbers... done.

** Currently installed version: v1.5.20241215 **

1) stable release "v1.5.20241215" (stable, most up to date)
2) beta build "v1.5.20241119beta" (no errors after 72 h runtime, long time testing needed)
3) nightly build "v2.0.20241217dev" (newest features and fixes, bugs possible)
4) specific branch (specific feature testing)
5) specific version
6) local tar file
7) quit

Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 1
> Selected: stable release "v1.5.20241215" (stable, most up to date)

Config.ini backed up to /data/etc/dbus-serialbattery_config.ini.backup
Config.ini restored to /data/etc/dbus-serialbattery/config.ini

Installing QML files for GUI V1...
done.

Installing GUIv2 web version...
ERROR: GUIv2 web version not found.

root@venus:/tmp# ls -l /data/etc/dbus-serialbattery/ext
-rwxrwxrwx    1 root     root        137100 Dec 17 15:07 minimalmodbus.py
-rwxrwxrwx    1 root     root          1851 Dec 17 15:07 update.py
drwxrwxrwx    2 root     root          4096 Dec 17 15:07 velib_python

mr-manuel, 2024-12-18:
Seems more like an issue in your system and/or unstable internet connection. There are over 50 installations of `v1.5.20241215` therefore it can't be a common problem.

You could try to manually download the tar file, check it and then use the [Local tar file](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/general/install#local-tar-file) install method.

mr-manuel, 2024-12-18:
This is correct. The file should be downloaded, if it's not there.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/7f82961591db5585c7badd73d54a7bae6a9375c5/etc/dbus-serialbattery/install-qml.sh#L236-L280

zigomatichub, 2024-12-18:
But the problem is in github release tar file generated which is not containing your complete repo of ext file. 
I just downloaded the file that script is taking (I launched with set -x to see the stuff)

zigomatichub, 2024-12-18:
Strange: 

root@venus:/data/etc/dbus-serialbattery# wget -O "/data/etc/dbus-serialbattery/ext/venus-os_dbus-serialbattery_gui-v2/venus-webassembly.zip" https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery_gui-v2/refs/heads/master/venus-webassembly.zip
/data/etc/dbus-serialbattery/ext/venus-os_dbus-serialbattery_gui-v2/venus-webassembly.zip: No such file or directory

root@venus:/data/etc/dbus-serialbattery# wget https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery_gui-v2/refs/heads/master/venus-webassembly.zip          --2024-12-18 12:45:56--  https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery_gui-v2/refs/heads/master/venus-webassembly.zip
Resolving raw.githubusercontent.com... 185.199.110.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13937651 (13M) [application/zip]
Saving to: ‘venus-webassembly.zip’

venus-webassembly.zip                       100%[=========================================================================================>]  13.29M  4.21MB/s    in 3.2s

2024-12-18 12:45:59 (4.21 MB/s) - ‘venus-webassembly.zip’ saved [13937651/13937651]

mr-manuel, 2024-12-18:
I don't know, why the folder is not created in your case. There is already a check for that which creates the folder if it's missing.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/7f82961591db5585c7badd73d54a7bae6a9375c5/etc/dbus-serialbattery/install-qml.sh#L258-L260

zigomatichub, 2024-12-18:
mhh I need to look at the output here then: 

root@venus:/data/etc/dbus-serialbattery# ./install-qml.sh
+ filesChanged=0
+ '[' -d /opt/victronenergy/gui ']'
+ echo ''

+ echo 'Installing QML files for GUI V1...'
Installing QML files for GUI V1...
+ '[' '!' -f /opt/victronenergy/gui/qml/PageBattery.qml.backup ']'
+ '[' '!' -f /opt/victronenergy/gui/qml/PageBatteryParameters.qml.backup ']'
+ '[' '!' -f /opt/victronenergy/gui/qml/PageBatterySettings.qml.backup ']'
+ '[' '!' -f /opt/victronenergy/gui/qml/PageLynxIonIo.qml.backup ']'
+ cmp -s /data/etc/dbus-serialbattery/qml/gui-v1/PageBattery.qml /opt/victronenergy/gui/qml/PageBattery.qml
+ cmp -s /data/etc/dbus-serialbattery/qml/gui-v1/PageBatteryCellVoltages.qml /opt/victronenergy/gui/qml/PageBatteryCellVoltages.qml
+ cmp -s /data/etc/dbus-serialbattery/qml/gui-v1/PageBatteryParameters.qml /opt/victronenergy/gui/qml/PageBatteryParameters.qml
+ cmp -s /data/etc/dbus-serialbattery/qml/gui-v1/PageBatterySettings.qml /opt/victronenergy/gui/qml/PageBatterySettings.qml
+ cmp -s /data/etc/dbus-serialbattery/qml/gui-v1/PageLynxIonIo.qml /opt/victronenergy/gui/qml/PageLynxIonIo.qml
++ head -n 1 /opt/victronenergy/version
+ versionStringToNumber v3.52
+ local p4=
+ local p5=
+ local p5=
+ local major=
+ local minor=
+ read major minor p4 p5 p6
++ echo v3.52
++ awk -v 'FS=[v.~-]' '{print $2, $3, $4, $5, $6}'
+ (( versionNumber = major * 1000000000 + minor * 1000000 ))
+ '[' -z ']'
+ (( versionNumber += 999 ))
+ '[' '!' -z ']'
+ '[' '!' -z ']'
+ (( venusVersionNumber = 3052000999 ))
+ versionStringToNumber v3.00~14
+ local p4=
+ local p5=
+ local p5=
+ local major=
+ local minor=
+ read major minor p4 p5 p6
++ echo v3.00~14
++ awk -v 'FS=[v.~-]' '{print $2, $3, $4, $5, $6}'
+ (( versionNumber = major * 1000000000 + minor * 1000000 ))
+ '[' -z 14 ']'
+ '[' 14 = large ']'
+ (( versionNumber += p4 ))
+ '[' '!' -z 14 ']'
+ '[' 14 = large ']'
+ '[' '!' -z ']'
+ qmlDir=/opt/victronenergy/gui/qml
+ ((  3052000999 < 3000000014  ))
+ echo done.
done.
+ '[' -d /opt/victronenergy/gui-v2 ']'
+ '[' -d /var/www/venus/gui-v2 ']'
+ '[' '!' -L /var/www/venus/gui-v2 ']'
+ pathGuiWww=/var/www/venus/gui-v2
+ '[' -f /var/www/venus/gui-v2/venus-gui-v2.wasm.sha256 ']'
++ cat /var/www/venus/gui-v2/venus-gui-v2.wasm.sha256
+ hash_installed='306405e8f3d62f85e7fa759c08156a6de008775fcd01fe7c3ae977f2aef9676b  venus-gui-v2.wasm'
++ curl -s https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery_gui-v2/refs/heads/master/venus-gui-v2.wasm.sha256
+ hash_online='306405e8f3d62f85e7fa759c08156a6de008775fcd01fe7c3ae977f2aef9676b  venus-gui-v2.wasm'
+ [[ 306405e8f3d62f85e7fa759c08156a6de008775fcd01fe7c3ae977f2aef9676b  venus-gui-v2.wasm == *\v\e\n\u\s\-\g\u\i\-\v\2\.\w\a\s\m* ]]
+ '[' '306405e8f3d62f85e7fa759c08156a6de008775fcd01fe7c3ae977f2aef9676b  venus-gui-v2.wasm' '!=' '306405e8f3d62f85e7fa759c08156a6de008775fcd01fe7c3ae977f2aef9676b  venus-gui-v2.wasm' ']'
+ '[' -f /data/etc/dbus-serialbattery/ext/venus-os_dbus-serialbattery_gui-v2/venus-gui-v2.wasm.sha256 ']'
+ hash_available='no hash available'
+ '[' '306405e8f3d62f85e7fa759c08156a6de008775fcd01fe7c3ae977f2aef9676b  venus-gui-v2.wasm' '!=' 'no hash available' ']'
+ echo ''

+ echo 'Installing GUIv2 web version...'
Installing GUIv2 web version...
+ '[' '!' -f /data/etc/dbus-serialbattery/ext/venus-os_dbus-serialbattery_gui-v2/venus-webassembly.zip ']'
+ echo 'ERROR: GUIv2 web version not found.'
ERROR: GUIv2 web version not found.
+ '[' 0 -gt 0 ']'

mr-manuel, 2024-12-18:
If the hash is equal, then it should be installed already once and therefore also already be downloaded once. I will not investigate further at this point.