mr-manuel/venus-os_dbus-serialbattery #133: JDB packet errors

pkeberly, 2024-12-17:
### Describe the bug

Driver log file indicates packet errors.  
Tried current release and nightly build, both show the errors

### How to reproduce

Install driver
reboot Cerbo
view log file

### Expected behavior

valid readings

### Driver version of the currently installed driver

v1.5.20241215

### Venus OS device type

Venus_GX

### Venus OS version

3.60~11

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

;------------ Setup for Luna's 8S energy battery ---------
;  12/16/2024 PWK  first connections, 300aH test
;  12/17/2024 PWK  driver update, bump top voltage to 3.52, tune charge ramp to get better balancing.

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description:
;     Calculates the minimum and maximum battery voltage based on the number of cells and their voltage.
;     ATTENTION: This does not prevent overcharging or overdischarging of individual cells.
;     To prevent overcharging and overdischarging, use these features (enabled by default):
[... 351 lines cut ...]
@4000000067619f7603659354 INFO:SerialBattery:> CVCM:     True
@4000000067619f76037477d4 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000067619f7603f45a1c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000067619f7604018534 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@4000000067619f76040e1fc4 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@4000000067619f76041c1214 INFO:SerialBattery:Serial Number/Unique Identifier: 39_304.0Ah
@4000000067619f7d1b37b46c ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 59 expected 45
@4000000067619f7d1b62661c INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.332 < 3.35
@4000000067619f881a033bac ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@4000000067619f881a12667c ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 15 expected 7
@4000000067619f881a29ceac INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.332 < 3.35
@4000000067619f91281a1ef4 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@4000000067619f912828f3d4 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 36 expected 7
@4000000067619f91283fd34c INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.332 < 3.35
@4000000067619f9421383044 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 54 expected 45
@4000000067619f94215316fc INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.333 < 3.35
@4000000067619f961abcfe5c ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
@4000000067619f961ad2d04c WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 192
@4000000067619f961aeac134 INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.332 < 3.35
@4000000067619f9826d05914 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067619f9826e6cf14 INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.332 < 3.35
@4000000067619f992770fa2c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000067619fab18423a5c ERROR:SerialBattery:>>> ERROR: No reply - returning [len:54/123]
@4000000067619fab18931594 INFO:SerialBattery:cell_voltages_good: True - min: 3.33 > 3.25 - max: 3.332 < 3.35
```

mr-manuel, 2024-12-17:
Did you check the FAQ?

https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#what-can-i-do-if-the-bms-communication-is-unstable

pkeberly, 2024-12-17:
Thanks, yes I did.  I just discovered that if I unplug the VE.Can cable to the RV-C tank level module, the driver and BMS communicates fine.  I use isolated ttl to USB for the BMS but I might be a missing ground between the 12V & 24V systems on the bench test.

closed 2024-12-17 by pkeberly