mr-manuel/venus-os_dbus-serialbattery #141: A Segmentation fault when using Bluetooth

RoRi3D, 2024-12-26:
### Describe the bug

I'm trying the bluetooth connection with my JK BMS but it is giving a "Segmentation fault" 
See screendump below
This reported data is repeating itself continuously (roughly every 18 sec).

Data only visible when running:
tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal

The issue seems to be:
./ run: line 12: 3631 Segmentation fault    python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:XXXXX
(I have replaced the MacAdress by XXX)

After this the bluetooth seems to restart (incl. pairing)
[image]

### How to reproduce

running:
tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal

### Expected behavior

I was not expecting a "Segmentation fault"

### Driver version of the currently installed driver

v1.5.20241215

### Venus OS device type

GX Card (integrated in Victron device)

### Venus OS version

V3.51

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
BLUETOOTH_BMS = Jkbms_Ble C8:xxxxx
```

### Relevant log output

```shell
Note: Serial number & Mac Address partly replaced by XXXXX manual

@40000000676d6cc62f2d1854 [^A^[[0;93m^BCHG^A^[[0m^B] Controller 68:XXXXXX Discovering: no
@40000000676d6cc62f2d2024 [^A^[[0;93m^BCHG^A^[[0m^B] Device C8:XXXXX ServicesResolved: no
@40000000676d6cc62f2d2bdc Successful disconnected
@40000000676d6ccb33b57aec Device C8:XXXXX (public)
@40000000676d6ccb33b5a5e4       Alias: JK_BD6A17S6P
@40000000676d6ccb33b5adb4       Paired: no
@40000000676d6ccb33b5b19c       Trusted: no
@40000000676d6ccb33b5b96c       Blocked: no
@40000000676d6ccb33b5bd54       Connected: no
@40000000676d6ccb33b5c13c       LegacyPairing: no
@40000000676d6ccb33b5c524       UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@40000000676d6ccb33b5d0dc       RSSI: -84
@40000000676d6ccb33c11f64
@40000000676d6cce2192a09c INFO:SerialBattery:
@40000000676d6cce219c5114 INFO:SerialBattery:Starting dbus-serialbattery
@40000000676d6cce21c31eac INFO:SerialBattery:Venus OS v3.51 running on MultiPlus-II^@
@40000000676d6cce21cd6f4c INFO:SerialBattery:dbus-serialbattery v1.5.20241215
@40000000676d6cce33a6f044 INFO:SerialBattery:Init of Jkbms_Ble at C8:XXXXX
@40000000676d6cce33b0a4a4 INFO:SerialBattery:Test of Jkbms_Ble at C8:XXXXX
@40000000676d6cd3167726ec INFO:SerialBattery:BAT: JKBMS 11.XA 16S (20241116)
@40000000676d6cd3168bcc14 INFO:SerialBattery:-- Connection established to Jkbms_Ble
@40000000676d6cd32a16aa74 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 1
@40000000676d6cd334de4764 INFO:SerialBattery:Use DeviceInstance: 1
@40000000676d6cd334f6190c INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ble_c8478010625c
@40000000676d6cd337061134 INFO:SerialBattery:Battery JKBMS BLE connected to dbus from ble_c8478010625c
@40000000676d6cd3378514d4 INFO:SerialBattery:========== Settings ==========
@40000000676d6cd33794f73c INFO:SerialBattery:> Connection voltage: 52.46 V | Current: 0.0 A | SoC: 76%
@40000000676d6cd337a1999c INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@40000000676d6cd337af64dc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000676d6cd337c53e9c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
@40000000676d6cd337d8f194 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
@40000000676d6cd337ed600c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 50.0 A (read from BMS)
@40000000676d6cd338001904 INFO:SerialBattery:> CVCM:     True
@40000000676d6cd338110cdc INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000676d6cd338317d8c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000676d6cd33852b18c INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000676d6cd3385fa9dc INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@40000000676d6cd3387f1ca4 INFO:SerialBattery:Serial Number/Unique Identifier: 401093XXXX
@40000000676d6cd514afd084 ./run: line 12:  8935 Segmentation fault      python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:XXXXXX
@40000000676d6cd51d122d44
@40000000676d6cd51d15751c
@40000000676d6cd51d157cec INFO:Preparing Bluetooth for connection to BMS
@40000000676d6cd51d17edec INFO:Bluetooth details
@40000000676d6cd51ffb6a84 [^A^[[0;93m^BCHG^A^[[0m^B] Device C8:XXXXX RSSI: -84
@40000000676d6cd51ffb9964 [^A^[[0;93m^BCHG^A^[[0m^B] Device C8:XXXXX RSSI: -84
```

mr-manuel, 2024-12-26:
See https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/86

closed 2024-12-26 by mr-manuel

RoRi3D, 2024-12-26:
Thank you !
The older version indeed seems (more) stable, for the last few minutes at least.