mr-manuel/venus-os_dbus-serialbattery #142: JK DZ-11 balancer

RichGorrod, 2024-12-27:
### Describe the problem

I have 4 x Seplos DIY batteries with Seplos V2 BMS in them. I have added a JK-DZ11-B2A24s balancer as the seplos BMS only has a 150mA balancer. The JK has a RS485 port on it that I trying to read the individual cell voltages on. I can communicated via Bluetooth but not via the RS485 port. - 
[JK-B1A24S-RS485-protocol-english.pdf](https://github.com/user-attachments/files/18256503/JK-B1A24S-RS485-protocol-english.pdf)

### Driver version of the currently installed driver

v1.5.20241119beta

### Driver version of the last known working driver

v1.5.20241119beta

### Venus OS device type

Raspberry Pi

### Venus OS version

V3.52~2

### BMS type

JKBMS (Heltec BMS), JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

4 280ah Batteries

### Connection type

Serial USB adapter to RS485

### Config file

```ini
; Example (remove the semicolon ";" to uncomment and activate the value/setting                                                                                                                                                                             >
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
MODBUS_ADDRESSES = 0x01, 0x02, 0x03, 0x04
BLUETOOTH_BMS = Jkbms_Ble C8:47:80:0C:04:5F, Jkbms_Ble C8:47:80:0B:DC:70, Jkbms                                                                                                                                                                             >
```

### Relevant log output

```shell
@40000000676dfbd538d8dd34 *** starting serial-starter ***
@40000000676dfbd61d2cecec serstart starting
@40000000676dfbd61e719634 INFO: loading config file /etc/venus/serial-starter.conf
@40000000676dfbd7351e07b4 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@40000000676dfd3b2f190b84 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@40000000676dfd4135988afc INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e02ba0517fe44 *** CCGX booted (0) ***
@40000000676e02ba296e9334 *** starting serial-starter ***
@40000000676e02bb214163bc serstart starting
@40000000676e02bb283ff674 INFO: loading config file /etc/venus/serial-starter.conf
@40000000676e02bd08dc0f0c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@40000000676e02bf36288274 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@40000000676e02f2194834c4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e03012e92c484 INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@40000000676e030737218874 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000676e030d1a1d5744 INFO: Create daemontools service gps-dbus.ttyUSB0
@40000000676e03131e2a7b14 INFO: Start service gps-dbus.ttyUSB0 once
@40000000676e031c3659cf8c INFO: Create daemontools service vedirect-interface.ttyUSB0
@40000000676e0322391af32c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000676e03260e76e374 INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e06691728fe44 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000676e066e2703f404 INFO: Start service gps-dbus.ttyUSB0 once
@40000000676e067739a4c46c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000676e067d0d87c884 INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e076d2b25e024 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000676e07730051b404 INFO: Start service gps-dbus.ttyUSB0 once
@40000000676e077c130f630c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000676e078122468b34 INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e096407a10a84 *** CCGX booted (0) ***
@40000000676e0964348a959c *** starting serial-starter ***
@40000000676e09651ec0c3bc serstart starting
@40000000676e09651fc14f84 INFO: loading config file /etc/venus/serial-starter.conf
@40000000676e0966364da25c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@40000000676e099504fa9ad4 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@40000000676e099b17f30cd4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e09aa3786fd1c INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@40000000676e09b103cea1dc INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000676e09b6384c29ac INFO: Create daemontools service gps-dbus.ttyUSB0
@40000000676e09bd06f8149c INFO: Start service gps-dbus.ttyUSB0 once
@40000000676e09c61b3ae0ec INFO: Create daemontools service vedirect-interface.ttyUSB0
@40000000676e09cc1e2fa364 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000676e09cf2f895d6c INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e0bf00761270c *** CCGX booted (0) ***
@40000000676e0bf0336e0dc4 *** starting serial-starter ***
@40000000676e0bf1303bc54c serstart starting
@40000000676e0bf134a19c24 INFO: loading config file /etc/venus/serial-starter.conf
@40000000676e0bf30234144c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@40000000676e0bf60bf7d1e4 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@40000000676e0bfc25b29ca4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000676e0c0d175620f4 INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@40000000676e0c131e14c864 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000676e0c4908090f6c INFO: Create daemontools service gps-dbus.ttyUSB0
@40000000676e0c4f143b2554 INFO: Start service gps-dbus.ttyUSB0 once
@40000000676e0c5826d5a0a4 INFO: Create daemontools service vedirect-interface.ttyUSB0
@40000000676e0c5e2a6a40e4 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000676e0c613aed1b1c INFO: Start service dbus-serialbattery.ttyUSB0 once
```

### Any other information that may be helpful

In an ideally world would like the seplos BMS to connect via can bus to the cerbo (Which they are doing now) and get the individual cell voltages from the JK balancer .

Many thanks for taking the time to look at this and for a fantastic project.

mr-manuel, 2024-12-27:
You asked that already in the past https://github.com/Louisvdw/dbus-serialbattery/discussions/1090.

Since it‘s a very edge case this will not be supported. It would take more money and time to add this, than swapping the BMS for you.

closed 2024-12-27 by mr-manuel

RichGorrod, 2024-12-27:
ok no probs. Will go down another route. - Probably leave them stand alone.

thanks again for all your great work with the project.

Rich