mr-manuel/venus-os_dbus-serialbattery #145: BatteryOperationalLimits/BatteryLowVoltage being update from dvcc.py with wrong value

nicobresseleers, 2024-12-28:
### Describe the problem

As soon as i start serialbattery I can see via dbus-spy that the BatteryOperationalLimits/BatteryLowVoltage on my multiplus is being set to 655.34 ( where i expect  1,9 / 2,1 *22 = 41,8/46 as "low voltage") , and triggering a "low battery" alarm on the portal

[image]

### Driver version of the currently installed driver

v1.5.20241215

### Venus OS device type

Raspberry Pi

### Venus OS version

3.52 2

### BMS type

JKBMS (Heltec BMS)

### Cell count

22

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 25.0
MAX_BATTERY_DISCHARGE_CURRENT = 40.0

; --------- Cell Voltages ---------
; Description:
;     Minimum and maximum cell voltages used for:
;         - Calculating the minimum and maximum battery voltage
;         - Triggering the SoC reset when SoC calculation is enabled
; Example:
;     16 cells * 3.45 V/cell = 55.2 V max charge voltage
;     16 cells * 2.90 V/cell = 46.4 V min discharge voltage
; Minimum voltage (can be seen as cut-off voltage)
MIN_CELL_VOLTAGE   = 1.900
; Maximum voltage (can be seen as absorption voltage)
MAX_CELL_VOLTAGE   = 2.650
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 2.550

; --------- SoC Reset Voltage (must match BMS settings) ---------
; +++ This section is independent and unrelated to the "SoC calculation" section below. +++
; This is used to reset the SoC to 100% periodically to correct SoC drift.
[... 465 lines cut ...]
; Use min/max cell voltage, CVL, CCL, and DCL values from the BMS.
SEPLOS_USE_BMS_VALUES = False
```

### Relevant log output

```shell
Don't see a log where the update is done, cannot debug dbus itself ( set a changepointer or breakpoint)
```

### Any other information that may be helpful

I am running 3x JK-BD4A24S-4P on 3 "CarboCaps" : 22 LTO cells from the manufacturer Yinlong, each with 40A

JK-BD4A24S-4P is connected via RS485 to 3x USB( 2+3+4), all works, just get these alarms on low voltage

Info/Config/MIN_CELL_VOLTAGE    is 1.9 via dbus-spy (correct) x22= 41.8
also
Info/MaxChargeVoltage is 61,49 equals the BatteryOperationalLimits/MaxChargeVoltage (61.5)

Info/BatteryLowVoltage   is "empty" in dbus-spy.. not sure that's normal

I have no clue where the 655,3499755859 V in BatteryOperationalLimits/BatteryLowVoltage comes from .. could not trace it back nor calculate it in one way or another.

I did stop the serial process, adapted value via dbus-spy and it did not change, as soon as i started serial again numbers where updated + also running aggregate on top, but also turned that one of for above test.

mr-manuel, 2024-12-29:
Please post also the log as requested when you opened the issue.

What do you see in the GUI under the Cell and Parameters page? Please post a Screenshot.

nicobresseleers, 2024-12-29:
I think i found the error... I had a -20° Celsius reading on 1 battery (C3) ( I assume I have a bad connection to the sensor, need to check in detail later)

If it measures negative values I also assume we block all from discharing and charging, ( and maybe also calculate wrong the min voltage?) ... maybe if you simulate with -50 we reach 1000v? :-)

it's still 655.35:
[image]

but as soon as I set 
CCCM_T_ENABLE = True -> False
DCCM_T_ENABLE = True -> False
the system became active and ignored the negative setting to not move at all... 

let me know what log you consider relevant for the issue and will be happy to post it.

screens:
Cell:
[image]

Parameters:
[image]
[image]
[image]

mr-manuel, 2024-12-29:
I was able to reproduce this issue on multiple systems, but have no additional informations yet.

nicobresseleers, 2024-12-29:
For the moment I turned of logging... systems seems to work as before  ( battery is in cellar.. will never be below 10 degrees Celsius.. not really an issue running without the temperature part)

nicobresseleers, 2024-12-29:
1 side effect ( you cannot help with but sharing anyway):

aggregate is not working since it does not get the parameters:

INFO:root:Sun Dec 29 21:16:01 2024: MultiPlus-II 48/3000/35-32 found.
ERROR:root:Sun Dec 29 21:16:02 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery SerialBattery(JKBMS)C2.
ERROR:root:Read trial nr. 1
......
ERROR:root:Sun Dec 29 21:16:12 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery SerialBattery(JKBMS)C2.
ERROR:root:Read trial nr. 11
ERROR:root:Sun Dec 29 21:16:12 2024: DBus read failed. Exiting.

Created https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/77 to check

mr-manuel, 2024-12-29:
Found the issue. The problem will be fixed in the next dev and stable version. If you wan't to fix it in the meanwhile add this line

```python
        self._dbusservice["/Info/BatteryLowVoltage"] = self.battery.min_battery_voltage
```

below this line https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/a7d7868555b37a4ba5c1f7ffd1df518fbbe72c0a/dbus-serialbattery/dbushelper.py#L967

mr-manuel, 2024-12-29:
Anyway, the value `655.35` seems to be some sort of `2^16` if `/Info/BatteryLowVoltage` is empty. I reported it also to Victron.

nicobresseleers, 2024-12-29:
don't even have "#Voltage control" in my local release :-)

Anyway will install the next version... 

This however does not fix my -20°C issue ( seems i cannot ignore it from the settings, it always takes the worse from the 4 temps?) 
if I turn of on the BMS sensors ( via the app.. this is a setting) .. it's even worse -> all become -20°C!
anything I can overwrite from a temp POV?

Honestly , I have seen MANY issue on the forums with unexplainable "Low Voltage alarms" .. we might just have found it!

mr-manuel, 2024-12-30:
If you are using JKBMS not the PB Model it has only two temperature sensors. In the config you can choose to use only one sensor for battery temp instead of all. What does the JKBMS app say?

nicobresseleers, 2024-12-30:
I have 2 sensors for sure... installed them myself, must have a bad connection ( had to extend the wires into the battery)
I can show for the other batteries;

C1: 2 temps
[image]
C2: 2 temps:
[image]
C3 : 3 temps, but 1 with -20 ( i assume a bad connection to the sensor)
[image]

Once I change to only select one sensor the one working, aggregate will fail  ( no update on that bug for now: https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/77) 

Is there a way to select the , let's say 1 sensor and you populate the 4 with the same value? ( they seem to have "None" on dbus)

mr-manuel, 2024-12-30:
No there is no way to select one temperature to display it as the others, since that would make no sense. The only reasonable solution is to fix the connected sensor.

nicobresseleers, 2024-12-30:
I get same values if I disable the temperature in the BMS app ... ( honestly i don't care about temp: it's LTO .. they work from -40 till +50)

they are in the basement at 15 degrees..

closed 2024-12-31 by mr-manuel

nicobresseleers, 2025-01-02:
I installed today the beta version ( nightly build will not run , pointing to wrong directories)

still have a high number in BatteryLowVoltage:

[image]

did you release the fix the beta release already?

commit b64ba0ea (2024-12-31): Fix #145 `/Info/BatteryLowVoltage` remaining `None`