mr-manuel/venus-os_dbus-serialbattery #149: JK BMS B2A8Sxxx CAN doesn't get detected as battery on raspi + CAN-hat

haraldhh, 2025-01-01:
### Describe the bug

I have connected my JK BMS 4-8S with it's CAN-port to a CAN-hat on a raspberry pi 3b+.

I can see a can9 interface, and when I configure the BMS to use a 500 kbps protocol and "CAN-bus BMS LV (500 kbit/s)" in Venus under Settings -> Services the Network Status shows "ERROR-ACTIVE" as state (which is a good thing as far as I can understand). I can see packets coming, droprate 0.25%. 

But no (additional) battery shows up. I do have the BMS connected with a RS485 - USB as well. 

canshow shows a lot of data coming in. 

I have also tried with other 250 kbps protocols to no avail. 

```
root@raspberrypi2:~# canshow 
000.051 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.052 < 0.0.0.00.04.F4| 0x00 0x0D 0x01 0x00 0x0D 0x04 0x00 0x00
000.053 < 6.0.0.F1.28.F4| 0xEE 0x0A 0xF0 0x0A 0x7B 0x24 0x03 0x00
000.134 [ 0.0.0.00.03.05| 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
000.135 [ 0.0.0.00.03.07| 0x12 0x34 0x56 0x78 0x56 0x49 0x43 0x00
000.151 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.152 < 0.0.0.00.04.F4| 0x01 0x0D 0x01 0x00 0x0D 0x04 0x00 0x00
000.251 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.252 < 0.0.0.00.04.F4| 0x00 0x0D 0x01 0x00 0x0D 0x04 0x00 0x00
000.253 < 6.0.0.F1.28.F4| 0xEE 0x0A 0xF0 0x0A 0x7B 0x24 0x03 0x00
000.254 < 6.0.0.F5.28.F4| 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00
000.255 < 6.0.0.06.E5.F4| 0x00 0x91 0x03 0xB6 0x00 0x00 0x00 0x00
000.256 < 0.0.0.00.05.F4| 0x41 0x00 0x41 0x01 0x41 0x00 0x00 0x00
000.257 < 6.0.0.F2.28.F4| 0x04 0x41 0x41 0x43 0x00 0x00 0x00 0x00
000.258 < 6.0.0.F4.28.F4| 0xD6 0x70 0x2C 0x00 0x00 0x00 0x64 0x00
000.351 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.352 < 0.0.0.00.04.F4| 0x00 0x0D 0x01 0x00 0x0D 0x04 0x00 0x00
000.451 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.452 < 0.0.0.00.04.F4| 0xFE 0x0C 0x01 0x00 0x0D 0x04 0x00 0x00
000.453 < 6.0.0.F1.28.F4| 0xEE 0x0A 0xF0 0x0A 0x7B 0x24 0x03 0x00
000.551 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.552 < 0.0.0.00.04.F4| 0xFE 0x0C 0x01 0xFE 0x0C 0x04 0x00 0x00
000.651 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.652 < 0.0.0.00.04.F4| 0x00 0x0D 0x01 0x00 0x0D 0x04 0x00 0x00
000.653 < 6.0.0.F1.28.F4| 0xEE 0x0A 0xF0 0x0A 0x7B 0x24 0x03 0x00
000.751 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00
000.752 < 0.0.0.00.04.F4| 0x00 0x0D 0x01 0x00 0x0D 0x04 0x00 0x00
000.753 < 6.0.0.F5.28.F4| 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00
000.754 < 6.0.0.06.E5.F4| 0x00 0x91 0x03 0xB6 0x00 0x00 0x00 0x00
000.755 < 6.0.0.E0.28.F4| 0x00 0x0D 0x00 0x0D 0xFE 0x0C 0x00 0x0D
000.851 < 0.0.0.00.02.F4| 0x85 0x00 0xA0 0x0F 0x64 0x00 0x00 0x00

```

### How to reproduce

1. Connect a 4-8S JK BMS with CAN to a CAN-hat on a raspberry
2.  Expect the battery to show up

### Expected behavior

I would expect the battery to show up. 

### Driver version of the currently installed driver

DRIVER_VERSION: str = "1.5.20241215"

### Venus OS device type

Raspberry Pi 3

### Venus OS version

3.52

### BMS type

JKBMS (Heltec BMS)

### Cell count

4

### Battery count

1

### Connection type

CAN

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; 2025-01-01 Harald
CAN_PORT = can8
```

### Relevant log output

```shell
2025-01-01 15:39:39.320714500 
2025-01-01 15:39:42.355391500 INFO:SerialBattery:
2025-01-01 15:39:42.355896500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-01 15:39:42.358161500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-01 15:39:42.358666500 INFO:SerialBattery:dbus-serialbattery v1.5.20241215
2025-01-01 15:39:43.886396500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-01 15:39:43.886864500 INFO:SerialBattery:Testing Daly_Can
2025-01-01 15:44:54.240946500 ERROR:SerialBattery:Exception occurred: CanError('Error receiving: [Errno 100] Network is down') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/daly_can.py line #97
2025-01-01 15:44:54.241664500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-01 15:44:54.248067500 ERROR:SerialBattery:Exception occurred: CanError('Error receiving: [Errno 100] Network is down') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_can.py line #194
2025-01-01 15:44:54.749437500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-01-01 15:44:54.749922500 INFO:SerialBattery:Testing Daly_Can
2025-01-01 15:46:10.159902500 ERROR:SerialBattery:Exception occurred: CanError('Error receiving: [Errno 100] Network is down') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/daly_can.py line #97
2025-01-01 15:46:10.160663500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-01 15:46:10.168412500 ERROR:SerialBattery:Exception occurred: CanError('Error receiving: [Errno 100] Network is down') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_can.py line #194
2025-01-01 15:46:10.668450500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2025-01-01 15:46:10.669002500 INFO:SerialBattery:Testing Daly_Can
2025-01-01 15:47:11.438728500 ERROR:SerialBattery:Exception occurred: CanError('Error receiving: [Errno 100] Network is down') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/daly_can.py line #97
2025-01-01 15:47:11.438743500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-01 15:47:11.442940500 ERROR:SerialBattery:Exception occurred: CanError('Error receiving: [Errno 100] Network is down') of type <class 'can.CanError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_can.py line #194
2025-01-01 15:47:11.944295500 ERROR:SerialBattery:ERROR >>> No battery connection at can8
2025-01-01 15:47:12.319060500 
2025-01-01 15:47:15.283787500 INFO:SerialBattery:
2025-01-01 15:47:15.284286500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-01 15:47:15.286306500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-01 15:47:15.286745500 INFO:SerialBattery:dbus-serialbattery v1.5.20241215
2025-01-01 15:47:16.778154500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-01 15:47:16.778584500 INFO:SerialBattery:Testing Daly_Can
^C
```

haraldhh, 2025-01-01:
If I choose "VE.Can & CAN-bus BMS (250 kbit/s)" in Venus, and "JK BMS CAN Protocol (250K) V2.0" in the BMS I also get ERROR-ACTIVE in Network status, and canshow shows packets flowing.

mr-manuel, 2025-01-01:
It seems you did not read the important nodes when opening the issue. You did not try the nightly.

Anyway, why are you writing can9 and in the config you have can8?

haraldhh, 2025-01-01:
Thanks, I'll try nightly. can9 was a typo.

haraldhh, 2025-01-01:
```
2025-01-01 16:16:39.956607500 *** CCGX booted (0) ***
2025-01-01 16:17:39.330300500 
2025-01-01 16:17:44.338942500 INFO:SerialBattery:
2025-01-01 16:17:44.339375500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-01 16:17:44.341301500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-01 16:17:44.341725500 INFO:SerialBattery:dbus-serialbattery v1.5.20241215
2025-01-01 16:17:46.883163500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-01 16:17:46.883816500 INFO:SerialBattery:Testing Daly_Can
2025-01-01 16:18:48.459601500 INFO:SerialBattery:
2025-01-01 16:18:48.460000500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-01 16:18:48.462129500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-01 16:18:48.462540500 INFO:SerialBattery:dbus-serialbattery v2.0.20241231dev
2025-01-01 16:18:50.232825500 INFO:SerialBattery:Detected CAN Bus bitrate: 250 kbps
2025-01-01 16:18:52.051819500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-01 16:18:52.054064500 INFO:SerialBattery:Testing Daly_Can
2025-01-01 16:18:52.160332500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-01 16:18:52.161110500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-01 16:18:52.164596500 INFO:SerialBattery:-- Connection established to Jkbms_Can
2025-01-01 16:18:53.961087500 INFO:SerialBattery:Use DeviceInstance: 2
2025-01-01 16:18:53.963676500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.can8
2025-01-01 16:18:54.024055500 INFO:SerialBattery:Polling interval: 1.000 s
2025-01-01 16:18:54.026912500 INFO:SerialBattery:Battery JKBMS CAN V2 connected to dbus from can8
2025-01-01 16:18:54.027479500 INFO:SerialBattery:========== Settings ==========
2025-01-01 16:18:54.028092500 INFO:SerialBattery:> Connection voltage: 13.300999999999998 V | Current: 0.0 A | SoC: 100%
2025-01-01 16:18:54.028680500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2025-01-01 16:18:54.029200500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2025-01-01 16:18:54.029938500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2025-01-01 16:18:54.030511500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A
2025-01-01 16:18:54.031001500 INFO:SerialBattery:> CVCM:     True
2025-01-01 16:18:54.031564500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-01-01 16:18:54.032117500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-01-01 16:18:54.032651500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-01-01 16:18:54.033178500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: None
2025-01-01 16:18:54.033688500 INFO:SerialBattery:Serial Number/Unique Identifier: can8
2025-01-01 16:19:11.027581500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-01-01 16:19:11.127452500 INFO:SerialBattery:CAN receiver stopped
2025-01-01 16:19:11.127460500 INFO:SerialBattery:Stopped dbus-serialbattery with exit code 0
2025-01-01 16:19:22.152568500 *** CCGX booted (0) ***

```

Thank You and sorry. Happy new year!

mr-manuel, 2025-01-01:
Happy new year too

closed 2025-01-01 by mr-manuel