mr-manuel/venus-os_dbus-serialbattery #151: Crash of 1 serial connector, other 2 keep running

nicobresseleers, 2025-01-03:
### Describe the problem

from current:

@4000000067775d383a00d6bc INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000067775d383a0923bc INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
@4000000067775d383a12b10c INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@4000000067775d383a490f04 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
@4000000067775d383a53ffcc INFO:SerialBattery:Serial Number/Unique Identifier: C1
@4000000067775edc0101ca24 ERROR:SerialBattery:Exception occurred: OSError(5, 'Input/output error') of type <class 'OSError'> in /opt/victronenergy/dbus-serialbattery/utils.py line #694
@4000000067775edc01838ce4 Traceback (most recent call last):
@4000000067775edc0183a454   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 698, in publish_battery
@4000000067775edc0183b7dc     result = self.battery.refresh_data()
@4000000067775edc0183c394   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 79, in refresh_data
@4000000067775edc01851b54     return self.read_status_data()
@4000000067775edc0185270c   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 90, in read_status_data
@4000000067775edc0186c51c     status_data = self.read_serial_data_jkbms(self.command_status)
@4000000067775edc0186d8a4   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms.py", line 335, in read_serial_data_jkbms
@4000000067775edc0186ec2c     start, length = unpack_from(">HH", data)
@4000000067775edc0186f7e4 TypeError: a bytes-like object is required, not 'NoneType'
@400000006777a6ab14d10c54 *** CCGX booted (0) ***

### Driver version of the currently installed driver

1.5.20241119beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.53~2

### BMS type

JKBMS (Heltec BMS)

### Cell count

22

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
root@raspberrypi2:/data/etc/dbus-serialbattery# cat config.ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings, and info messages are logged
; DEBUG: Errors, warnings, info, and debug messages are logged
LOGGING = INFO

; --------- Battery Current Limits ---------
; +++ Limits apply to each individual battery/BMS. +++
; +++ If you have multiple batteries, you need a battery aggregator. +++
MAX_BATTERY_CHARGE_CURRENT    = 40.0
MAX_BATTERY_DISCHARGE_CURRENT = 40.0

; --------- Cell Voltages ---------
[... 512 lines cut ...]
@400000006777b2f420e6a15c INFO:SerialBattery:Starting dbus-serialbattery
@400000006777b2f42101de04 INFO:SerialBattery:Venus OS v3.53~2
@400000006777b2f4210a7924 INFO:SerialBattery:dbus-serialbattery v1.5.20241119beta

last line tai64nlocal:
root@raspberrypi2:/data/etc/dbus-serialbattery# tai64nlocal
@400000006777b2f4210a7924 INFO:SerialBattery:dbus-serialbattery v1.5.20241119bet
2025-01-03 09:50:34.554334500 INFO:SerialBattery:dbus-serialbattery v1.5.20241119bet
```

### Any other information that may be helpful

the other 2 serials are running fine.. could it be hardware ( 
[image]
i already ordered 3 new RS485 -> USB converters since i keep getting errors with the pl2303 chipset from kernel ( dmesg)

[ 3339.811729] pl2303 ttyUSB1: error sending break = -32
[ 3339.826722] pl2303 ttyUSB1: pl2303_get_line_request - failed: -32
[ 3339.834713] pl2303 ttyUSB1: pl2303_get_line_request - failed: -32
[ 3340.021626] pl2303 ttyUSB4: error sending break = -32
[ 3340.236727] pl2303 ttyUSB2: pl2303_get_line_request - failed: -32
[ 3340.471475] pl2303 ttyUSB2: error sending break = -32
[ 3340.789597] pl2303 ttyUSB4: pl2303_get_line_request - failed: -32
[ 3341.021478] pl2303 ttyUSB4: error sending break = -32
[ 3341.237860] pl2303 ttyUSB2: pl2303_get_line_request - failed: -32

nicobresseleers, 2025-01-03:
tried the nightly build .. but that one I cannot installer , errors with /dev/ ?

ThomasDesloovere, 2025-01-03:
I think i am having the same problem but i did not look into logs my pi works fine and then about once every 2 days it loses serial connection to all devices (sometimes not completely to my JKbms and solar charger but the BMV712 gets disconnected) this only started happening when i installed this driver.

mr-manuel, 2025-01-03:
Please do not merge to many problems into one issue. This only makes it harder to troubleshoot and understand the problem.

So first of all: When you install the nightly, what errors do you get when installing? Did you read the changelog? The path `/opt/victronenergy/dbus-serialbattery` is completely wrong.

mr-manuel, 2025-01-03:
If you did not already, check also the FAQ section [What can I do, if the BMS communication is unstable?](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#what-can-i-do-if-the-bms-communication-is-unstable)

nicobresseleers, 2025-01-03:
> If you did not already, check also the FAQ section [What can I do, if the BMS communication is unstable?](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#what-can-i-do-if-the-bms-communication-is-unstable)

all is fine except for :
Cheap USB Hubs: Make sure you are using a qualitative USB Hub with enough power.

-> I ordered with a different chipset, same one as I am using to read my Elec meter ( that never has issues)

will keep you posted. just a shame it crashes...

mr-manuel, 2025-01-03:
Even better of a USB hub, would be one USB to 4x UART or RS485 adapter.

nicobresseleers, 2025-01-03:
> Please do not merge to many problems into one issue. This only makes it harder to troubleshoot and understand the problem.
>
> So first of all: When you install the nightly, what errors do you get when [...]

created a new one: https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/153

ThomasDesloovere, 2025-01-03:
Yeah same here... and no its not the cables or anything like that because everything has worked for the past 3 years really stable no problems and now I installed this driver the connection to some serial devices gets lost every couple of days.

nicobresseleers, 2025-01-03:
support/compatibility  for the PL2303 is not that great on PI i am afraid
could be related to these errors as they pile up... but not sure

[14136.618273] pl2303 ttyUSB2: error sending break = -32
[14136.638390] pl2303 ttyUSB3: error sending break = -32
[14136.794374] pl2303 ttyUSB4: pl2303_get_line_request - failed: -32
[14137.028263] pl2303 ttyUSB4: error sending break = -32
[14137.385628] pl2303 ttyUSB2: pl2303_get_line_request - failed: -32
[14137.412002] pl2303 ttyUSB3: pl2303_get_line_request - failed: -32
[14137.618265] pl2303 ttyUSB2: error sending break = -32
[14137.638516] pl2303 ttyUSB3: error sending break = -32
[14137.795881] pl2303 ttyUSB4: pl2303_get_line_request - failed: -32

have a new converter hopefully next week ( hope it solves it) if not: will keep digging.

nicobresseleers, 2025-01-03:
night build:

2025-01-03 14:59:54.302582500 *** CCGX booted (0) ***
2025-01-03 15:01:30.388858500 INFO:SerialBattery:
2025-01-03 15:01:30.389325500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-03 15:01:30.391812500 INFO:SerialBattery:Venus OS v3.53~2 running on Raspberry Pi 3 Model B Rev 1.2
2025-01-03 15:01:30.392332500 INFO:SerialBattery:dbus-serialbattery v2.0.20250103dev
2025-01-03 15:01:46.407251500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-03 15:01:46.407975500 INFO:SerialBattery:Testing Jkbms
2025-01-03 15:01:46.659495500 INFO:SerialBattery:-- Connection established to Jkbms
2025-01-03 15:01:48.488236500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 4
2025-01-03 15:01:48.864193500 INFO:SerialBattery:Use DeviceInstance: 4
2025-01-03 15:01:48.865774500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB4
2025-01-03 15:01:48.967291500 INFO:SerialBattery:Polling interval: 1.000 s
2025-01-03 15:01:48.970010500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB4
2025-01-03 15:01:48.971307500 INFO:SerialBattery:========== Settings ==========
2025-01-03 15:01:48.972048500 INFO:SerialBattery:> Connection voltage: 53.76 V | Current: -6.99 A | SoC: 99%
2025-01-03 15:01:48.972744500 INFO:SerialBattery:> Cell count: 22 | Cells populated: 22
2025-01-03 15:01:48.973311500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2025-01-03 15:01:48.973977500 INFO:SerialBattery:> MIN CELL VOLTAGE: 1.900 V | MAX CELL VOLTAGE: 2.650 V| FLOAT CELL VOLTAGE: 2.550 V
2025-01-03 15:01:48.974755500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 40.0 A
2025-01-03 15:01:48.975507500 INFO:SerialBattery:> CVCM:     True
2025-01-03 15:01:48.976947500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-01-03 15:01:48.977015500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2025-01-03 15:01:48.978237500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-01-03 15:01:48.979226500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-01-03 15:01:48.979993500 INFO:SerialBattery:Serial Number/Unique Identifier: C3
2025-01-03 15:02:36.878002500 ERROR:SerialBattery:Exception occurred: OSError(5, 'Input/output error') of type <class 'OSError'> in /data/apps/dbus-serialbattery/utils.py line #694
2025-01-03 15:02:36.889737500 Traceback (most recent call last):
2025-01-03 15:02:36.889743500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 767, in publish_battery
2025-01-03 15:02:36.889748500     result = self.battery.refresh_data()
2025-01-03 15:02:36.889751500   File "/data/apps/dbus-serialbattery/bms/jkbms.py", line 80, in refresh_data
2025-01-03 15:02:36.889756500     return self.read_status_data()
2025-01-03 15:02:36.889758500   File "/data/apps/dbus-serialbattery/bms/jkbms.py", line 91, in read_status_data
2025-01-03 15:02:36.889763500     status_data = self.read_serial_data_jkbms(self.command_status)
2025-01-03 15:02:36.889854500   File "/data/apps/dbus-serialbattery/bms/jkbms.py", line 336, in read_serial_data_jkbms
2025-01-03 15:02:36.889860500     start, length = unpack_from(">HH", data)
2025-01-03 15:02:36.889863500 TypeError: a bytes-like object is required, not 'NoneType'
2025-01-03 15:03:23.264302500 INFO:SerialBattery:
2025-01-03 15:03:23.264763500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-03 15:03:23.267188500 INFO:SerialBattery:Venus OS v3.53~2 running on Raspberry Pi 3 Model B Rev 1.2
2025-01-03 15:03:23.267671500 INFO:SerialBattery:dbus-serialbattery v2.0.20250103dev
2025-01-03 15:03:39.284675500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-03 15:03:39.285161500 INFO:SerialBattery:Testing Jkbms
2025-01-03 15:03:39.549146500 INFO:SerialBattery:-- Connection established to Jkbms
2025-01-03 15:03:41.103528500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 4
2025-01-03 15:03:41.346652500 INFO:SerialBattery:Use DeviceInstance: 4
2025-01-03 15:03:41.348176500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB4
2025-01-03 15:03:41.426310500 INFO:SerialBattery:Polling interval: 1.000 s
2025-01-03 15:03:41.426316500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB4
2025-01-03 15:03:41.426321500 INFO:SerialBattery:========== Settings ==========
2025-01-03 15:03:41.426324500 INFO:SerialBattery:> Connection voltage: 53.43 V | Current: -7.68 A | SoC: 98%
2025-01-03 15:03:41.426329500 INFO:SerialBattery:> Cell count: 22 | Cells populated: 22
2025-01-03 15:03:41.426332500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2025-01-03 15:03:41.426336500 INFO:SerialBattery:> MIN CELL VOLTAGE: 1.900 V | MAX CELL VOLTAGE: 2.650 V| FLOAT CELL VOLTAGE: 2.550 V
2025-01-03 15:03:41.426428500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 40.0 A
2025-01-03 15:03:41.446248500 INFO:SerialBattery:> CVCM:     True
2025-01-03 15:03:41.446252500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-01-03 15:03:41.446256500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2025-01-03 15:03:41.446259500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-01-03 15:03:41.446263500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2025-01-03 15:03:41.446268500 INFO:SerialBattery:Serial Number/Unique Identifier: C3
root@raspberrypi2:/data/log/dbus-serialbattery.ttyUSB4#

mr-manuel, 2025-01-03:
@ThomasDesloovere in your case you could check the CPU utilization.

ThomasDesloovere, 2025-01-03:
So I installed htop and i could see that nodered was using quite a lot of resources and when I open nodered from the VRM the BMV 712 lost connection again and 2 minutes later the connection was reestablished. Sometimes the connection to my JK-bms also gets lost but this was not the case now.

[image]
[image]
[image]

mr-manuel, 2025-01-03:
You have another issue, or do you see the same in the log? If you don't see the same in the log, please stop posting in this issue.

nicobresseleers, 2025-01-05:
As promised an update:

After replacing the cheap (2,25€) pl2302 with the even cheaper( 2,16€)  FT232:

lsusb -t- v
        |__ Port 5: Dev 6, If 0, Class=Hub, Driver=hub/4p, 12M
            ID 0a05:7211 Unknown Manufacturer hub
            |__ Port 2: Dev 7, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
            |__ Port 3: Dev 8, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
            |__ Port 4: Dev 9, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
root@raspberrypi2:~#

NO more errors from kernel / dmesg AND for now all seems stable!

I would stay away from cheap PL2303:
[image]

and us FT232/CH340 instead
[image]

as it seems more stable on the RPi

Could be the stability on the OS also... not necessary the device itself ( as i have used them previously in other project successful)

As always keeping an eye on the system and will keep you posted! (running currently nightly build: v2.0.20250103dev)

Not sure you want to investigate if we can do more checks to avoid errors, if not feel free to close.

Regards
Nico

mr-manuel, 2025-01-05:
I'm waiting for a collaboration, then I can provide direct links to high quality adapters. Anyway I will add to the docs, that FTDI chips are recommended.

closed 2025-01-05 by mr-manuel