mr-manuel/venus-os_dbus-serialbattery #153: Night build fails

nicobresseleers, 2025-01-03:
### Describe the problem

Install is going fine:

root@raspberrypi2:/data/log/dbus-systemcalc-py# wget -O /tmp/install.sh https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/master/dbus-serialbattery/install.sh
--2025-01-03 13:28:02--  https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/master/dbus-serialbattery/install.sh
Resolving raw.githubusercontent.com... 2606:50c0:8003::154, 2606:50c0:8002::154, 2606:50c0:8000::154, ...
Connecting to raw.githubusercontent.com|2606:50c0:8003::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 15306 (15K) [text/plain]
Saving to: ‘/tmp/install.sh’

/tmp/install.sh                                                               100%[==============================================================================================================================================================================================>]  14.95K  --.-KB/s    in 0.004s

2025-01-03 13:28:02 (3.30 MB/s) - ‘/tmp/install.sh’ saved [15306/15306]

root@raspberrypi2:/data/log/dbus-systemcalc-py#
root@raspberrypi2:/data/log/dbus-systemcalc-py# bash /tmp/install.sh

*** Welcome to the dbus-serialbattery installer from mr-manuel! ***

Fetch available version numbers... done.

** Currently installed version: v1.5.20241215 **

1) stable release "v1.5.20241215" (stable, most up to date)                                 4) specific branch (specific feature testing)                                               7) quit
2) beta build "v1.5.20241119beta" (no errors after 72 h runtime, long time testing needed)  5) specific version
3) nightly build "v2.0.20250103dev" (newest features and fixes, bugs possible)              6) local tar file

Select which version you want to install from m-rmanuel's repo and enter the corresponding number: 3
> Selected: nightly build "v2.0.20250103dev" (newest features and fixes, bugs possible)

Config.ini backed up to /data/etc/dbus-serialbattery_config.ini.backup
--2025-01-03 13:28:11--  https://github.com/mr-manuel/venus-os_dbus-serialbattery/archive/refs/heads/master.zip
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/mr-manuel/venus-os_dbus-serialbattery/zip/refs/heads/master [following]
--2025-01-03 13:28:11--  https://codeload.github.com/mr-manuel/venus-os_dbus-serialbattery/zip/refs/heads/master
Resolving codeload.github.com... 140.82.121.9
Connecting to codeload.github.com|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘/tmp/master.zip’

/tmp/master.zip                                                                   [                     <=>                                                                                                                                                                       ]  14.22M  3.66MB/s    in 4.3s

2025-01-03 13:28:16 (3.34 MB/s) - ‘/tmp/master.zip’ saved [14916409]

Config.ini restored to /data/apps/dbus-serialbattery/config.ini

Remove old dbus-serialbattery folders on root fs...
The app "dbus-serialbattery" was already added to the directory "/opt/victronenergy/service-templates" in the config file.
The app "dbus-serialbattery_gui" was already added to the directory "/opt/victronenergy/gui" in the config file.
The app "dbus-serialbattery_gui" was already added to the directory "/var/www/venus" in the config file.

Installing QML files for GUI V1...
done.

Stop all dbus-serialbattery services...

No Bluetooth battery configuration found in "/data/apps/dbus-serialbattery/config.ini".
You can ignore this, if you are using only a serial connection.

No CAN port configuration found in "/data/apps/dbus-serialbattery/config.ini".
You can ignore this, if you are using only a serial connection.

#################################
# First activation instructions #
#################################

SERIAL battery connection: The activation is complete. You don't have to do anything more.

BLUETOOTH battery connection: There are a few more steps to complete activation.

    1. Add your Bluetooth BMS to the config file "/data/apps/dbus-serialbattery/config.ini".
       Check the default config file "/data/apps/dbus-serialbattery/config.default.ini" for more informations.
       If your Bluetooth BMS are nearby you can show the MAC address with "bluetoothctl devices".

    2. Make sure to disable Bluetooth in "Settings -> Bluetooth" in the remote console/GUI to prevent reconnects every minute.

    3. Run "/data/apps/dbus-serialbattery/restart.sh", if the Bluetooth BMS were not added to the "config.ini" before.

    ATTENTION!
    If you changed the default connection PIN of your BMS, then you have to pair the BMS first using OS tools like the "bluetoothctl".
    See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl for more details.

CAN battery connection: There are a few more steps to complete activation.

    1. Add your CAN port to the config file "/data/apps/dbus-serialbattery/config.ini".
       Check the default config file "/data/apps/dbus-serialbattery/config.default.ini" for more informations.

    2. In the remote console/GUI, go to 'Settings -> Services -> VE.Can port -> CAN-bus profile' and select a profile with a bitrate that
       matches your BMS.

    3. Re-run "/data/apps/dbus-serialbattery/restart.sh", if the CAN port was not added to the "config.ini" before.

CUSTOM SETTINGS: If you want to add custom settings, then check the settings you want to change in "/data/apps/dbus-serialbattery/config.default.ini"
                 and add them to "/data/apps/dbus-serialbattery/config.ini" to persist future driver updates.

*** dbus-serialbattery v2.0.20250103dev was installed. ***

When I read the logging:

-rw-r--r--    1 root     root             0 Jan  3 13:28 state
root@raspberrypi2:/data/log/dbus-serialbattery.ttyUSB3# cat current
@400000006777e6002d9d6bec ./run: line 3: /opt/victronenergy/dbus-serialbattery/start-serialbattery.sh: No such file or directory
@400000006777e6012ede776c ./run: line 3: /opt/victronenergy/dbus-serialbattery/start-serialbattery.sh: No such file or directory

so nightly build does not work (going back to beta)

### Driver version of the currently installed driver

v2.0.20250103dev

### Driver version of the last known working driver

v1.5.20241119beta 

### Venus OS device type

Raspberry Pi

### Venus OS version

3.53~2

### BMS type

JKBMS (Heltec BMS)

### Cell count

22

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
NA
```

### Relevant log output

```shell
When I read the logging:

-rw-r--r--    1 root     root             0 Jan  3 13:28 state
root@raspberrypi2:/data/log/dbus-serialbattery.ttyUSB3# cat current
@400000006777e6002d9d6bec ./run: line 3: /opt/victronenergy/dbus-serialbattery/start-serialbattery.sh: No such file or directory
@400000006777e6012ede776c ./run: line 3: /opt/victronenergy/dbus-serialbattery/start-serialbattery.sh: No such file or directory
```

nicobresseleers, 2025-01-03:
root@raspberrypi2:/opt/victronenergy/dbus-serialbattery# cat start-serialbattery.sh
#!/bin/bash

# remove comment for easier troubleshooting
#set -x

. /opt/victronenergy/serial-starter/run-service.sh

# app=$(dirname $0)/dbus-serialbattery.py

# start -x -s $tty
app="python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py"
args="/dev/$tty"
start $args
root@raspberrypi2:/opt/victronenergy/dbus-serialbattery#

mr-manuel, 2025-01-03:
You reboot after installation, right?

nicobresseleers, 2025-01-03:
i restarted processes , not rebooted ( also not mentioned it should be ..) 

night build is now working ( just need to check and alter parameters in config.ini

closed 2025-01-03 by mr-manuel