mr-manuel/venus-os_dbus-serialbattery #155: wrong displayed cell voltage when cell voltage is over 3.45V (or higher than float cell voltage?)

Olilala, 2025-01-06:
### Describe the problem

I saw the following wrong cell voltage ( > 19V !!!) , when real cell voltage is higher than 3.45V respectively 3.46V. I don't know the right limit value. Battery is in this moment by 100% SoC.

Venus OS:
[image]

BMS:
[image]

[image]

[image]

A few Minutes later, the deye inverter needs power from Battery and starts to discharge, the battery voltage drops and everything runs back in normal condition, the cell voltage seems In a good range again:
[image]

I don't know why, but this time the tai64nlocal report doesn't save this failure. I had the same error on 2024-12-28 the first time, there was a tai64nlocal report with this error! I attached both reports.  

Further information: I had two times same battery Pack, but they are not connected together. One battery works for the victron pv-system, for which I have the raspi with Venus os, the other (in this case covered) battery is for a second pv plant and I use Venus os only for monitoring battery voltages and conditions.

### Driver version of the currently installed driver

1.5.20241512

### Driver version of the last known working driver

/

### Venus OS device type

Raspberry Pi

### Venus OS version

3.51

### BMS type

EG4 LiFePower (Revov, Tian Power)

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 160.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0
FLOAT_CELL_VOLTAGE = 3.45
BMS_TYPE = EG4_Lifepower
```

### Relevant log output

```shell
2024-12-28 13:42:16.254206500 *** CCGX booted (0) ***
2024-12-28 13:44:31.180144500 INFO:SerialBattery:
2024-12-28 13:44:31.180520500 INFO:SerialBattery:Starting dbus-serialbattery
2024-12-28 13:44:31.182426500 INFO:SerialBattery:Venus OS v3.51 running on Raspberry Pi 3 Model B Rev 1.2
2024-12-28 13:44:31.182867500 INFO:SerialBattery:dbus-serialbattery v1.5.20241215
2024-12-28 13:44:47.194003500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-12-28 13:44:47.194554500 INFO:SerialBattery:Testing EG4_Lifepower at address "\x01"
2024-12-28 13:44:47.246508500 INFO:SerialBattery:Hardware Version for address \x01: BTBI23061100091 
2024-12-28 13:44:47.307440500 INFO:SerialBattery:Firmware Version for address \x01: 3BN-HES16S48V200LT55-V1.1.0
2024-12-28 13:44:47.570945500 INFO:SerialBattery:-- Connection established to EG4_Lifepower
2024-12-28 13:44:48.538707500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 2
2024-12-28 13:44:48.783373500 INFO:SerialBattery:Use DeviceInstance: 2
2024-12-28 13:44:48.784583500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.ttyUSB5
2024-12-28 13:44:48.833177500 INFO:SerialBattery:Polling interval: 2.000 s
2024-12-28 13:44:48.840840500 INFO:SerialBattery:Battery EG4 Lifepower connected to dbus from /dev/ttyUSB5
2024-12-28 13:44:48.844051500 INFO:SerialBattery:========== Settings ==========
2024-12-28 13:44:48.845292500 INFO:SerialBattery:> Connection voltage: 54.76 V | Current: 0.0 A | SoC: 100.0%
2024-12-28 13:44:48.846667500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-12-28 13:44:48.847816500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-12-28 13:44:48.848804500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.400 V
2024-12-28 13:44:48.849501500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 160.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A
2024-12-28 13:44:48.850196500 INFO:SerialBattery:> CVCM:     True
2024-12-28 13:44:48.850931500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-12-28 13:44:48.851715500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-12-28 13:44:48.852507500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-12-28 13:44:48.853200500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-12-28 13:44:48.854047500 INFO:SerialBattery:Serial Number/Unique Identifier: BTBI23061100091_280.0Ah
2024-12-28 20:37:45.279982500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-28 20:37:45.281738500 INFO:SerialBattery:cell_voltages_good: True - min: 3.318 > 3.25 - max: 3.326 < 3.35
2024-12-29 00:36:28.395661500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-29 00:36:28.397511500 INFO:SerialBattery:cell_voltages_good: True - min: 3.315 > 3.25 - max: 3.318 < 3.35
2025-01-02 10:39:46.890222500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-02 10:39:46.891973500 INFO:SerialBattery:cell_voltages_good: True - min: 3.259 > 3.25 - max: 3.263 < 3.35
2025-01-03 12:50:05.178909500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-03 12:50:05.180634500 INFO:SerialBattery:cell_voltages_good: True - min: 3.337 > 3.25 - max: 3.34 < 3.35
2025-01-04 09:17:52.586965500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-04 09:17:52.588758500 INFO:SerialBattery:cell_voltages_good: True - min: 3.297 > 3.25 - max: 3.303 < 3.35
```

### Any other information that may be helpful

Last Time, when there was the same error:

2024-12-28 12:30:16.846740500 INFO:SerialBattery:========== Settings ==========
2024-12-28 12:30:16.847805500 INFO:SerialBattery:> Connection voltage: 53.9 V | Current: 0.0 A | SoC: 100.0%
2024-12-28 12:30:16.848689500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-12-28 12:30:16.850348500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-12-28 12:30:16.851499500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.400 V
2024-12-28 12:30:16.852628500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 160.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A
2024-12-28 12:30:16.853775500 INFO:SerialBattery:> CVCM:     True
2024-12-28 12:30:16.854748500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-12-28 12:30:16.856700500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-12-28 12:30:16.857715500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-12-28 12:30:16.858756500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-12-28 12:30:16.859792500 INFO:SerialBattery:Serial Number/Unique Identifier: BTBI23061100091_280.0Ah
2024-12-28 12:44:58.052822500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-28 12:44:58.054304500 INFO:SerialBattery:cell_voltages_good: False - min: 3.346 > 3.25 - max: 19.801 < 3.35
2024-12-28 13:31:31.079402500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-28 13:31:31.080812500 INFO:SerialBattery:cell_voltages_good: False - min: 3.343 > 3.25 - max: 19.786 < 3.35
2024-12-28 13:32:05.116117500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-12-28 13:32:05.117668500 INFO:SerialBattery:cell_voltages_good: False - min: 3.343 > 3.25 - max: 19.787 < 3.35

mr-manuel, 2025-01-06:
@dchiquito could you check?

dchiquito, 2025-01-06:
I believe the issue is that the second highest bit is being set as part of the EG4 LifePower protocol. What information that's supposed to convey I have no idea, but masking that bit away makes the aberrant voltages seem normal again: 19.845V = `19845 & 0x3fff = 3461` = 3.461 V, possibly an overvoltage but it's at least a plausible number.

closed 2025-01-06 by mr-manuel

mr-manuel, 2025-01-06:
@Olilala please re-install the nightly and check, if the issue is solved.

Olilala, 2025-01-06:
Nightly version (2.0.20250103dev) is installed, but I think it will takes a few days, before battery is still charged back to 100% because of less sunny days in germany :-)

Olilala, 2025-01-12:
Hi, today battery reached again 100% and I saw cell_voltages above 3.6V in Venus OS without wrong values about 19V.
So I think nightly version (2.0.20250103dev) is ok in this part.

PR #156 (2025-01-06): Correct the mask on EG4 LifePower Cell voltages
The reverse engineered EG4 LifePower protocol documentation mentioned that the high bit on cell voltages would ocassionally be set, possibly relating to a high power alarm: https://github.com/slim-bean/powermon?tab=readme-ov-file#group-1

Apparently the second highest bit can also be set sometimes, so that also needs to be masked out. Adjust the mask accordingly.

Fixes #155