mr-manuel/venus-os_dbus-serialbattery #159: Upgrading Venus OS to v3.53 breaks serialbattery "ModuleNotFoundError: No module named 'dataclasses'"

haraldhh, 2025-01-08:
### Describe the bug

I upgraded Venyus OS on my Raspberry 3b+ to v3.53 through VRM. After a reboot serialbattery failed with "ModuleNotFoundError: No module named 'dataclasses'". 

I re-installed using nightly, still the same issue. 

### How to reproduce

1. Upgrade Venus OS to v3.53
2.  reinstall serialbattery nightly
3. expect it to load

### Expected behavior

expect it to load

### Driver version of the currently installed driver

v2.0.20250107dev

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.53

### BMS type

JKBMS (Heltec BMS)

### Cell count

4

### Battery count

1

### Connection type

CAN

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
; MAX_BATTERY_CHARGE_CURRENT = 100.0
; MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; 2025-01-01 Harald
CAN_PORT = can8
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
CVCM_ENABLE = True

; 2025-01-02 Harald, balancing @ 3.45V, get them some more time by using 3.500V
MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.500

;  2025-01-02 Harald
BMS_TYPE = Jkbms_Can

; 2025-01-03 
; Victron float;
;FLOAT_CELL_VOLTAGE = 3.375
; Andy / Offgrid-garage float;
;FLOAT_CELL_VOLTAGE = 3.35
; My float? 
FLOAT_CELL_VOLTAGE = 3.4
; Specify the SoC level where CVL limit is reset to max voltage.
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 95
; -- CVL reset based on cell voltage difference (linear mode)
; Specify the cell voltage difference where CVL limit is maintained until the difference is equal or lower.
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL        = 0.005

; 2025-01-08
BATTERY_CELL_DATA_FORMAT = 1
```

### Relevant log output

```shell
2025-01-08 13:25:32.697938500 INFO:SerialBattery:
2025-01-08 13:25:32.698368500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-08 13:25:32.700561500 INFO:SerialBattery:Venus OS v3.53 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-08 13:25:32.701023500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev
2025-01-08 13:25:32.858358500 Traceback (most recent call last):
2025-01-08 13:25:32.858364500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 503, in <module>
2025-01-08 13:25:32.858369500     main()
2025-01-08 13:25:32.858371500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 349, in main
2025-01-08 13:25:32.858375500     from bms.daly_can import Daly_Can
2025-01-08 13:25:32.858378500   File "/data/apps/dbus-serialbattery/bms/daly_can.py", line 24, in <module>
2025-01-08 13:25:32.858383500     from can import Message, CanOperationError
2025-01-08 13:25:32.858470500   File "/data/apps/dbus-serialbattery/ext/can/__init__.py", line 81, in <module>
2025-01-08 13:25:32.858475500     from . import util  # isort:skip
2025-01-08 13:25:32.858478500   File "/data/apps/dbus-serialbattery/ext/can/util.py", line 36, in <module>
2025-01-08 13:25:32.858482500     from .interfaces import VALID_INTERFACES
2025-01-08 13:25:32.858486500   File "/data/apps/dbus-serialbattery/ext/can/interfaces/__init__.py", line 7, in <module>
2025-01-08 13:25:32.858513500     from can._entry_points import read_entry_points
2025-01-08 13:25:32.858516500   File "/data/apps/dbus-serialbattery/ext/can/_entry_points.py", line 3, in <module>
2025-01-08 13:25:32.858521500     from dataclasses import dataclass
2025-01-08 13:25:32.858524500 ModuleNotFoundError: No module named 'dataclasses'
```

haraldhh, 2025-01-08:
Downgraded to v3.52. Now serialbattery complains;

```
2025-01-08 13:28:06.206546500 INFO:SerialBattery:
2025-01-08 13:28:06.207009500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-08 13:28:06.209145500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-08 13:28:06.209613500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev
2025-01-08 13:28:07.367744500 INFO:SerialBattery:Detected CAN Bus bitrate: 500 kbps
2025-01-08 13:28:09.375173500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-08 13:28:09.375179500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-08 13:28:09.375322500 INFO:SerialBattery:-- Connection established to Jkbms_Can
2025-01-08 13:28:09.375751500 INFO:SerialBattery:Successful battery connection at can8 and this address None
2025-01-08 13:28:12.033505500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 2
2025-01-08 13:28:12.416490500 INFO:SerialBattery:Use DeviceInstance: 2
2025-01-08 13:28:12.420253500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.can8
2025-01-08 13:28:12.481670500 INFO:SerialBattery:Polling interval: 1.000 s
2025-01-08 13:28:12.483686500 INFO:SerialBattery:Battery JKBMS CAN V2 connected to dbus from can8
2025-01-08 13:28:12.484330500 INFO:SerialBattery:========== Settings ==========
2025-01-08 13:28:12.485804500 INFO:SerialBattery:> Connection voltage: 13.229000000000001 V | Current: 0.0 A | SoC: 83%
2025-01-08 13:28:12.486932500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2025-01-08 13:28:12.488369500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-01-08 13:28:12.498360500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.500 V| FLOAT CELL VOLTAGE: 3.400 V
2025-01-08 13:28:12.501244500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2025-01-08 13:28:12.502403500 INFO:SerialBattery:> CVCM:     True
2025-01-08 13:28:12.503384500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-01-08 13:28:12.504399500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-01-08 13:28:12.505228500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-01-08 13:28:12.505960500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: None
2025-01-08 13:28:12.506872500 INFO:SerialBattery:Serial Number/Unique Identifier: can8
2025-01-08 13:28:13.522060500 Traceback (most recent call last):
2025-01-08 13:28:13.522067500   File "/data/apps/dbus-serialbattery/dbushelper.py", line 855, in publish_battery
2025-01-08 13:28:13.522072500     self.battery.manage_charge_voltage()
2025-01-08 13:28:13.522075500   File "/data/apps/dbus-serialbattery/battery.py", line 571, in manage_charge_voltage
2025-01-08 13:28:13.522080500     self.manage_charge_voltage_limit()
2025-01-08 13:28:13.522083500   File "/data/apps/dbus-serialbattery/battery.py", line 913, in manage_charge_voltage_limit
2025-01-08 13:28:13.522188500     + (f" • penalty_sum: {penalty_sum:.3f} V" if utils.CVL_CONTROLLER_MODE == 1 else "")
2025-01-08 13:28:13.522194500 UnboundLocalError: local variable 'penalty_sum' referenced before assignment
2025-01-08 13:28:17.181331500 INFO:SerialBattery:
2025-01-08 13:28:17.181798500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-08 13:28:17.184231500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-08 13:28:17.184730500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev

```

mr-manuel, 2025-01-08:
Your installed driver version is not matching the config file. Read the changelog please.

haraldhh, 2025-01-08:
Thank You, 

```
2025-01-08 19:32:48.409043500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-08 19:32:48.411209500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-08 19:32:48.411791500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev
2025-01-08 19:32:49.211191500 INFO:SerialBattery:Detected CAN Bus bitrate: 500 kbps
2025-01-08 19:32:51.215138500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-08 19:32:51.215622500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-08 19:32:51.218672500 INFO:SerialBattery:-- Connection established to Jkbms_Can
2025-01-08 19:32:51.219147500 INFO:SerialBattery:Successful battery connection at can8 and this address None
2025-01-08 19:32:51.654217500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-01-08 19:32:51.654979500 INFO:SerialBattery:Stopped dbus-serialbattery with exit code 0
2025-01-08 19:32:54.867431500 INFO:SerialBattery:
2025-01-08 19:32:54.867880500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-08 19:32:54.870239500 INFO:SerialBattery:Venus OS v3.52 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-08 19:32:54.870688500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev
2025-01-08 19:32:55.685520500 INFO:SerialBattery:Detected CAN Bus bitrate: 500 kbps
2025-01-08 19:32:57.690072500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-08 19:32:57.690728500 INFO:SerialBattery:Testing Jkbms_Can
2025-01-08 19:32:57.695883500 INFO:SerialBattery:-- Connection established to Jkbms_Can
2025-01-08 19:32:57.696374500 INFO:SerialBattery:Successful battery connection at can8 and this address None
2025-01-08 19:32:58.603655500 INFO:SerialBattery:Reconnected to previously identified battery with DeviceInstance: 2
2025-01-08 19:32:58.961458500 INFO:SerialBattery:Use DeviceInstance: 2
2025-01-08 19:32:58.975634500 INFO:SerialBattery:Use dbus ServiceName: com.victronenergy.battery.can8
2025-01-08 19:32:59.031071500 INFO:SerialBattery:Polling interval: 1.000 s
2025-01-08 19:32:59.034653500 INFO:SerialBattery:Battery JKBMS CAN V2 connected to dbus from can8
2025-01-08 19:32:59.036549500 INFO:SerialBattery:========== Settings ==========
2025-01-08 19:32:59.037711500 INFO:SerialBattery:> Connection voltage: 13.181999999999999 V | Current: 0.0 A | SoC: 83%
2025-01-08 19:32:59.039918500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2025-01-08 19:32:59.041020500 INFO:SerialBattery:> CHARGE MODE: Linear
2025-01-08 19:32:59.043595500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.500 V| FLOAT CELL VOLTAGE: 3.400 V
2025-01-08 19:32:59.044701500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2025-01-08 19:32:59.045313500 INFO:SerialBattery:> CVCM:     True
2025-01-08 19:32:59.045903500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2025-01-08 19:32:59.046499500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2025-01-08 19:32:59.047073500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2025-01-08 19:32:59.047639500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: None
2025-01-08 19:32:59.048223500 INFO:SerialBattery:Serial Number/Unique Identifier: can8

```

closed 2025-01-08 by mr-manuel

haraldhh, 2025-01-08:
I now booted v3.53 of Venus OS and now I have the "No module named 'dataclasses'" error again

```
2025-01-08 19:35:39.226826500 INFO:SerialBattery:
2025-01-08 19:35:39.227280500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-08 19:35:39.229577500 INFO:SerialBattery:Venus OS v3.53 running on Raspberry Pi 3 Model B Plus Rev 1.3
2025-01-08 19:35:39.230063500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev
2025-01-08 19:35:39.387901500 Traceback (most recent call last):
2025-01-08 19:35:39.387908500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 503, in <module>
2025-01-08 19:35:39.387913500     main()
2025-01-08 19:35:39.387914500   File "/data/apps/dbus-serialbattery/dbus-serialbattery.py", line 349, in main
2025-01-08 19:35:39.387919500     from bms.daly_can import Daly_Can
2025-01-08 19:35:39.387922500   File "/data/apps/dbus-serialbattery/bms/daly_can.py", line 24, in <module>
2025-01-08 19:35:39.387926500     from can import Message, CanOperationError
2025-01-08 19:35:39.388013500   File "/data/apps/dbus-serialbattery/ext/can/__init__.py", line 81, 
```in <module>
2025-01-08 19:35:39.388019500     from . import util  # isort:skip
2025-01-08 19:35:39.388021500   File "/data/apps/dbus-serialbattery/ext/can/util.py", line 36, in <module>
2025-01-08 19:35:39.388026500     from .interfaces import VALID_INTERFACES
2025-01-08 19:35:39.388029500   File "/data/apps/dbus-serialbattery/ext/can/interfaces/__init__.py", line 7, in <module>
2025-01-08 19:35:39.388057500     from can._entry_points import read_entry_points
2025-01-08 19:35:39.388061500   File "/data/apps/dbus-serialbattery/ext/can/_entry_points.py", line 3, in <module>
2025-01-08 19:35:39.388065500     from dataclasses import dataclass
2025-01-08 19:35:39.388068500 ModuleNotFoundError: No module named 'dataclasses'

Booted up v3.52 and it's working again without errors. Some googling suggested the error was due to different ptyhon versions, but both v3.52 and v.3.53 seem to be running 3.8.18.

haraldhh, 2025-01-08:
Looks like "dataclasses" is missing in v3.53 ;

# On v3.52 ; 
root@raspberrypi2:/data/apps/dbus-serialbattery# python
Python 3.8.18 (default, Nov 20 2024, 09:58:26)
[GCC 9.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dataclasses
>>>

# On v3.53 ; 
root@raspberrypi2:~# python
Python 3.8.18 (default, Jan  7 2025, 15:21:51)
[GCC 9.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dataclasses
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'dataclasses'
>>>

Should I file a bug at Victron maybe?

mr-manuel, 2025-01-08:
Very odd. On my end it works. You could try to update from v3.52 to v3.53 again. Maybe they already fixed it.

haraldhh, 2025-01-08:
I ran a "opkg update && opkg install python3-modules" and now serialbattery is happy on v3.53. Is python3-modules something that should be installed manually before serialbattery as a dependency or am I missing something?

mr-manuel, 2025-01-08:
Nothing should be needed to do manually except for running the install script.

mr-manuel, 2025-01-08:
Interesting. Seems only on the Raspberry Pi.

```
root@einstein:~# uname -a
Linux einstein 5.10.109-venus-17 #1 SMP Tue Jan 7 16:10:16 UTC 2025 armv7l GNU/Linux
root@einstein:~# python
Python 3.8.18 (default, Jan  7 2025, 15:21:51)
[GCC 9.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dataclasses
>>>
```

haraldhh, 2025-01-08:
Sorry, I don't have a real Cerbo device here to test with. I might get access to one for testing purposes if needed?

mr-manuel, 2025-01-08:
I have all devices here, don't worry. I already contacted Victron.

mr-manuel, 2025-01-09:
The large images has that module, the normal one doesn't.

It can be installed with `opkg install python3-misc`, it is fixed in v3.60, where it is moved to python3-core and available by default in both image types.

You are one of the few users which uses CAN, therefore no one had this error until now.

Thanks for reporting it!

haraldhh, 2025-01-09:
Thank You for all the work You do! Hope I helped a little. 

Have to try the large image,  nodered is interesting.

spacecabbie, 2025-02-12:
Just wanted to confirm This issue on a new install venus os then update to 3.53 then install off venus-os_dbus-serialbattery (beta release) on a raspberry pi 4b

mr-manuel, 2025-06-03:
Thank you very much for your donation!