mr-manuel/venus-os_dbus-serialbattery #161: Only one Seplos BMS showing up

Panamabu, 2025-01-12:
### Describe the problem

Good day,

I have curretly two battery Packs with the Seplos V16 BMS. Both are connected to USB directly to the two USB of the Original Cerbo. unfortunately only one BMS is recognized by the cerbo. Each pack individually is working, but together only one is showing up.

am i doing anything wrong here?

hope i captured the relevant log files, im not an expert unfortunately\Thank you for the help

### Driver version of the currently installed driver

v1.5.20241215 stable

### Driver version of the last known working driver

none

### Venus OS device type

Cerbo GX

### Venus OS version

V3.52

### BMS type

Seplos

### Cell count

16

### Battery count

1 working individual 2 connected only 1 showing up

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0
BMS_TYPE = Seplos
```

### Relevant log output

```shell
lsusb:

Bus 003 Device 005: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 003 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

root@einstein:~# /data/log/serial-starter/current
-sh: /data/log/serial-starter/current: Permission denied

  GNU nano 4.9.3                                                       /data/log/serial-starter/current                                                                 
@40000000678382e33242cc64 *** starting serial-starter ***
@40000000678382e4127a1464 serstart starting
@40000000678382e413e21874 INFO: loading config file /etc/venus/serial-starter.conf
@40000000678382e50a3f6f9c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@40000000678382e609a9f9e4 INFO: Create daemontools service mk2-dbus.ttyS4
@40000000678382e7269c8f44 INFO: Create daemontools service vedirect-interface.ttyS5
@40000000678382e90ef4ae94 INFO: Create daemontools service vedirect-interface.ttyS6
@40000000678382ea3440462c INFO: Create daemontools service vedirect-interface.ttyS7
@40000000678382ec19f2a594 INFO: Start service mk2-dbus.ttyS4
@40000000678382ec3b4777ec INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@40000000678382ed33086e24 INFO: Start service vedirect-interface.ttyS5
@40000000678382ef11651254 INFO: Create daemontools service dbus-serialbattery.ttyUSB1
@40000000678382ef1f0254e4 INFO: Start service vedirect-interface.ttyS6
@40000000678382f103e8a604 INFO: Start service vedirect-interface.ttyS7
@40000000678382f310557afc INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000678382f522437624 INFO: Start service dbus-serialbattery.ttyUSB1 once
@40000000678382f7266c827c INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@40000000678382fd3aba89cc INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000678382fe034c3724 INFO: Create daemontools service dbus-cgwacs.ttyUSB1
@400000006783830410a9ffb4 INFO: Start service dbus-cgwacs.ttyUSB1 once
@400000006783830414db0704 INFO: Create daemontools service gps-dbus.ttyUSB0
@400000006783830a23381ef4 INFO: Start service gps-dbus.ttyUSB0 once
@400000006783830a3037dd4c INFO: Create daemontools service gps-dbus.ttyUSB1
@4000000067838311091c830c INFO: Start service gps-dbus.ttyUSB1 once
@400000006783831526c0111c INFO: Create daemontools service vedirect-interface.ttyUSB0
@400000006783831b33c61104 INFO: Create daemontools service vedirect-interface.ttyUSB1
@400000006783831b35326e5c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000678383213068d85c INFO: Start service dbus-serialbattery.ttyUSB0 once
@400000006783832206332e5c INFO: Start service vedirect-interface.ttyUSB1 once
@400000006783832710c5d89c INFO: Start service dbus-serialbattery.ttyUSB1 once
@40000000678383763556982c INFO: Start service dbus-cgwacs.ttyUSB0 once
@400000006783837c10ed5dcc INFO: Start service gps-dbus.ttyUSB0 once
@400000006783838524134484 INFO: Start service vedirect-interface.ttyUSB0 once
@400000006783838a32ed6ffc INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000678383e32d646e14 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000678383e915f062ec INFO: Start service gps-dbus.ttyUSB0 once
@40000000678383f22740168c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000678383f735ba3fe4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@400000006783845303939894 INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000067838458145e12bc INFO: Start service gps-dbus.ttyUSB0 once
@4000000067838461288880c4 INFO: Start service vedirect-interface.ttyUSB0 once
@40000000678384663abfc1bc INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000678384bf3182d7c4 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000678384c518637244 INFO: Start service gps-dbus.ttyUSB0 once
@40000000678384ce2dd60fec INFO: Start service vedirect-interface.ttyUSB0 once
@40000000678384d41152789c INFO: Start service dbus-serialbattery.ttyUSB0 once
@400000006783852d034f82e4 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000678385321780809c INFO: Start service gps-dbus.ttyUSB0 once
@400000006783853b29aae49c INFO: Start service vedirect-interface.ttyUSB0 once
@400000006783854039f931b4 INFO: Start service dbus-serialbattery.ttyUSB0 once
```

mr-manuel, 2025-01-12:
Hello, some logfiles are missing. Please read the instructions again: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#serial-bms-connection

Panamabu, 2025-01-12:
ok, i guess the answer is written in the LOG, i gave to change the capacity in the seplos tool by 1Ah then the identical serial number should be fixed....

Panamabu, 2025-01-12:
SOLVED... changing the Ah ratings of the Pack solved the serial number issue, both packs did show up afterwards

mr-manuel, 2025-01-12:
What was not clear in the troubleshooting section, that you did not check the log and that you did not used the command from the docs?

closed 2025-01-12 by mr-manuel