mr-manuel/venus-os_dbus-serialbattery #162: No more access to "Cell Voltages"

DIVERSDIVERS, 2025-01-14:
### Describe the bug

When i go to a Daly Battery doesn´t matter which one and select cell Voltages the Brower freeze and nothing happens, doenst matter if Tablet oder Mobile or Computer (on old GUI it works)

### How to reproduce

1. go to a Battery
2. select Cell Voltages

### Expected behavior

should display Cell Voltages

### Driver version of the currently installed driver

v1.5.20241215

### Venus OS device type

Cerbo GX

### Venus OS version

v3.53 (20250107135716)

### BMS type

Daly BMS

### Cell count

16

### Battery count

5

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

LOGGING = ERROR

MAX_BATTERY_CHARGE_CURRENT = 500.0
MAX_BATTERY_DISCHARGE_CURRENT = 500.0

MIN_CELL_VOLTAGE   = 2.750
MAX_CELL_VOLTAGE   = 3.600
FLOAT_CELL_VOLTAGE = 3.405

SOC_RESET_VOLTAGE  = 3.600

SOC_RESET_AFTER_DAYS = 5

BLOCK_ON_DISCONNECT = False

LINEAR_LIMITATION_ENABLE = True

LINEAR_RECALCULATION_EVERY = 60

LINEAR_RECALCULATION_ON_PERC_CHANGE = 5

CVCM_ENABLE = True

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.025

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.065

CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.075

SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 97

CVL_ICONTROLLER_MODE   = True
CVL_ICONTROLLER_FACTOR = 0.2

CCCM_CV_ENABLE = True

DCCM_CV_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING   = 3.630, 3.625, 3.615, 3.605, 3.59, 3.58, 3.57, 3.56, 3.53
MAX_CHARGE_CURRENT_CV_FRACTION =     0, 0.010, 0.015, 0.025, 0.05, 0.15, 0.20, 0.35,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.75, 2.85, 2.90, 2.95, 3.00
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0, 0.02, 0.25,  0.5,    1

CCCM_T_ENABLE = True

DCCM_T_ENABLE = True

TEMPERATURES_WHILE_CHARGING      = 0,   3,   7,  10,  15,  40,  45,  50,  55, 60
MAX_CHARGE_CURRENT_T_FRACTION    = 0, 0.2, 0.5, 0.8,   1,   1, 0.8, 0.5, 0.2,  0

TEMPERATURES_WHILE_DISCHARGING   = -35,  -25, -10,   0,  5,  40,  50,  55, 60
MAX_DISCHARGE_CURRENT_T_FRACTION =   0, 0.4,  0.6, 0.8,  1,   1, 0.5, 0.2,  0

CCCM_SOC_ENABLE = True

DCCM_SOC_ENABLE = True

SOC_WHILE_CHARGING                 =   99,   98,   97,   96
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.15, 0.25, 0.40, 1.00

SOC_WHILE_DISCHARGING              =    5,   10,   15,   20
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.10, 0.20, 0.50, 1.00

TIME_TO_GO_ENABLE = True

BMS_TYPE = Daly

BATTERY_CELL_DATA_FORMAT = 1

AUTO_RESET_SOC = True

TELEMETRY = False

SOC_LOW_WARNING = 15
SOC_LOW_ALARM   = 10

BATTERY_CAPACITY = 302

INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
2025-01-14 15:07:00.679010500 INFO: Start service dbus-serialbattery.ttyUSB1 once
2025-01-14 15:07:02.412523500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2025-01-14 15:07:04.280981500 INFO: Start service dbus-serialbattery.ttyUSB3 once
2025-01-14 15:07:06.000734500 INFO: Start service dbus-serialbattery.ttyUSB4 once
2025-01-14 15:08:10.635451500 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
2025-01-14 15:08:18.439924500 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
2025-01-14 15:08:20.388418500 INFO: Create daemontools service dbus-serialbattery.ttyUSB1
2025-01-14 15:08:21.986749500 INFO: Create daemontools service dbus-serialbattery.ttyUSB2
2025-01-14 15:08:24.083555500 INFO: Create daemontools service dbus-serialbattery.ttyUSB3
2025-01-14 15:08:24.634926500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2025-01-14 15:08:26.152152500 INFO: Create daemontools service dbus-serialbattery.ttyUSB4
2025-01-14 15:08:26.568760500 INFO: Start service dbus-serialbattery.ttyUSB1 once
2025-01-14 15:08:28.130468500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2025-01-14 15:08:30.408801500 INFO: Start service dbus-serialbattery.ttyUSB3 once
2025-01-14 15:08:32.500008500 INFO: Start service dbus-serialbattery.ttyUSB4 once
2025-01-14 15:09:09.711826500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2025-01-14 15:19:57.568771500 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
2025-01-14 15:20:05.510221500 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
2025-01-14 15:20:07.525122500 INFO: Create daemontools service dbus-serialbattery.ttyUSB1
2025-01-14 15:20:09.493246500 INFO: Create daemontools service dbus-serialbattery.ttyUSB2
2025-01-14 15:20:10.983997500 INFO: Create daemontools service dbus-serialbattery.ttyUSB3
2025-01-14 15:20:11.719199500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2025-01-14 15:20:12.679945500 INFO: Create daemontools service dbus-serialbattery.ttyUSB4
2025-01-14 15:20:13.726167500 INFO: Start service dbus-serialbattery.ttyUSB1 once
2025-01-14 15:20:15.785648500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2025-01-14 15:20:17.247955500 INFO: Start service dbus-serialbattery.ttyUSB3 once
2025-01-14 15:20:18.986373500 INFO: Start service dbus-serialbattery.ttyUSB4 once
2025-01-14 15:21:27.986445500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2025-01-14 15:21:29.788666500 INFO: Start service dbus-serialbattery.ttyUSB1 once
2025-01-14 15:21:33.849447500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2025-01-14 15:21:35.550645500 INFO: Start service dbus-serialbattery.ttyUSB3 once
2025-01-14 15:21:37.510284500 INFO: Start service dbus-serialbattery.ttyUSB4 once
```

RaveRebel, 2025-01-14:
Same Problem since today (14.01.25) with JkBMS

Driver version of the currently installed driver
v1.5.20241215

Venus OS device type
VenusOS (RPI4)

Venus OS version
v3.52 

BMS type
JkBMS

Cell count
16

Battery count
1

Connection type
Serial USB adapter to RS485

mr-manuel, 2025-01-15:
Thanks for reporting. I fixed the issue. Reboot the device, then the GUI is updated automatically.

closed 2025-01-15 by mr-manuel