mr-manuel/venus-os_dbus-serialbattery #163: New Daly BMS Version is not working

AlexanderLux, 2025-01-15:
### Describe the bug

I already have 2 old Version daly bms successful connected over RS485 USB connector.

https://a.aliexpress.com/_ExpwUVS

I added a 3th daly BMS of the New Version (M Series). Same Like the others over RS485 USB connector.
But the BMS get Not connected .

https://a.aliexpress.com/_EQXgVQ0

Maybe you can help fixing it
Thanks ;)

### How to reproduce

Connect New Daly BMS (Series M) to Cerbo GX over RS485 USB connector.

### Expected behavior

New Daly BMS (Series M) Is supported

### Driver version of the currently installed driver

1.5.20241215

### Driver version of the last known working driver

1.5.20241215

### Venus OS device type

Cerbo GX

### Venus OS version

3.50

### BMS type

Daly BMS

### Cell count

16

### Battery count

3

### Connection type

Serial USB adapter to RS485

### Config file

```ini
ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are log
ged
LOGGING = DEBUG

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 180.0
MAX_BATTERY_DISCHARGE_CURRENT = 180.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used for:
;         - Calculating the min/max battery voltage
;         - Trigger the SoC reset when SoC calculation is 
enabled
; Example:
;     16 cells * 3.45 V/cell = 55.2 V max charge voltage.
;     16 cells * 2.90 V/cell = 46.4 V min discharge voltag
e
MIN_CELL_VOLTAGE   = 2.800
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.6
```

### Relevant log output

```shell
@400000006786f6902fa39c2c INFO:SerialBattery:Starting dbus-serialbattery
@400000006786f6903003c944 INFO:SerialBattery:Venus OS v3.50 running on Cerbo GX^@
@400000006786f690304ae07c INFO:SerialBattery:dbus-serialbattery v1.5.20241215
@400000006786f6a03117e4dc INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006786f6a03124aa64 INFO:SerialBattery:Testing Daly at address "\x40"
@400000006786f6a039ccd254 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006786f6a039dedb84 INFO:SerialBattery:Testing Daly at address "\x80"
@400000006786f6a10628f144 ERROR:SerialBattery:>>> ERROR: No reply - returning

@400000006786f6902fa39c2c INFO:SerialBattery:Starting dbus-serialbattery
@400000006786f6903003c944 INFO:SerialBattery:Venus OS v3.50 running on Cerbo GX^@
@400000006786f690304ae07c INFO:SerialBattery:dbus-serialbattery v1.5.20241215
@400000006786f6a03117e4dc INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006786f6a03124aa64 INFO:SerialBattery:Testing Daly at address "\x40"
@400000006786f6a039ccd254 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006786f6a039dedb84 INFO:SerialBattery:Testing Daly at address "\x80"
@400000006786f6a10628f144 ERROR:SerialBattery:>>> ERROR: No reply - returning
```

### Any other information that may be helpful

…

mr-manuel, 2025-01-15:
Please ask the vendor, if they have the same protocol amd if not ask for the protocol documentation.

transistorgit, 2025-02-10:
Hi,

I have 2 100balance BMS and connected them successfully with serial battery on a raspi. I used a newer driver version, so maybe you want to give it another try.
 
Make sure to check the following items:
* Wiring (reverse A/B for test)
* termination resistor (configurable by BMS App)
* Address in BMS App configured to "1"
* protocol type in BMS configured to None/RS485
* is the new daly-supplied RS485 converter compatible to cerbo? check syslog and lsusb if it is recognised.
* newest driver version

best regards

transistorgit, 2025-02-21:
Hi, I use the original Rs485 converters that were supplied as accessory to the 100balance bms. One converter per bms. I use an isolating USB hub between the converters and the pi. 
See the attached picture. At bottom right is the usb hub with the 2 converters. 

Best regards
Bernd

[image]

mr-manuel, 2025-03-10:
No update from @AlexanderLux since nearly two months. If someone has the same problem, we need the protocol documentation to verify that.

closed 2025-03-10 by mr-manuel

kr0815, 2025-03-22:
> Hi,
>
> I have 2 100balance BMS and connected them successfully with serial battery on a raspi. I used a newer driver version, so maybe you want to give it another try.
>
> Make sure to check the [...]

ery interesting - trying to get a second battery pack to work, ordered one of the new blue daly bms

So far no communication with dbus-srialbatery 

But i you got it working, there must be a way ...