mr-manuel/venus-os_dbus-serialbattery #165: Second JBD BMS not detected on /dev/ttyUSB2 (No reply error)

thenebu, 2025-01-16:
### Describe the bug

I have two JBD BMS connected to my Cerbo GX via RS485-to-USB adapters. Both BMS devices are connected and configured using dbus-serialbattery. The first BMS is correctly detected and working via /dev/ttyUSB1. However, the second BMS on /dev/ttyUSB2 fails to initialize. It appears that dbus-serialbattery is assigned to /dev/ttyUSB2, but the logs show repeated errors indicating no reply from the BMS.

Additional Notes:
	•	The BMS connected to /dev/ttyUSB1 is working perfectly and is displayed in Venus OS.
	•	The BMS connected to /dev/ttyUSB2 is not displayed in Venus OS despite being assigned to dbus-serialbattery.
	•	Both RS485-to-USB adapters are identical and verified to be working.
	•	The second BMS does not seem to respond to dbus-serialbattery.

Could this be an issue with the handling of multiple JBD BMS devices by dbus-serialbattery? Or might it be related to a misconfiguration or compatibility issue?

Any guidance or suggestions for resolving this issue would be greatly appreciated.

### How to reproduce

I’ve tried the following steps to troubleshoot, but the issue persists:
	1.	Verified the /dev/ttyUSB* devices using ls -l /dev/ttyUSB*.
	2.	Checked the service status using svstat:
	•	/service/dbus-serialbattery.ttyUSB1 is up and working.
	•	/service/dbus-serialbattery.ttyUSB2 repeatedly goes down after failing to initialize.
	3.	Inspected the logs using:
	`tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB2/current | tai64nlocal`
	4.	Rebooted the Cerbo GX multiple times.
	5.	Tried using the latest nightly build of dbus-serialbattery.

Service Status (svstat): `/service/dbus-serialbattery.ttyUSB1: up (pid 2911) 453 seconds
/service/dbus-serialbattery.ttyUSB2: down 14 seconds, normally up`

### Expected behavior

both bms shown in VenusOS

### Driver version of the currently installed driver

latest stable and nightly 2.0.20250107dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.53

### BMS type

Jiabaida JDB BMS (LLT Power, Overkill Solar, Smart BMS)

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
USE_PORT_AS_UNIQUE_ID = TRUE
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = TRUE
PUBLISH_CONFIG_VALUES = True
BMS_TYPE = LltJbd
```

### Relevant log output

```shell
(/data/log/dbus-serialbattery.ttyUSB2/current):
2025-01-16 18:41:36.985384500 INFO:SerialBattery:Starting dbus-serialbattery
2025-01-16 18:41:37.027187500 INFO:SerialBattery:Venus OS v3.53 running on Cerbo GX
2025-01-16 18:41:37.028157500 INFO:SerialBattery:dbus-serialbattery v2.0.20250107dev
2025-01-16 18:41:53.045344500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2025-01-16 18:41:53.046154500 INFO:SerialBattery:Testing LltJbd at address "\x00"
2025-01-16 18:41:53.345050500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-16 18:41:53.853896500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2025-01-16 18:41:53.854760500 INFO:SerialBattery:Testing LltJbd at address "\x00"
2025-01-16 18:41:54.143519500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-16 18:41:54.654008500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2025-01-16 18:41:54.654018500 INFO:SerialBattery:Testing LltJbd at address "\x00"
2025-01-16 18:41:54.949028500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2025-01-16 18:41:55.454789500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB2
2025-01-16 18:41:55.455633500 INFO:SerialBattery:Exit signal received, exiting gracefully...
2025-01-16 18:41:55.456376500 INFO:SerialBattery:Stopped dbus-serialbattery with exit code 1
```

mr-manuel, 2025-01-16:
You could try to switch the serial cable connectors on the BMS side to see if BMS2 works with the cable and adapter of BMS1.

thenebu, 2025-01-16:
Sorry, I forgot to mention earlier, but I already tested this as one of the very first troubleshooting steps. I have 4 RS485-to-USB adapters available, and none of them worked with BMS2. However, BMS1 works perfectly fine even with the cable and adapter from BMS2.

To further isolate the issue, I set up a Raspberry Pi 4 and connected BMS2 to it, running the same setup. Unfortunately, I encountered the exact same issue as on the Cerbo GX, with the same “no reply” error in the logs.

I also happen to have an identical spare BMS here, and I’m planning to swap out BMS2 with this spare tomorrow to see if it’s a hardware issue with the second BMS itself.

Do you have any other suggestions or ideas in the meantime? Any help would be greatly appreciated!

mr-manuel, 2025-01-17:
If the BMS are identical, then it‘s very likely a hardware issue.

thenebu, 2025-01-17:
I sincerely apologize for the oversight on my part. After hours of troubleshooting, I realized that I had only swapped the one side of the cable (the RS485-to-USB adapter side). Once I swapped the entire cable, including the connection from the BMS to the adapter, everything started working correctly!

It turns out the issue was with the cable itself. I deeply regret any time I took away from you while trying to assist me. Thank you for your patience and support!

closed 2025-01-17 by thenebu

mr-manuel, 2025-01-17:
You‘re welcome! Feel free to make a donation :-) any amount is welcome

thenebu, 2025-01-17:
Did already yesterday! 15€ + fee