mr-manuel/venus-os_dbus-serialbattery #171: How to make the dubs serialdriver dumb?

disaster123, 2025-01-25:
### Describe the problem

Hello,

I tried to search GitHub and the faq as well as the documentation, but didn’t found a solution.

I like the serialdriver especially for having much more information about the bms status. But I still want that the bms controls everything. This makes it more easier for me to debug problems.

So I would like that all decisions are made by my victron or the bms not by the serialdriver. It should just pass all values it gets from the bms.

How can I make it dumb like the normal can bus connection?

Greets,
Stefan

### Driver version of the currently installed driver

v1.5.20241215

### Driver version of the last known working driver

None

### Venus OS device type

Raspberry Pi

### Venus OS version

V3.52

### BMS type

JKBMS Inverter (JKBMS PB Model)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
Defaults
```

### Relevant log output

```shell
None
```

disaster123, 2025-01-25:
Or i might call this passthrough mode. I looked into the code but it seems there is no mode like that. Is there any reason?

disaster123, 2025-01-25:
OK it seems SEPLOS driver has such a functionality: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/d08cec11c30877ad05af50851068f557c8930c8b/dbus-serialbattery/bms/seplosv3.py#L208

=> USE_BMS_DVCC_VALUES

I wanted to like to have the same for my JK inverter BMS.

mr-manuel, 2025-01-26:
Just disable all CVL, CCL and DCL limits and set the voltages and charge/discharge current the same as on the BMS.

disaster123, 2025-01-26:
Sadly that does not work as the bms changes
those values based on several rules. But I saw that the bms does not seem to publish those values through rs485 but only CAN… so it might be impossible or we need to find a way to get those values through rs485 as well

mr-manuel, 2025-01-29:
The driver does not provide this values. Anyway I would not recommend it, since the BMS is only the last way of protection.

closed 2025-01-29 by mr-manuel